espressif/esp-hosted #255: 使用iw工具和wpa工具扫描不了wifi

JS-Ye, 2023-08-10:
### ESP 芯片/产品平台：
* **ESP芯片:** ESP32-WROOM-32UE
* **产品平台:** ESP-Hosted-NG Release 1.0.2
* **软件:** https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng/host

### 问题描述
客户执行 ifconfig -a 可以找到espsta0无线网卡驱动，但是使用 iw工具 和 wpa工具，扫描不了wifi，报错了

JS-Ye, 2023-08-10:
This is the information for running "ifconfig -a" :
[image]

kapilkedawat, 2023-08-10:
Could you please stop network manager, kill wpa_supplicant and hostapd and retry?

sudo service NetworkManager stop
sudo killall -9 wpa_supplicant
sudo killall -9 hostapd

JS-Ye, 2023-08-11:
@kapilkedawat There is no NetworkManager and hostap.
[image]

JS-Ye, 2023-08-11:
@kapilkedawat I could only kill wpa_supplicant, but I couldn't scan the WiFi after restarting.

kapilkedawat, 2023-08-11:
hi @JS-Ye , which OS is this? there should be something similar to network manager.

JS-Ye, 2023-08-11:
@kapilkedawat Embedded linux system. The compiled kernel size is about 2M. The file system is made by buildroot, and the file system is 10M.

kapilkedawat, 2023-08-11:
In my opinion, wpa_supplicant or similar utility is already controlling the interface which is causing new one to fail. 

Can you please try following?
1. reboot the system
2. issue scan command directly to wpa_cli without killing the supplicant (wpa_cli -i espsta0 scan)

JS-Ye, 2023-08-11:
@kapilkedawat Now with the image method, the "rfkill: Cannot open RFKILL control device" error does not appear.
[image]

JS-Ye, 2023-08-11:
Still can't scan the Wi-Fi
[image]

kapilkedawat, 2023-08-11:
command is getting timed out, can you share firmware logs as well(esp32 side of logs, can be obtained using idf.py monitor).

kapilkedawat, 2023-08-14:
@JS-Ye these are host side logs, could you pls get esp32 firmware side of logs?

JS-Ye, 2023-08-18:
@kapilkedawat This is esp32 firmware side of logs.  
[image]

log：[esp模块日志.txt](https://github.com/espressif/esp-hosted/files/12375333/esp.txt)

kapilkedawat, 2023-08-19:
Hi @JS-Ye , thanks for providing the logs.

It looks like either chip was not able to start the scan or it didn't send the response back to host for scan issue command. 
We would need you help to debug it further. Can you please apply attached patch and [set firmware logs level to debug](https://github.com/espressif/esp-idf/blob/master/components/log/Kconfig#L25) and reshare firmware side logs?

shreyassheth1, 2023-08-20:
Also please add this patch with the one shared by @kapilkedawat and take logs for host and firmware simultaneously.
[host_debug.zip](https://github.com/espressif/esp-hosted/files/12388393/host_debug.zip)

JS-Ye, 2023-08-21:
@kapilkedawat @shreyassheth1 
Excuse me，where do I add these two patches?

shreyassheth1, 2023-08-21:
@JS-Ye You have to download these both the patches and then have to apply it using git apply "patch_name" in esp_hosted folder.

JS-Ye, 2023-08-22:
@shreyassheth1 esp-hosted/esp_hosted_fg/esp/esp_driver/network_adapter Compile fail.
[image]

zrb2002, 2023-08-23:
Is there any progress on this issue

shreyassheth1, 2023-08-23:
Have combined and updated the patch please try to compile with this patch and try.
[host_print_debug.zip](https://github.com/espressif/esp-hosted/files/12416162/host_print_debug.zip)
Also please make sure you are using correct compiling environment from esp-idf folder sdk esp_driver folder as mentioned in these [instructions](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/README.md)

JS-Ye, 2023-08-24:
@shreyassheth1   I have added the patch for you provided, but the compilation still failed. Could you please provide the compiled firmware for us to test first?  
[image]  
[image]

zrb2002, 2023-08-24:
Can you provide a test bin file

shreyassheth1, 2023-08-25:
Here are the compiled bins with patch and debug logs enabled.
[hosted_patch_bin.zip](https://github.com/espressif/esp-hosted/files/12436633/hosted_patch_bin.zip)

JS-Ye, 2023-08-25:
@shreyassheth1 What are the burning addresses of these bin files?

JS-Ye, 2023-08-25:
@shreyassheth1 Can you refer to this?  
[image]

shreyassheth1, 2023-08-25:
Please use this command @JS-Ye 
esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32c3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x0 bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 network_adapter.bin

shreyassheth1, 2023-08-25:
Here a compiled bin for esp32.
[hosted_patch_bin.zip](https://github.com/espressif/esp-hosted/files/12437550/hosted_patch_bin.zip)

JS-Ye, 2023-08-28:
@shreyassheth1 We need is SPI only, but you give is SDIO only. Please provide a bin file about SPI only of ESP32 again.  

[image]

shreyassheth1, 2023-08-28:
Hey @JS-Ye here are compiled binaries for esp32 spi only with debug logs 
[hosted_patch_bin.zip](https://github.com/espressif/esp-hosted/files/12452802/hosted_patch_bin.zip)

JS-Ye, 2023-08-29:
@shreyassheth1 I downloaded the bin file you provided, but I still have problems in debugging.  

[image]

### LOG
[esp32调试信息.txt](https://github.com/espressif/esp-hosted/files/12459328/esp32.txt)  
[host调试信息.txt](https://github.com/espressif/esp-hosted/files/12459329/host.txt)

tinswf, 2023-08-31:
more debug log
[esp32_20230831_2.txt](https://github.com/espressif/esp-hosted/files/12483087/esp32_20230831_2.txt)
[host_20230831_2.txt](https://github.com/espressif/esp-hosted/files/12483088/host_20230831_2.txt)

tinswf, 2023-08-31:
Prepare command request for scan
Authentication request: 42:bd:1c:9e:d8:06 1 0 0 0
esp32: Command[11] timed out

tinswf, 2023-08-31:
log
[esp32_20230831_3.txt](https://github.com/espressif/esp-hosted/files/12484617/esp32_20230831_3.txt)
[host_20230831_3.txt](https://github.com/espressif/esp-hosted/files/12484618/host_20230831_3.txt)

tinswf, 2023-08-31:
[image]

tinswf, 2023-08-31:
now we do not use iwlist to scan wifi
1、ifconfig espsta0 up
2、export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/clou/ppp/lib:/clou/lib:/mnt/data/testwifi/lib
3、/mnt/data/testwifi/wpa_supplicant -D nl80211 -c /mnt/data/testwifi/wpa_supplicant.conf -i espsta0

> Hey @JS-Ye we have taken a look at logs and it seems that you were able to resolve the earlier issue in which supplicant was not able to issue scan command. Now supplicant is able to scan and is trying to connect with ap(ssid: abcde). Looks like the connection has failed and failed during the assoc state.
> 
> Can you please capture sniffer logs which will help us to know why this is failing.
> 
> We will also share you a debug firmware with enhanced debugging. Please use that while capturing the logs and sniffer. [hosted_esp32_debug.zip](https://github.com/espressif/esp-hosted/files/12484087/hosted_esp32_debug.zip)

shreyassheth1, 2023-09-01:
Hey @tinswf 
Thanks for the recent set of logs. it looks like host has issued the command however slave didn't receive that. Can you please check the SPI connection whether it's correct, you can run raw throughput test for the spi to validate the connection [Raw_TP_Testing.md](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md)

kapilkedawat, 2023-10-10:
Hi @tinswf , any update on this issue. Were you able to proceed further?

mantriyogesh, 2023-11-28:
@JS-Ye any update ahead?
Please feel free to close the issue if everything is working fine.