espressif/esp-hosted #259: Nothing Happens

hackintoshlover, 2023-08-31:
I am using a Raspberry Pi 4 with Raspbian OS Buster 32bit.
Flashed the firmware on a esp32 dev board.
Flashed efuse to support emmc.
Connected to raspberry pi using sdio pins with pullup-resistors.
Compiled esp_hosted-ng, followed all steps.
BUT NOTHING HAPPENS.
Attached is the dmesg log for your reference.
Need Help.
[Dmesg.txt](https://github.com/espressif/esp-hosted/files/12484186/Dmesg.txt)

hackintoshlover, 2023-09-01:
> Can you please state the firmware/source code used, esp side logs, raspberry pi side logs (dmesg at that time)?
> 
> how much ext pull-up values?
> 
> Also, what is the **wire length**, is it possible to test over **pcb**? problem is that sdio (any sdio) is very much prone to the signal integrity, and gives issues on manually connected wires.
> 
> you can also try to reduce the frequency of sdio and try? FG reference code here: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c#L774-L778 copy the piece in your **NG**'s esp_spio.c .
> 
> Also, setup image? Finally, [timing adjustment](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md#242-sdio) to 'NP' could be tried? The default timing is PP. changing to NP is last sort (we know Raspberry Pi follows PP for ESP32 SDIO, suspecting wires introducing timing issues)

Used pre-built firmware from esp hosted releases page https://github.com/espressif/esp-hosted/releases/tag/release%2Fng-v1.0.2

Attached error.txt files contains all logs from esp32 as well as raspberry pi.

Used 10k pull-up resistors with wire length smaller than 10cm, made a breakout board and tested over pcb.

Tried reducing frequency, no effect.

Tried timing adjustment, no effect.

[Error.txt](https://github.com/espressif/esp-hosted/files/12496046/Error.txt)
[esp32 console log.txt](https://github.com/espressif/esp-hosted/files/12496047/esp32.console.log.txt)

hackintoshlover, 2023-09-13:
Hi,
Any help here?

mantriyogesh, 2023-09-14:
In esp log, the data path is not started.
Have you configured the /boot/config.txt as per documentation?

mantriyogesh, 2023-09-14:
You can also try manual flashing the esp binary on master and use master code on host , with logs added in cases where returns are there in:

https://github.com/espressif/esp-hosted/blob/43267cfee077276907518f3eba53fda93304f736/esp_hosted_ng/host/sdio/esp_sdio.c#L656-L691

mantriyogesh, 2023-09-26:
any updates here?

hackintoshlover, 2023-09-26:
Hi yogesh,
Couldn't post was a bit occupied.
I tried all the above steps also tried esp-hosted-fg, still nothing works.
I can confirm sdio is working since sdmmc works also raspberry pi dmesg identifies it.
Looking at the esp32 serial i don't see a communication being initiated.
Any steps you would suggest?

mantriyogesh, 2023-09-26:
Actually, in your ESP log, I couldn't see 'start data path' or something like that in the end.
that really means that https://github.com/espressif/esp-hosted/blob/43267cfee077276907518f3eba53fda93304f736/esp_hosted_ng/host/sdio/esp_sdio.c#L780 was never called.

This means that esp_probe was failed. generally the reason for probe failing is sdio signal integrity issue.
meaning, the cables are prone to interference. small (<6cm cables) and good quality cables, verifying external pull-ups, connecting all grounds of host and slave etc can help.

you can also try 1 bit sdio mode, which faces less signal integrity probs :
https://github.com/espressif/esp-hosted/issues/242#issuecomment-1713662854

once tested, you can form integrated pcb to test 4 bit mode.

mantriyogesh, 2023-11-27:
is 

```
dtoverlay=sdio,poll_once=off
```
added in /boot/config.txt?

pls send output of:
```sh
$ cat /boot/config.txt
```

mantriyogesh, 2023-11-28:
Let us know current state, to assist ahead..
If the things working fine, please help to close the issue..