espressif/esp-hosted #260: ad-hoc(mesh) mode support

Jason-xy, 2023-08-31:
Will this project support ad-hoc (mesh) mode in the future?
Is it feasible to apply the esp32 mesh library to implement ad-hoc features under Linux in this project?

kapilkedawat, 2023-09-01:
Hi @Jason-xy , esp mesh is not supported at the moment with both fg and ng solution. Do let us know your use-case and we will evaluate the same.

Jason-xy, 2023-09-01:
> Hi @Jason-xy , esp mesh is not supported at the moment with both fg and ng solution. Do let us know your use-case and we will evaluate the same.

Thank you for your response. We are conducting research on robot swarm, and the robots we are using run on Ubuntu. What we need is a network card that supports ad-hoc networking to enable decentralize and distributed data exchange among the robots. Previously, we attempted to use a Broadcom network card that supports AirDrop on MacBook, but we encountered issues with discontinuation and drivers. Therefore, our plan is to develop our own network card that supports ad-hoc networking using the ESP32 based on this project.

kapilkedawat, 2023-10-06:
Hi @Jason-xy, sorry for the delayed response, we currently have no plan to implement adhoc mode for ng solution in near future. I'll update this thread if that changes.