espressif/esp-hosted #262: ESP32C3 espsta0 interface does not appear

linchanghe123, 2023-09-13:
I transplanted the esp-hosted code to the stm32mp157 platform, and the esp32c3 firmware uses ESP-Hosted-NG Release 1.0.2 firmware. espsta0 device is not generated under linux.

The device tree configuration of the Linux system is as follows:
 
[image]

The linux driver only modified the configuration of handshake, data_ready, and reset pins.
 
[image]

However, after loading esp32_spi.ko, the bootup information of esp32c3 was not received, and the espsta0 device was not generated.
 
[image]

Both sides use SPI mode 2.
This is information about related signals when esp32_spi.ko is loaded
[image]
[image]
[image]
What could be the reason for this?

mantriyogesh, 2023-09-13:
Hello @linchanghe123 

This problem is already been answered I believe.
@Shreyas0-7  had sent you the response:

> Subject: Re: question about esp-hosted
>  
> can you please mention which commit you are using.
> 
> **Question 1: I cannot compile the esp32c3 firmware using the idf v5.0.1 tool**
>  Can you please make sure you have followed these steps correctly https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/README.md
> most importantly please verify you are using idf tools using export.sh from this path only esp-hosted/master/esp_hosted_ng/esp/esp_driver/esp-idf in a fresh terminal and then try to build the firmware
> In addition, compiling esp_hosted_ng firmware using idf tool v5.0.1 is not supported and is only supported by idf tool present in this path esp-hosted/master/esp_hosted_ng/esp/esp_driver/esp-idf.
> 
> **Question 2: The espsta0 device cannot be generated under Linux**
> Once the firmware is flashed and spi kernel module is loaded you will be able to see espsta0
> 
> **Question 3: both the master and the slave use mode 2**
> 
> This is the default configuration and works correctly hence it might not be a SPI problem.
> 
> Additional information: Currently we use DuPont cable to connect ESP32C3 and the main MCU, and the cable length is about 10CM.
> There is no suitable oscilloscope at the current office location that can capture waveform analysis
> 
> You can verify your spi connection by following same steps for esp_hosted_ng https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md
> 
> 
> Regards,
> Shreyas Ashish Sheth

linchanghe123, 2023-09-13:
Hello @mantriyogesh

Regarding question 2, after loading the esp32c3 firmware and esp32_spi.ko under Linux, Linux did not receive the event that generated espsta0. I grabbed the waveform and found that there was no valid data. My wiring has been confirmed to be error-free. Could you please help me check the waveform information? Give me some guidance.
[image]
[image]

Regards,
Lin

mantriyogesh, 2023-09-13:
Can you please share full waveform to check in detail?

linchanghe123, 2023-09-13:
[untitled.csv](https://github.com/espressif/esp-hosted/files/12595107/untitled.csv)

Can you open this file？
[image]
[image]
[image]
[image]

linchanghe123, 2023-09-13:
I enabled spidev of Linux and shorted MISO and MOSI. It is normal to use the spidev_test command to test the data sending and receiving functions.
[image]

mantriyogesh, 2023-09-13:
Cannot analyze csv, can you please send raw format and software to be used?

mantriyogesh, 2023-09-13:
From the images, I see the clock is not 50% duty cycle.
Not sure, but looks like it is not 50%.

ESP may not perform correctly if the clock is not released 50% duty.
can you please attach esp full log and dmesg full log (all textual logs please).

mantriyogesh, 2023-09-13:
in esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c

Please after spi_slave_initialize(),
please add 
```c
    gpio_set_drive_capability(GPIO_SCLK, GPIO_DRIVE_CAP_3);
    gpio_set_drive_capability(GPIO_MISO, GPIO_DRIVE_CAP_3);
    gpio_set_pull_mode(GPIO_MISO, GPIO_PULLDOWN_ONLY);
```
Sets GPIO strength to highest.

Below code may not exactly match, just to show the idea of adding pull-ups/pull-downs & gpio drive capability.

```c
    /* Enable pull-ups on SPI lines
     * so that no rogue pulses when no master is connected
     */
    gpio_set_pull_mode(CONFIG_ESP_SPI_GPIO_HANDSHAKE, GPIO_PULLDOWN_ONLY);
    gpio_set_pull_mode(CONFIG_ESP_SPI_GPIO_DATA_READY, GPIO_PULLDOWN_ONLY);
    gpio_set_pull_mode(GPIO_MOSI, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_SCLK, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_CS, GPIO_PULLUP_ONLY);

    ESP_LOGI(TAG, "SPI Ctrl:%u mode: %u, InitFreq: 10MHz, ReqFreq: %uMHz\nGPIOs: MOSI: %u, MISO: %u, CS: %u, CLK: %u HS: %u DR: %u\n",
            ESP_SPI_CONTROLLER, slvcfg.mode, SPI_CLK_MHZ,
            GPIO_MOSI, GPIO_MISO, GPIO_CS, GPIO_SCLK,
            CONFIG_ESP_SPI_GPIO_HANDSHAKE, CONFIG_ESP_SPI_GPIO_DATA_READY);

    ESP_LOGI(TAG, "Hosted SPI queue size: Tx:%u Rx:%u", SPI_TX_QUEUE_SIZE, SPI_RX_QUEUE_SIZE);
    register_hs_disable_pin(GPIO_CS);

    /* Initialize SPI slave interface */
    ret=spi_slave_initialize(ESP_SPI_CONTROLLER, &buscfg, &slvcfg, DMA_CHAN);
    assert(ret==ESP_OK);

    //gpio_set_drive_capability(CONFIG_ESP_SPI_GPIO_HANDSHAKE, GPIO_DRIVE_CAP_3);
    //gpio_set_drive_capability(CONFIG_ESP_SPI_GPIO_DATA_READY, GPIO_DRIVE_CAP_3);
    gpio_set_drive_capability(GPIO_SCLK, GPIO_DRIVE_CAP_3);
    gpio_set_drive_capability(GPIO_MISO, GPIO_DRIVE_CAP_3);
    gpio_set_pull_mode(GPIO_MISO, GPIO_PULLDOWN_ONLY);
```

Also with referring, https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md
Verify that all the GPIOs are working fine. 
check `SPI Timing issues - Tuning` section as well.

Also add debug logs in `spi_data_ready_interrupt_handler()` and `spi_interrupt_handler()` at linux and get us the log.

linchanghe123, 2023-09-13:
Hello mantriyogesh，
Can you help me provide a firmware for esp32c3? I can't compile it here

linchanghe123, 2023-09-13:
[Uploading Logic Setup 1.1.15 (64-bit).zip…]()
[spi.zip](https://github.com/espressif/esp-hosted/files/12595453/spi.zip)

[image]

mantriyogesh, 2023-09-13:
From your log, I could see there :
1. Clock released looks very wrong in duty cycle.
[image]
2. there was no bootup event sent from ESP to host on MISO.
Please check if MISO pin (or other SPI pins) are correctly connected to master.
MISO must have some input on startup/reset.
It may be possible that host is not correctly sampling because of wrong duty cycle, But for that first MISO at least have to be sent.

unless first event is seen on logic, we should not move ahead.
@Shreyas0-7 , Can you please help @linchanghe123  to build on master ?
@linchanghe123 , please make sure that you use **exactly same git commit at esp and host side**.
If code is different at esp and host, it may create issues.

So, next steps:
1. Check why host is sending such a **unstable duty cycle.**
- Possible cause could be incorrect input power supplied on host
- Disable power saving at host (at least for the time being)
- Wires could be shorter and with equal sizes
- Lower the clock further low, so that the duty cycle is stable. 
2. **Build** the ESP NG firmware on your own. you can use master git commit at both ESP and host for building. @Shreyas0-7 can help in resolving the build issues. Code is open source so you should be able to tweak the code easily to make sure that base platform is up and running. Sharing Firmware binaries is dead slow even for small change and not suggested.
3. you might also want to comment two lines (while debugging):
https://github.com/espressif/esp-hosted/blob/43267cfee077276907518f3eba53fda93304f736/esp_hosted_ng/host/spi/esp_spi.c#L205-L206
to avoid two step frequency adjustment from slave to host.
but this will anyway be in effect once the bootup event is received.
4. You can also **test SPI base 4 lines & handshake using esp-idf example**, https://github.com/espressif/esp-idf/tree/master/examples/peripherals/spi_slave with one part running on **ESP** chipset and other part on **host** (even user space library which provides spi is okay for now)

shreyassheth1, 2023-09-13:
Hey @linchanghe123 
For esp3c3 firmware compilation errors, please mention which commit you are using.
Can you please make sure you have followed these steps correctly https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/README.md
Most importantly please verify you are using idf tools using export.sh from this path only esp-hosted/master/esp_hosted_ng/esp/esp_driver/esp-idf in a fresh terminal and then try to build the firmware.
Also can you please attach the compilation failed logs.

linchanghe123, 2023-09-13:
Hello @shreyassheth1
I download esp-hosted code and run "cmake ."，The following error occurs：
[image]
[image]

shreyassheth1, 2023-09-15:
Hey can you please folllow these steps in esp_hosted_ng\esp\esp_driver folder:
1) git reset --hard
2) git submodule update --init --depth=1

Go in esp-idf directory 
3) git submodule update --init --depth=1

(make sure this is a fresh terminal and you have installed python)
In esp-idf directory
4) Enter command install.bat 

In esp_driver(cd ..) folder use these command to setup firmware libraries

5) rmdir /s /q esp-idf\components\esp_wifi\lib
6) xcopy /s /e lib\ esp-idf\components\esp_wifi\lib\

After completing these steps you will be able to build the firmware.

linchanghe123, 2023-09-19:
hello，
I use the idf v4.4.5 version and comment out the wifi-related functions that cannot be defined. Now the spi communication is normal. However, when I overwrite the relevant lib libraries of esp_wifi and release the commented out wifi-related functions, it will report that other functions cannot be found (esp_wifi_sta_get_use_h2e_internal, etc., defined in the original libnet80211.a). How should this be handled?
[image]

shreyassheth1, 2023-09-19:
Hey @linchanghe123 Please use **Idf tools present in esp_idf in esp_driver folder.**
Please follow the steps mentioned in the previous comment.
Also, these commands (as mentioned above) in esp_driver folder will take care of errors occurring in libnet80211.a
`
    rmdir /s /q esp-idf\components\esp_wifi\lib
`
`
    xcopy /s /e lib\ esp-idf\components\esp_wifi\lib\
`

linchanghe123, 2023-09-21:
The code downloaded through commands cannot be downloaded completely, nor can it be compiled successfully. After I installed the V4.4.5 version, I replaced the “component” with the ”component“under the esp_driver/esp-idf that was not downloaded well, and then replaced the lib library under esp_wifi. Finally, the code compiles normally. Now esp-hosted mode works normally. Thanks
[image]

shreyassheth1, 2023-09-25:
Hey glad to here it's working for you but I will recommend you to follow these steps in order to make it work as expected 

git clone https://gitlab.espressif.cn:6688/app-frameworks/esp_hosted.git

Now everything will get cloned under esp_hosted folder.
In esp_driver directory 
`cd esp_hosted\esp_hosted_ng\esp\esp_driver`

perform these steps 
`git reset --hard`
`git submodule update --init --depth=1`

This will create esp-idf directory under esp_driver folder
**in esp_driver folder**
`cd esp-idf`
` git submodule update --init --depth=1`

(make sure this is a fresh terminal and you have installed python)

`install.bat` 

These steps will installs all the tools required to build firmware.

To have required libraries for building firmware follow these steps
In  **esp_driver(cd ..)** folder use these commands.`

`rmdir /s /q esp-idf\components\esp_wifi\lib`
` xcopy /s /e lib\ esp-idf\components\esp_wifi\lib\`

kapilkedawat, 2023-10-06:
@linchanghe123 feel free to close the issue if this is resolved.

closed 2023-10-08 by linchanghe123