espressif/esp-hosted #263: ESP_hosted-NG: ESP32 panic'ed because of illegal instruction executed in libbtdm_app.a

franziskanerda, 2023-09-18:
Dear,

We found out an issue that is being triggered sporadically when we have WiFi and BT communication ongoing simultaneously for a longer time.

This is a tail of a log file obtained from ESP32 side:

I (152094) FW_MAIN: Set IP Address
I (152136) FW_MAIN: Set multicast mac address list
I (555278) FW_MAIN: Add key request
I (555280) FW_CMD: process_add_key:1761
I (555280) FW_CMD: Setting GTK [1]
I (555282) FW_CMD: process_add_key:1822 auth done
I (1035296) FW_MAIN: Add key request
I (1035296) FW_CMD: process_add_key:1761
I (1035296) FW_CMD: Setting GTK [2]
I (1035298) FW_CMD: process_add_key:1822 auth done
I (1515326) FW_MAIN: Add key request
I (1515326) FW_CMD: process_add_key:1761
I (1515326) FW_CMD: Setting GTK [1]
I (1515328) FW_CMD: process_add_key:1822 auth done
I (1995340) FW_MAIN: Add key request
I (1995340) FW_CMD: process_add_key:1761
I (1995340) FW_CMD: Setting GTK [2]
I (1995342) FW_CMD: process_add_key:1822 auth done
I (2475348) FW_MAIN: Add key request
I (2475350) FW_CMD: process_add_key:1761
I (2475350) FW_CMD: Setting GTK [1]
I (2475352) FW_CMD: process_add_key:1822 auth done

ASSERT_PARAM(1024 0), in rwbt.c at line 381
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008f8d0: f01d020c 00004136 f01d0000

Core  0 register dump:
PC      : 0x4008f8d7  PS      : 0x00060b34  A0      : 0x80083e09  A1      : 0x3ffc1a68
A2      : 0x00000000  A3      : 0x00000400  A4      : 0x00000000  A5      : 0x3f4097a8
A6      : 0x0000017d  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc19d8
A10     : 0x00000000  A11     : 0x3ffc19fb  A12     : 0x3ffc19a7  A13     : 0x00000031
A14     : 0x00000000  A15     : 0x3ffc19ac  SAR     : 0x00000004  EXCCAUSE: 0x00000000

EXCVADDR: 0x00000000  LBEG    : 0x40083d11  LEND    : 0x40083d19  LCOUNT  : 0x00000000

 
Backtrace:0x4008f8d4:0x3ffc1a68 |<-CORRUPTED

 
ELF file SHA256: b03e6aae4654bd96

 
Rebooting...

 
Further investigation pointed out that it might be a known bug in libbtdm_app.a already resolved with this commit:
https://github.com/espressif/esp-idf/commit/a83b16f18d78c29a99a3cd57f223f9eaa2200055

Could you please provide an update of the current FW sources that includes newer library with the changes that resolve this issue?

mantriyogesh, 2023-09-19:
@kapilkedawat ++

franziskanerda, 2023-10-06:
Thanks for your message and for the information, that v5.0 should resolve our problem. Have you already scheduled v5.0 and can you say something about the estimated release date?

kapilkedawat, 2023-10-10:
@franziskanerda, it should be available by the end of this month.

franziskanerda, 2023-11-09:
Hi kapilkedawat, do you have any news on the release schedule for v5.0 of the 'hosted_ng' firmware? Would be great if we could test in this year if the new version brings improvements to our problem.

franziskanerda, 2023-11-16:
@kapilkedawat : sorry, I forgot the @ letter -:). Do you have any news on the release schedule for v5.0 of the 'hosted_ng' firmware? I would be very grateful for a short answer!

kapilkedawat, 2023-11-16:
Hey @franziskanerda.. it's almost ready and currently in review phase, we will be releasing it shortly. Do note that it will based on IDF 5.1 public release.