espressif/esp-hosted #264: Compilation issue

gauthamik9, 2023-09-29:
Hi yogesh,
I tried with raspberry pi -- esp32 setup, wifi is connected.With same configuration , I am trying with arm processor as host and esp32 as slave.
I compiled in kernel level esp32 folder it was loaded and I got the nodes for both station and ap. I didnt add the c_support folder so I could not able to start the sudo ./test.out softap_start process.
So I want to add this folder ,while doing  cross compiling I could not found the cmsis_os.h file. Could you please tell the exact path.
gauthami@gauthami-Latitude-3420:~/esp_timing/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support$ . /opt/poky-tiny/2.5.2/environment-setup-cortexa5hf-neon-poky-linux-musleabi 
gauthami@gauthami-Latitude-3420:~/esp_timing/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support$ ${CC} test.c -o test
In file included from serial_if.h:12:0,
                 from test.c:23:
platform_wrapper.h:8:10: fatal error: cmsis_os.h: No such file or directory
 #include "cmsis_os.h"
          ^~~~~~~~~~~~

mantriyogesh, 2023-09-29:
From 
```file
 I  ~/c/h5   $  esp_hosted_fg/host/linux/host_control/c_support  cat Makefile                                                                                                                                                                     Fri Sep 29 14:22:24 2023
CC = gcc

CROSS_COMPILE :=

CFLAGS = -C -Wall --sysroot=$(SDKTARGETSYSROOT)

CFLAGS_SANITIZE = -fsanitize=undefined  -fsanitize-address-use-after-scope -fsanitize-undefined-trap-on-error  -fstack-protector-all -fstack-check -fsanitize=address -fsanitize=pointer-compare -fno-omit-frame-pointer -static-libasan

LINKER = -lpthread -lrt

DIR_COMMON = $(PWD)/../../../../common
DIR_CTRL_LIB = $(PWD)/../../../control_lib
DIR_SERIAL = $(PWD)/../../../virtual_serial_if
DIR_COMPONENTS = $(PWD)/../../../components
DIR_LINUX_PORT = $(PWD)/../../port

INCLUDE += -I$(DIR_COMMON)/protobuf-c
INCLUDE += -I$(DIR_COMMON)/include
INCLUDE += -I$(DIR_CTRL_LIB)/include
INCLUDE += -I$(DIR_CTRL_LIB)/src/include
INCLUDE += -I$(DIR_SERIAL)/include
INCLUDE += -I$(DIR_COMPONENTS)/include
INCLUDE += -I$(DIR_LINUX_PORT)/include
INCLUDE += -I.

SRC += $(DIR_COMMON)/protobuf-c/protobuf-c/protobuf-c.c
SRC += $(DIR_COMMON)/esp_hosted_config.pb-c.c
SRC += $(DIR_CTRL_LIB)/src/ctrl_core.c
SRC += $(DIR_CTRL_LIB)/src/ctrl_api.c
SRC += $(DIR_SERIAL)/src/serial_if.c
SRC += $(DIR_COMPONENTS)/src/esp_queue.c
SRC += $(DIR_LINUX_PORT)/src/platform_wrapper.c

PWD := $(shell pwd)

SRC += ./test_utils.c

all: test

test:
	$(CROSS_COMPILE)$(CC) $(CFLAGS) $(INCLUDE) $(SRC) $(LINKER) $(@).c -o $(@).out

stress:
	$(CROSS_COMPILE)$(CC) $(CFLAGS) $(CFLAGS_SANITIZE) $(INCLUDE) $(SRC) $(LINKER) $(@).c -o $(@).out -ggdb3 -g

clean:
	rm -f *.out *.o
```

gauthamik9, 2023-10-05:
Dear Yogesh,
When I try to cross compile with this toolchain CROSS_COMPILE := arm-poky-linux-musleabi-
getting some errors.
root@0d4c4bb56531:/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support# . /opt/poky-tiny/2.5.2/environment-setup-cortexa5hf-neon-poky-linux-musleabi 
root@0d4c4bb56531:/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support# make test
arm-poky-linux-musleabi-gcc  -O2 -pipe -g -feliminate-unused-debug-types  -I/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/protobuf-c -I/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/include -I/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/include -I/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/src/include -I/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../virtual_serial_if/include -I/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../components/include -I/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../port/include -I. /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/protobuf-c/protobuf-c/protobuf-c.c /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/esp_hosted_config.pb-c.c /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/src/ctrl_core.c /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/src/ctrl_api.c /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../virtual_serial_if/src/serial_if.c /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../components/src/esp_queue.c /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../port/src/platform_wrapper.c ./test_utils.c -lpthread -lrt test.c -o test.out
/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/protobuf-c/protobuf-c/protobuf-c.c:48:10: fatal error: stdlib.h: No such file or directory
 #include <stdlib.h> /* for malloc, free */
          ^~~~~~~~~~
compilation terminated.
In file included from /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/include/esp_hosted_config.pb-c.h:7:0,
                 from /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/esp_hosted_config.pb-c.c:9:
/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/protobuf-c/protobuf-c/protobuf-c.h:199:10: fatal error: assert.h: No such file or directory
 #include <assert.h>
          ^~~~~~~~~~
compilation terminated.
/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/src/ctrl_core.c:4:10: fatal error: stdlib.h: No such file or directory
 #include <stdlib.h>
          ^~~~~~~~~~
compilation terminated.
In file included from /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/include/esp_hosted_config.pb-c.h:7:0,
                 from /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/include/ctrl_api.h:13,
                 from /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/src/ctrl_api.c:7:
/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/protobuf-c/protobuf-c/protobuf-c.h:199:10: fatal error: assert.h: No such file or directory
 #include <assert.h>
          ^~~~~~~~~~
compilation terminated.
/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../virtual_serial_if/src/serial_if.c:5:10: fatal error: string.h: No such file or directory
 #include <string.h>
          ^~~~~~~~~~
compilation terminated.
In file included from /home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../../components/src/esp_queue.c:8:0:
/opt/poky-tiny/2.5.2/sysroots/x86_64-pokysdk-linux/usr/lib/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/include-fixed/stdio.h:22:10: fatal error: features.h: No such file or directory
 #include <features.h>
          ^~~~~~~~~~~~
compilation terminated.
/home/wifi/esp_timing_fg/esp-hosted/esp_hosted_fg/host/linux/host_control/c_support/../../port/src/platform_wrapper.c:8:10: fatal error: stdlib.h: No such file or directory
 #include <stdlib.h>
          ^~~~~~~~~~
compilation terminated.
In file included from ./test_utils.c:22:0:
/opt/poky-tiny/2.5.2/sysroots/x86_64-pokysdk-linux/usr/lib/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/7.3.0/include-fixed/stdio.h:22:10: fatal error: features.h: No such file or directory
 #include <features.h>
          ^~~~~~~~~~~~
compilation terminated.
In file included from test.c:22:0:
test.h:24:10: fatal error: sys/types.h: No such file or directory
 #include <sys/types.h>
          ^~~~~~~~~~~~~
compilation terminated.
Makefile:44: recipe for target 'test' failed
make: *** [test] Error 1
can you suggest any solution for this.

mantriyogesh, 2023-10-05:
Are you able to cross compile simple demo.c

demo.c:
```c
#include <stdlib.h>
void main()
{
printf("running cross compiled on target\n");
}
```

Before moving to demo app, please cross compile this (or any sample C file) and run it on target.
Once you are able to do it, youncan use same command to hook in Makefile of demo app.

gauthamik9, 2023-10-05:
Dear Yogesh,
Yes It is running in the board.
root@rugged-board-a5d2x-sd1:/data# ./test
running cross compiled on target

mantriyogesh, 2023-10-05:
Okay. Use same config such that the build (command) output you get from app demo matches to that of demo.c above.

Make sure no extra flags added while building and compiler is visible.
With comparison, you can make it out.

You might have to mend the cross_compile and other options such that the final build command (output from make) resembles.

gauthamik9, 2023-10-06:
Dear Yogesh,
I tried to cross compile the application.when I try to set sta_connect getting error.
root@rugged-board-a5d2x-sd1:~# modprobe esp32_sdio.ko
esp_probe: ESP network device detected
esp_sdio: probe of mmc0:0001:2 failed with error -22
root@rugged-board-a5d2x-sd1:~# 
Received INIT event from ESP32 peripheralEVENT: 2
ESP slave Chip ID: 0x0
EVENT: 0
ESP peripheral capabilities: 0x1d
Features supported are:
         * WLAN
         * BT/BLE
           - HCI over SDIO
           - BT/BLE dual mode
EVENT: 3

root@rugged-board-a5d2x-sd1:~# ifocnfig -a
-sh: ifocnfig: command not found
root@rugged-board-a5d2x-sd1:~# ifconfig -a
can0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          NOARP  MTU:16  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:10 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:48 

eth0      Link encap:Ethernet  HWaddr 04:91:62:D3:1D:AA  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:28 Base address:0x8000 

ethap0    Link encap:Ethernet  HWaddr 00:00:00:00:00:00  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ethsta0   Link encap:Ethernet  HWaddr 00:00:00:00:00:00  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          LOOPBACK  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4  
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

root@rugged-board-a5d2x-sd1:~# cd /data/
root@rugged-board-a5d2x-sd1:/data# ls
test.out
root@rugged-board-a5d2x-sd1:/data# sudo ./test.out get_wifi_mode
-sh: sudo: command not found
root@rugged-board-a5d2x-sd1:/data# ./test.out get_wifi_mode
serial interface open failed, Is the driver loaded?
Failed to serial_init
init hosted control lib failed
Err Exit

Suggestion please.

mantriyogesh, 2023-10-06:
expected behaviour.

check https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#22-rpi_initsh

You need to port rpi_init.sh, not bypass it.
There are other things in rpi_init.sh, which you are simply skipping, results in above error.

closed 2023-11-28 by mantriyogesh