espressif/esp-hosted #265: Long Range Protocol

sueng66, 2023-09-30:
I want to use Long Range Protocol in here, is it possible ? Try to look for esp_wifi_set_protocol only found libnet8211.a

mantriyogesh, 2023-10-06:
Can you please let me know which flavour (ESP-Hosted-NG or ESP-Hosted-FG)?

sueng66, 2023-10-06:
I will using Linux as the host so I prefer ESP-Hosted-NG

mantriyogesh, 2023-10-06:
@kapilkedawat ++

kapilkedawat, 2023-10-10:
@sueng66 Long range support is not available at the moment since its an espressif proprietary protocol and doesn't fit into hosted-ng's open source based implementation. Also it needs both AP and station to be ESP devices to work.

mantriyogesh, 2023-10-11:
Hello @sueng66 ,

ESP-Hosted-FG also supports Linux. So ESP-Hosted-FG can be used for implementation of Long range.
You might have to try it on directly using ESP firmware change first. Once PoC is okay, you can also set the protocol from Linux using RPC (control path)

sueng66, 2023-10-11:
> Hi @sueng66, Long-range support is currently unavailable because it is an Espressif proprietary protocol and does not fir into hosted-ng's open-source-based implementation. Additionally, it requires both the Access Point (AP) and the station to be ESP devices for it to function.

Sure I will using one esp-hosted as AP and another several esp-hosted as station

kapilkedawat, 2023-10-12:
@sueng66 suggest to use esp-hosted-fg solution since that supports both AP and station mode. As @mantriyogesh explained, you can set the mode directly in firmware first and if initially PoC is okay, we will add the support to set the mode using RPC.

sueng66, 2023-11-09:
done by inserting esp_wifi_set_protocol at file slave_control.c , in function req_start_softap_handler and req_connect_ap_handler before calling ret = esp_wifi_get_mac(ESP_IF_WIFI_STA , mac);

closed 2023-11-09 by sueng66