espressif/esp-hosted #266: esp-hosted FG on ESP32-C2

Albert702, 2023-10-02:
Hello. 
Because NG support for ESP32-C2 not ready while, i trying now to use FG for my linux system.

ESP IDF 5.1 , esp-hosted-release-ng-v1.0.2 , only changed Data Ready and Handshake gpios according  to my hardware,- ESP32 chip on device pcb.

I found some issues.

1 When i build firmware from "default" configuration,  a connection to  my AP with test host utility is failed.
errors are:
wifi:bcn_timeout,ap_probe_send_start
wifi:ap_probe_send over, resett wifi status to disassoc
assert failed: queue_next_transaction spi_slave_api.c:464 (spi_trans->rx_buffer)
See "adapter+bt.log". 

Seems like  insufficient memory. 

I switched-off bluetooth in menuconfig. In this firmware remain only error 'bcn_timeout'. See "adapter_only.log".
Network is operational, but big delays and errors at network commands (ping) .See "ping_host.log".
If do not use connection some time (several minuts), connection is break. See "ping1.log"

A "station" example from SDK is connected to my AP and worked without problems in same "environment". See "station.log"
[log.zip](https://github.com/espressif/esp-hosted/files/12779269/log.zip)

mantriyogesh, 2023-10-02:
Hello @Albert702 ,

Which git commit do you refer to for FG at esp and host?

mantriyogesh, 2023-10-02:
Also,

1. What is host
2. Can you check the transport speed first using raw throughput testing of TX and RX?
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md

mantriyogesh, 2023-10-03:
### assert failed: queue_next_transaction spi_slave_api.c:464 (spi_trans->rx_buffer)

This means you are lacking memory as ESP32-C2 has lower memory capacity, issue is observed.
Can you please try **master** branch (always use same commit at both ESP and host) ?
We had tried to limit the memory usage https://github.com/espressif/esp-hosted/blob/4730f9dda379ec0199c1e83b8316ecfe3c88aa1f/esp_hosted_fg/esp/esp_driver/network_adapter/sdkconfig.defaults.esp32c2#L30-L31

### ESP IDF 5.1 , esp-hosted-release-ng-v1.0.2 , only changed Data Ready and Handshake gpios according to my hardware,- ESP32 chip on device pcb.
Actually, In `esp-hosted-release-ng-v1.0.2` , as name suggests, it is NG targetted release. 
```
For FG, you can try the master code for now.
There are multiple issue fixes for FG recently added into master branch.
```
### If firmware with RAW_TP option, host is never successfully obtained ip address.

Raw_TP is transport testing, where focus is on running traffic over transport without any Wi-Fi connection or any other software involvement. This would help to understand the transport layer problem.
It is not expected to get IP address in this test.

### measure of throughput is impossible

It is not iperf throughput. it is just transport level throughout check where a dummy packet is sent in the ineterested direction (either `Host->ESP` or `ESP->Host`).

### I do not know how to work
esp debug output, polling or interrupt-based.

Lets not jump on the conclusions, first lets understand where is the problem.

### When network is operational, file - copy - through - ftp transfer rate about 150...170 KBytes/s.

So involving complete system and testing would be difficult to localise the problematic leg.
that is the reason, try to run raw throughput first.

### When disabled all debug output in menuconfig, a speed is decreased by ~ 30 times. I think, this is sync issue.

Possibly looks like some issue in SPI. Have you had a look at https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md ? As you already able to communicate (ping), the porting is done fine. but still some issues might exist, which might be just ignored or not known to you, which need to be taken care. for example, length of jumper wires or max spi slave frequency support etc.

Albert702, 2023-10-04:
Hello, @mantriyogesh

>1.6 Mbps is extremely low

Peak speed is ~1.6 megabytes per second, not megabits.
 
>Can you please check the logic analyzer waveform?

In attach waveform of SPI clock signal. Scales : 0.5 ms and 5 uS.

Try to build FG host driver from master branch is failed, because :
...........................................
esp_kernel_port.h:100:8: error: too many arguments to function ‘ether_addr_copy’
  100 |        ether_addr_copy(dev->dev_addr, addr, ETH_ALEN);
      |        ^~~~~~~~~~~~~~~
In file included from drivers/net/wireless/espressif/main.c:26:
./include/linux/etherdevice.h:276:20: note: declared here
  276 | static inline void ether_addr_copy(u8 *dst, const u8 *src)
.............................................

Kernel 5.4.66
Need to add new 'issue'?

[waveform.zip](https://github.com/espressif/esp-hosted/files/12802472/waveform.zip)

mantriyogesh, 2023-10-05:
Ohh, I mistaken with mega bits ps.

The throughput is correct (11+ mbps TCP) and is consistent with our tests on Linux.

> 100 | ether_addr_copy(dev->dev_addr, addr, ETH_ALEN);

Actually, this code is changed and under review taking little time to get pushed over master.
You can change it as 

```diff
-- ether_addr_copy(dev->dev_addr, addr, ETH_ALEN);
++ ether_addr_copy(dev->dev_addr, addr);
```

The expected change for now I attached which takes care of Linux 5/6:
[MR307.patch](https://github.com/espressif/esp-hosted/files/12812105/MR307.patch)
might take up to week to get reflected on master.

mantriyogesh, 2023-10-05:
@Albert702 
Above change is already added on GitHub master branch. PTAL.

Please help in closing the issue.

Albert702, 2023-10-05:
Hello, @mantriyogesh.

Ok, thanks.

closed 2023-10-05 by Albert702