espressif/esp-hosted #267: What is the situation will let esp send the ESP_PRIV_IF cmd second times  ?

LuciaferKelvin, 2023-10-18:
I found that case when the esp connected the ap while using sta mode.
I need to know the situation let the esp send the ESP_PRIV_IF cmd .
As I konw , that cmd will be send first time after esp power on.
Bye the way , I firmware version had I used is Relases V0.4.

mantriyogesh, 2023-10-18:
Hello @LuciaferKelvin 

Yes
https://github.com/espressif/esp-hosted/blob/74e707b604280ecb2eafa60ba5ed6993ce57b457/host/linux/host_driver/esp32/main.c#L346-L348

In 0.4, ESP_PRIV_EVENT_INIT message on top of ESP_PRIV_IF means that the ESP has been freshly started.
If you receive such event in middle of scenario, would mean that ESP firmware had aborted or asserted.

ESP logs can be taken and confirmed on this.

By the way, Priv interface is extendable for any user use. Own message type similar to  ESP_PRIV_EVENT_INIT can be created, to handle any custome message.

LuciaferKelvin, 2023-10-18:
sure , that ist it. 
I had test it .The esp is reboot.
Thank you

closed 2023-10-18 by LuciaferKelvin