espressif/esp-hosted #268: About ESP Tx power

LuciaferKelvin, 2023-10-20:
Is the esp hosted firmware automatic  change the tx power in working?

mantriyogesh, 2023-10-20:
NG also supports power tuning from host from wpa_supplicant/cfg80211:

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/esp_cmd.c#L249-L253

LuciaferKelvin, 2023-10-20:
But the host just could set set max tx power， if I set the max tx power， the esp will using that power level？

LuciaferKelvin, 2023-10-20:
for example ，If I set the 15dbm max Tx power,  Will the esp rearch the tx power level?

kapilkedawat, 2023-10-20:
> for example ，If I set the 15dbm max Tx power, Will the esp actual tx power reach it value and keep it?

@LuciaferKelvin it will the maximum value of transmit power that will be used during transmission subjected to regulatory restrictions.

mantriyogesh, 2023-11-28:
can we close this?

closed 2023-11-28 by LuciaferKelvin