espressif/esp-hosted #269: 请问ESP-Hosted-NG什么时候能够支持AP

zhengqianguuu, 2023-10-24:
ESP-Hosted-NG版本从去年开始就提示AP模式会尽快发布，一直在等这个功能，请问什么时候能够支持，或是否有支持SDIO的AP和STA

mantriyogesh, 2023-10-24:
Hello @zhengqianguuu ,

yes, It is still in development and might take some time to get released.
You can use FG version for AP, STA, and AP+STA cases for time being over SDIO.

zhengqianguuu, 2023-10-26:
我测试过您说的FG版本，但是在I.MU6ULL的开发板上没有驱动起来，该版本是在树莓派上面运行的吗？所以才测试了NG的版本，STA可以使用。

该邮件从移动设备发送

mantriyogesh, 2023-10-26:
### Common Transport for FG and NG
The underlying transport is the same for FG and NG.
Meaning, if you are not able to run FG, then you wouldn't be able to run NG either.

### Porting ESP-Hosted to your Linux
To port the ESP-Hosted solution to your Linux, please check [porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md)

If you face any problems, let us know.
Porting is user's responsibility and would not be considered ESP-Hosted bug, but we can help as much as possible to get your platform up for ESP-Hosted.

### SDIO limitations
If you use SDIO as transport, please switch to PCB with external pull-ups.
Using jumper wire in SDIO make signal integrity issues.
This is a protocol limitation and not ESP-Hosted as such.

### Transport preference
You can try using SPI solution if using small jumper wires, at first.
If you need SDIO with jumper cables only, you can also try 1 bit SDIO mode first, which faces less signal integrity probs.

### 1 Bit SDIO Raspberry Pi config
Raspberry Pi config for 1 bit:
https://github.com/espressif/esp-hosted/issues/242#issuecomment-1713662854
You would need to change your DeviceTree accordingly for your Linux (this is anyway needed irrespective of FG/NG).
Once tested, you can form integrated pcb to test 4 bit mode.