espressif/esp-hosted #270: Ported to STM32H7B with lwip - but there are issues

claydonkey, 2023-10-31:
Good day sir and may I say, you are doing great work...

I have migrated the host code to work on a STM32H7B0 (Single core) and had good results running a UDP server and IPerf. My hope was to setup a UDP Server capable of sustaining (1MBytes ~ 10Mbits) /sec - that's all I require (I managed to get 0,6 Mbytes with SPI and a homebrew version of WIFININA  for the same chip. However I had to ditch that as it  didn't quite reach the desired target bandwidth - due mainly to the fact that the Slave SPI on the ESP32 is limited to 10Mhz.

So I have the following queries/problems.

1.  running iperf with the SDIO at 20Mhz I am getting a max bench of 6Mbits/s and running the clock any faster yields a similar result. So I decided to TASK_RAW_TP, but currently as soon as the host scans the network APs it bombs. The esp32 is pushing packets at 10Mbits/s and the SDMMC_FLAG_DTIMEOUT interrupt is triggered on the STM. - So I "IDMACTRL = SDMMC_DISABLE_IDMA"ed it  , the sdio_api.c signals "sdio_driver_read_bytes 63 CMD53 error" leaving the console to repeat
xxx-xxx sec 0.00000 Kbits/sec.
xxx-xxx sec 0.00000 Kbits/sec.
....

How can I :

a. Lower the stream output from the ESP32 ) - I am not at all accustomed with the esp32 slave code...
b. Ditch the extra packets on the host and carry on until I find the max bandwidth of the STM32..

2. I have scoured the [Getting started with STM32H7 Series SDMMC host controller]
(https://www.st.com/resource/en/application_note/an5200-getting-started-with-stm32h7-series-sdmmc-host-controller-stmicroelectronics.pdf) and have implemented its recommendations as best I could it in "sdio_ll.c" Also I have referred to the STM32 code from  edp-at example/peripherlas/SDIO adding :
if (HAL_GPIO_ReadPin(GPIOC, GPIO_PIN_9) == RESET) {    // D1 if low, means SDIO interrupt
		xSemaphoreGive(sdio_recv_SemHandle);
	}
and the appropriate semaphore stuff

The result - consistent but slow. 

3. The hardware
I have really small 1-2cm wires soldered to the across the pins of each MCU with all the right pull ups. 
On the ESP32 : 
Turned off the logging (LED lights), QDIO 80Mhz, Compiled Ofast. 
On the STM32:
Using DCache and ICache  (with SCB_CleanInvalidateDCache())  - I wanted to place the buffers in Non Cached memory but it a the buffers are created on the heap (I think) and  placing them in partitioned memory is kinda tedious. Compiled with -Ofast. SDIO doesn't complain, IPerf gives results upto 100Mhz. 

4. I briefly looked at your other more upto date/experimental branch of esp-hosted. Have you ditched the STM32s for good. If so  a. Would it be easy to port the Host code to other MCUs?
b. Would it ill advised to try that at the moment?

I'm sorry if this all a bit incoherent I have been up all night trying to finish this project ...
Thanks for any advice

mantriyogesh, 2023-10-31:
Hello @claydonkey ,

We have another branch , https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host where we use 
ESP (slave) - SPI/SDIO - ESP (host)
where we have:
- Improved our code (bugfixes and features)
- Better iperf throughput
- Clear port layer, which lets easy adaptation for any MCUs.

#### Limitations with STM32:
1. SDIO: As STM32 doesn't really provide pure SDIO (AFAIK when we implemented), we had to take make SDIO HAL layer initialization manually.
2. HAL layer thus copied is really not maintainable, as it varies from STM32 to STM32 in the customer pick.
3. Users have to really migrate from one STM32 HAL driver to desired STM32 HAL driver code (as we had to import sdio_ll as HAL port code), which sometimes become tricky to reproduce due to hardware dependencies.

This was the reason we implement ESP32 itself being Host.
Additionally, Using ESP as host helps to use full potential feature set. 
 
Although we support 'SDIO' yet with specific STM32 till now, STM32F412ZGT6, I would really suggest to use evaluate https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host
with SPI as transport initially.

SDIO (ESP host-SDIO-ESP slave) transport is also supported but broken on latest changes, which we expect to fix in upcoming week.

Do let me know if you wish to try this. I will explain (probably create small document to use this branch for evaluation) the port layer, pin connections etc.

claydonkey, 2023-10-31:
I am really impressed by your work and thanks for the quick reply.... I noticed ALOT of esp headers in the new host is most of the code platform agnostic?

claydonkey, 2023-10-31:
Just a quick ask. How can I reduce the packets/speed of the Raw Test?

claydonkey, 2023-10-31:
**fantastic!** @mantriyogesh 

I am pretty satisfied with the UDP packet bandwidth so far.  However, it remains to be seen how much the combination of the lwip stack and esp-hosted slave cap the performance of my main app. I really need to keep it svelte as I am only using UDP to transmit what used to be sent over cdc virtual serial. Still I am approaching USB fs speeds here...

I just got hold of a ESP32-C6. Is it's SPI still capped at 10Mhz?

Also I noticed that there is a the **DEFAULT_LISTEN_INTERVAL** macro. 

I just looked at the 1YN using the CY43439.  They had a trivial breakdown of   SDIO bus speeds:
| SDIO bus speed modes | Max SDIO clock frequency | Max bus speed | Signaling voltage according to M.2 specification | Supported in 3.3V VDDIO Override Mode |
|----------------------|--------------------------|---------------|--------------------------------------------------|---------------------------------------|
| ddDS (Default speed) | 25 MHz                   | 12.5 MByte/s  | 1.8 V                                            | Yes                                   |
| HS (High speed)      | 50 MHz                   | 25 MByte/s    | 1.8 V                                            | Yes                                   |
| SDR12                | 25 MHz                   | 12.5 MByte/   | 1.8 V                                            | Yes                                   |
| SDR25                | 50 MHz                   | 25 MByte/s    | 1.8 V                                            | Yes                                   |
| DDR50                | 50 MHz                   | 50 MByte/     | 1.8 V                                            | Yes                                   |

claydonkey, 2023-10-31:
Haha I'm just rabbiting on incoherently  @mantriyogesh 
Whereas your posts are like brochures

mantriyogesh, 2023-10-31:
## Max Peripheral Freqs for ESP32-C6

As per C6 datasheet,
SDIO upto 50MHz supported with 4Bit mode .
[image]

SPI slave as per documentation :
[image]
For SPI in practicality, we have tested till 26MHz with around 10cm jumper wires.
Reason to limit to this: IDF throughput testing uses max **26MHz**. see: https://github.com/espressif/esp-idf/blob/b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e/components/idf_test/include/esp32c6/idf_performance_target.h#L20

You can change chipset name in above link to see `IDF_PERFORMANCE_MAX_SPI_CLK_FREQ` values for all chipsets.
In case of ESP32-C2/C3, this freq testing happened be 40MHz.
Although, iMax SPI supported freq depends upon a number of parameters. You should gradually increase the frequency to test with your chipset and ESP.

Unfortunately, SPI timing are handled by each vendor independently (no standard as such). And depending upon the timing, you can achieve max frequency. This also relies much on length of jumper wires. Ideally, 10MHz should be used for wires. For production, anyway, PCBs would be needed where you can get to real better timings. Once PCB tested, the timings can be tuned better using TRM (If really needed to be)

## DEFAULT_LISTEN_INTERVAL
Yes, this parameter should be really set to 3 (default) value.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv4N17wifi_sta_config_t15listen_intervalE

listen_interval typically affects DTIM periodicity and recommended to keep default 3. higher listen_interval is kind of non-standard way of power save, which is supported on ESP chips.
There are other better ways to save power so, to repeat, let listen_inveral be 3.

## Other considerations
Typically, the max throughput will be limited on multiple factors.
 Transport bandwidth is important aspect as you pointed. But there are other factors, like Wi-Fi driver performance (from ESP-IDF)
 Real question is how much the throughput is needed by the application, being IoT application.

mantriyogesh, 2023-10-31:
Do let us know if you can follow [2020.10.31__gh270_interim_branch_DoNOTUseInProduction.tgz](https://github.com/espressif/esp-hosted/files/13217485/2020.10.31__gh270_interim_branch_DoNOTUseInProduction.tgz) above..
that would be the baseline for our code in esp_as_mcu_host branch (later).

mantriyogesh, 2023-11-01:
> Just a quick ask. How can I reduce the packets/speed of the Raw Test?

You may just want to add a small vTaskDelay (msleep would be very big at this place) in TX task each side. But anyway, the tasks are blocking in nature and would not push unless the queues/bus is free to handle next packet.

claydonkey, 2023-11-01:
Thanks again for all you help.
I now have a genuine issue:
I am not entirely sure if this is related to my port but it seems that MULTI_BLOCK (Host Transmission in this case) does not work as expected. 
Point of fact I have assumed that MULTI_BLOCK transmissions are not allowed in esp-hosted. However, the Raw test specifies the Transmission size at 1460, so I am probably wrong in this assumption.
According to
It may be related to the SD Command SDMMC_CMD_STOP_TRANSMISSION : 12. - which certainly the STM32 sdmmc library assumes is the case.
[image]

mantriyogesh, 2023-11-01:
Worth check similar problem: https://github.com/espressif/esp-hosted/issues/246#issuecomment-1665051857

1. Multi block (basically block transfer mode) is supported in ESP-Hosted. Block size = 512
2. CMD12 is in context of SDMMC, We rely on CMD52 and CMD53 for communication.
CMD52 for registers and CMD53 for Rx/Tx data.

Very nice document where SDIO slave communication is explained for ESP: https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-reference/peripherals/sdio_slave.html

3. Init seq for SDMMC is different and SDIO is different (This is the reason you can't use STM SDMMC driver as is for SDIO). We had to create SDIO specific init sequence to avoid SDMMC related errors. Check 
https://github.com/espressif/esp-hosted/blob/479e76e80f2200a4f620809e715394c76135b2fc/esp_hosted_fg/host/stm32/driver/transport/sdio/sdio_ll.c#L355-L472  
Reference is https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-reference/protocols/esp_sdio_slave_protocol.html#esp-slave-init

Please have a look at '1031 bytes' example and how CMD53 is being used for data transfer in multiple blocks.

For simplicity, you can try to align the buffer length (and buffer) to 512 bytes and send? ESP-Hosted would send/receive that by default in block format.
In Rx side, anyway, the actual buffer payload length is interpreted from `ESP-Hosted payload header-> length` 

https://github.com/espressif/esp-hosted/blob/479e76e80f2200a4f620809e715394c76135b2fc/esp_hosted_fg/common/include/adapter.h#L27

mantriyogesh, 2023-11-01:
> However, the Raw test specifies the Transmission size at 1460
Raw throughput task is just taking 1460 as example (to mimic the practical IP packet). Ideally this could have been set to 512*3 which is current max SDIO size, but to keep practical max size, 1460 was used.

This size is also common for SPI max payload size which is 1600. In SPI case anything beyond Max IP packet + ESP_Hosted header size would, although, be testable (and would yeild higher bandwidth of transport), but would be non-realistic/non-practical scenario.

claydonkey, 2023-11-01:
By Jingo it's working.
The '1031 bytes' example was key. I rechecked the CMD53 argument I was sending....

**The mistake:**
Sending the wrong address to the esp32.

I mistakenly was setting the buffer length rather than the number of blocks to the CMD53 argument.

where arg was say 1024 instead of 2 for Ccmd53_flags |= CMD53_BLOCKMODE
(write_flag<< 28) | (addr<< 9) | (arg & 0x1ff) | cmd53_flags;

I was lucky/unlucky I was able to send 1 block before

mantriyogesh, 2023-11-01:
That's great..
Pure block mode could be assessed if faster than the block(s) + remaining bytes.

claydonkey, 2023-11-01:
https://clipchamp.com/watch/j8zXP3zsvhd

https://github.com/espressif/esp-hosted/assets/5519575/508dab4e-083f-4f35-aa82-af27595718d2

claydonkey, 2023-11-01:
Actually it's getting better!
STM32 is reporting 20Mbit/s
but I can't see that could be the case

mantriyogesh, 2023-11-01:
Yes.. happy to help!

mantriyogesh, 2023-11-02:
hello @claydonkey ,

The branch [esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host) can be referred for SDIO. ESP as SDIO master is tested in this branch.
Other MCUs which want to use their SDIO master, can refer the port layer below:

Files:
[esp_as_mcu_host/host/hosted_os_adapter.h](https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/host/hosted_os_adapter.h) and 
[esp_as_mcu_host/host/port](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host/host/port) (port layer for third party MCU host)
[esp_as_mcu_host/host/port/sdio_wrapper.c](https://github.com/espressif/esp-hosted/blob/feature/esp_as_mcu_host/host/port/sdio_wrapper.c) (SDIO master porting file)

claydonkey, 2023-11-02:
I'm gonna take a stab at porting to the new branch...
Where all the changes pretty much resided in sdio_ll.c previously where should I add the HAL in the new branch?

claydonkey, 2023-11-02:
Ideally I would like to examine the esp host code in either Visual Studio Code or the Espressif IDE. How do you maintain and manage this yourself?

mantriyogesh, 2023-11-02:
Thats correct.
You can actually use VSCode extension https://github.com/espressif/vscode-esp-idf-extension for code organization.
I personally do not use it, still rely on the basic vim and no IDE. 

Actually, we are planning to integrate [esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host) branch as external component similar to https://components.espressif.com/components/mantriyogesh/esp_hosted

This is yet very initial (hence the private branch yet). Once this is integrated, this example directory would be **removed** and https://github.com/espressif/esp-idf/tree/master/examples/wifi/iperf will be aware of ESP-Hosted external component.
Then this will be easier than ever, But this is going to take a lot of time.

By the time you can use this extension and ESP-Hosted as directory in addition to your MCU code (You would know better VSCode I have to agree)

claydonkey, 2023-11-02:
You are the **DUDE** @mantriyogesh

claydonkey, 2023-11-02:
It seems like some functionality has now been delegated to esp_hosted_ng. Specifically I am looking at the headers  like esp_wifi.h
Are you sort of merging the two codebase into esp_hosted_fg?

mantriyogesh, 2023-11-02:
Esp-Hosted-NG is actually more focussed towards Linux with standard WPA-Supplicant.

claydonkey, 2023-11-02:
g_rxcb array is not defined in sdio.drv
line 494,: 		
	`if (g_rxcb[netif_interface_type]) {
				if (buf_handle->payload_zcopy)
					g_rxcb[netif_interface_type](buf_handle->payload,
						buf_handle->payload_len, buf_handle->priv_buffer_handle);
				else
					g_rxcb[netif_interface_type](buf_handle->payload,
						buf_handle->payload_len, NULL);
			}`

mantriyogesh, 2023-11-02:
Apologies for this. Give us couple days time to fix this branch.

claydonkey, 2023-11-02:
Also the semaphore sdio_recv_SemHandle  in the master branch does not seem to accomplish anything. 
In esp-at it relates to :

sdio_host_get_intr(&intr_raw, &intr_st);  (sdio_host_transport.c) 
where an interrupt is flagged on the DATA1 pin

In esp-hosted you have the function:

ret = sdio_host_get_intr(&intr_st); where an interrupt is asked by 
sdio_driver_read_bytes(SDIO_FUNC_1, SDIO_REG(ESP_SLAVE_INT_ST_REG), (uint8_t*)intr_st, 4);
Does this accomplish the same effect or should I look to esp-at for managing the task state?

mantriyogesh, 2023-11-02:
Right, but the code is consistent with the https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_sdio_slave_protocol.html#sending-fifo:

[image]

But As I see, every code part (even the IDF example) uses one of these interrupts to determine new packet.
https://github.com/espressif/esp-idf/blob/b4268c874a4cf8fcf7c0c4153cffb76ad2ddda4e/examples/peripherals/sdio/host/main/app_main.c#L353

But I see your question valid and I think both,

mantriyogesh, 2023-11-02:
SLC0HOST_SLC0_RX_NEW_PACKET_INT_RAW

claydonkey, 2023-11-02:
Do you have iperf / raw tp results for esp sdio host on the new branch?

mantriyogesh, 2023-11-03:
I think @SohKamYung-Espressif had taken some time back for intermediate branch https://github.com/espressif/esp-hosted/issues/270#issuecomment-1787517504.

New branch, still fixing, @SohKamYung-Espressif is working on it and he would let you know as it is done.

SohKamYung-Espressif, 2023-11-03:
@claydonkey Based on an intermediate branch, these are the throughput numbers I am getting.

Setup:

- Host is ESP32, connected to a ESP32-C6 as slave. SDIO is 4-bit, running at 40MHz. Sending packets to a PC connected to the same Wi-Fi network.

Raw throughput (ESP32 <---> ESP32-C6 only) as seen on Host side:

- when Host is doing both Tx and Rx to slave, Tx is **27 Mbps**, Rx is **13 MBps**
- when Host is only doing Tx to slave, Tx is **53MBps**
- when Host is only doing Rx from slave, Rx is **27 MBps**

Iperf UDP results (iperf on ESP32 communicating with iperf on PC through ESP32-C6)

- when ESP is sending UDP to PC: **35.6 MBps**
- when ESP is receving UDP from PC: **23 MBps**

Iperf TCP results are not stable. This may be an issue with the ESP32 iperf being used in the intermediate branch.

SohKamYung-Espressif, 2023-11-03:
Forgot to add this: a packet size of 1460 bytes was used in the raw throughput transfer test.

claydonkey, 2023-11-03:
@SohKamYung-Espressif fantastic.
Thanks for detailed breakdown

mantriyogesh, 2023-11-04:
https://github.com/espressif/esp-hosted/issues/270#issuecomment-1791195233

Just to confirm on this,

Usually in ESP peripherals, INT_ST is the masked interrupt status, INT_ST = INT_RAW & INT_ENA. If you know that the specific interrupt was enabled in the ENA register, checking the RAW and ST registers are equivalent.
So in our case, it doesn't really matter.

SohKamYung-Espressif, 2023-11-06:
For reference, only one bit, `new_packet_intr_mask`, is being used in the interrupt register.

From the [ESSL SDIO code](https://github.com/espressif/idf-extra-components/blob/master/esp_serial_slave_link/essl_sdio_defs.c):

````
/*
 * SPDX-FileCopyrightText: 2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Apache-2.0
 */

// Definitions of Espressif SDIO Slave hardware

#include "essl_sdio.h"

essl_sdio_def_t ESSL_SDIO_DEF_ESP32 = {
    .new_packet_intr_mask = BIT(23),
};

essl_sdio_def_t ESSL_SDIO_DEF_ESP32C6 = {
    .new_packet_intr_mask = BIT(23),
};
````

claydonkey, 2023-11-06:
@mantriyogesh 
Am I correct in my assumption when migrating to the new branches semaphore handling:
Within SDIO transfers:

**start transfer:**

old:
`xSemaphoreTake(semahandle, portMAX_DELAY);`
new: 
`g_h.funcs->_h_get_semaphore(rpc_tx_sem,portMAX_DELAY);`

**end/failed:**

old:
`xSemaphoreGive(semahandle);`
new:
`g_h.funcs->_h_post_semaphore(rpc_tx_sem);`

**within ISR:**

old:	
```xSemaphoreGiveFromISR(sdio_recv_SemHandle, &xHigherPriorityTaskWoken);
	if (xHigherPriorityTaskWoken) {
		portYIELD_FROM_ISR(xHigherPriorityTaskWoken);
	}
```
new:	
`g_h.funcs->_h_post_semaphore_from_isr(rpc_tx_sem);`

What about Yielding?

And finally: 

**The interrupt Data Pin:**

`/* Blocking fn call. Returns when SDIO slave device generates a SDIO interupt */
int hosted_sdio_wait_slave_intr(uint32_t ticks_to_wait) {
	SDIO_FAIL_IF_NULL(hsd1.Instance);

	__SDMMC_CLEAR_FLAG(SDMMC1, SDMMC_IT_SDIOIT);
	__HAL_SD_DISABLE_IT(&hsd1, SDMMC_IT_SDIOIT);

	if (HAL_GPIO_ReadPin(GPIOC, GPIO_PIN_9) == RESET) {    // D1 if low, means SDIO interrupt
		return ESP_OK;
	}
	g_h.funcs->_h_get_semaphore(rpc_tx_sem, HOSTED_BLOCKING);
	__HAL_SD_ENABLE_IT(&hsd1, SDMMC_IT_SDIOIT);
}`

claydonkey, 2023-11-06:
Actually I think the semaphore is:
"sdio_bus_lock"
as defined and instantiated in 
`void transport_init(void(*transport_evt_handler_fp)(uint8_t)){...}`
in sdio.drv

mantriyogesh, 2023-11-07:
1. portMAX_DELAY -> HOSTED_BLOCK_MAX
2. Caution: return types are different from that of FreeRTOS standards. Generally, 0 represents success.
3. > SDIO Interrupt on D1 Data Pin

`g_h.funcs->_h_get_semaphore(rpc_tx_sem, HOSTED_BLOCKING);`
I am trying to understand why is this line needed in SDIO wait intr?
rpc_tx_sem is only specific to RPC and not related directly with SDIO as such.

> i(!g_h.funcs->_h_read_gpio(GPIOC, GPIO_PIN_9))

Instead of just relying on the pin signal on D1,
You might have to check if any reliable function which would tell that interrupt received on SDIO device. You might have to check SDIO HAL driver for your MCU.  (if the HAL supports easy way).

Also, the interrupt is verified again in second stage anyway, using 
https://github.com/espressif/esp-hosted/blob/89aee43a65ff1ccd97cba3254eee1ea1835a0aa7/host/drivers/transport/sdio/sdio_drv.c#L378 
Please check return value of this function & on failure, continue.
Ahead in the **caller** of this function, check the BIT23 is set. This means new packet received interrupt is set. clear the interrupt irrespective of bit23 being set or not.

We will also update this logic in our branch also.

4. `sdio_bus_lock` is expected to be used as SDIO bus lock and not a semaphore.

claydonkey, 2023-11-09:
@mantriyogesh 
Maybe I should open another thread but this is a continuation of my endeavours so...

**This applies to both entry points in examples:**
**iperf / arping** 

**esp-hosted hanging in prvCheckTasksWaitingTermination**

https://github.com/espressif/esp-hosted/assets/5519575/60ccd421-6841-4c50-8dd6-16cb010e3659

The patched esp-hosted compiles without any errors/warnings.

The sdio init works as is seen in the video (I assume the CIS data is correct)

**The problem**
As soon as the sdio_process_rx_task is called from pxPortInitialiseStack in sdio_drv.c
The program hangs after entering .

**sdio_drv.c** (from  **pxPortInitialiseStack**)
```
static void sdio_process_rx_task(void const* pvParameters)
{
...
while (1) {
		if (g_h.funcs->_h_dequeue_item(from_slave_queue, &buf_handle_l,	portMAX_DELAY)) {
			continue;
		}
}
...
```
**to os_wrapper.c** 
a. `item_retrieved = xQueueReceive(*q_id, item, pdMS_TO_TICKS(SEC_TO_MILLISEC(timeout)));`
or
b. `item_retrieved = xQueueReceive(*q_id, item, 0);`

Finally hanging around in FreeRTOS tasks.c

`static void prvCheckTasksWaitingTermination( void )`

I noticed that
`int hosted_dequeue_item(void * queue_handle, void *item, int timeout)`
maybe should be
`int hosted_dequeue_item(void * queue_handle, void *item, uint32_t timeout)`
as the RTOS tick is uint32_t in size. but all this does is choose a or b (above)

Seems so close

claydonkey, 2023-11-09:
@mantriyogesh
Maybe I should open another thread but this is a continuation of my endeavours so...

This applies to both entry points in examples:
iperf / arping

esp-hosted hanging in prvCheckTasksWaitingTermination

 Untitled.video.mp4 
The patched esp-hosted compiles without any errors/warnings.

The sdio init works as is seen in the video (I assume the CIS data is correct)

The problem
As soon as the sdio_process_rx_task is called from pxPortInitialiseStack in sdio_drv.c
The program hangs after entering .

sdio_drv.c (from pxPortInitialiseStack)

static void sdio_process_rx_task(void const* pvParameters)
{
...
while (1) {
		if (g_h.funcs->_h_dequeue_item(from_slave_queue, &buf_handle_l,	portMAX_DELAY)) {
			continue;
		}
}
...
to os_wrapper.c
a. item_retrieved = xQueueReceive(*q_id, item, pdMS_TO_TICKS(SEC_TO_MILLISEC(timeout)));
or
b. item_retrieved = xQueueReceive(*q_id, item, 0);

Finally hanging around in FreeRTOS tasks.c

static void prvCheckTasksWaitingTermination( void )

I noticed that
int hosted_dequeue_item(void * queue_handle, void *item, int timeout)
maybe should be
int hosted_dequeue_item(void * queue_handle, void *item, uint32_t timeout)
as the RTOS tick is uint32_t in size. but all this does is choose a or b (above)

Seems so close

claydonkey, 2023-11-09:
Strangely when I am stepping through the debugger. If I break at certain point this problem does not occur...

claydonkey, 2023-11-09:
bus_width 64. Clock Div 350
bus_width 66. Clock Div 4
BlockNbr:0 BlockSize:65536 CardSpeed:0 CardType:0 CardVersion:0 Class:0 LogBlockNbr:0 LogBlockSize:0 RelCardAdd:1
TUPLE: DEVICE, size: 3: D9 01 FF
TUPLE: MANFID, size: 4
  MANF: 6666, CARD: 2222
TUPLE: FUNCID, size: 2: 0C 00
TUPLE: FUNCE, size: 4: 00 00 02 32
TUPLE: CONFIG, size: 5: 01 01 00 02 07
TUPLE: CFTABLE_ENTRY, size: 8
  INDX: C1, Intface: 1, Default: 1, Conf-Entry-Num: 1
  IF: 41
  FS: 30, misc: 0, mem_space: 1, irq: 1, io_space: 0, timing: 0, power: 0
  IR: 30, mask: 1,   IRQ: FF FF
  LEN: FFFF
TUPLE: END
I Function 0 Blocksize: 512
I Function 1 Blocksize: 512
                           ERROR
I Starting SDIO process rx task
I Received INIT event from ESP32 peripheral
I EVENT: 12
I EVENT: 11
I capabilities: 0x1d
I Features supported are:
I        * WLAN
I        * BT/BLE
I          - HCI over SDIO
I          - BT/BLE dual mode
I EVENT: 13
I ESP board type is : 0

I Received INIT event
I Base transport is set-up

E ERR [0] ret[4294967295] for Req[0x116]

E Response not received for [116]

mantriyogesh, 2023-11-09:
Hello @claydonkey , 
Sorry, got busy in other activities, will check your msg tomorrow in detail and reply back.
But the timers and sleeps need to be ported for your MCU, and FreeRTOS or OS timing granularity.

By the way, are you using the feature/esp_as_mcu_host and ESP as SDIO  host?

If yes, please wait, the branch fix should be available tomorrow or Monday by @SohKamYung-Espressif ..

We will notify you once it is done.
I will check and get back to your original question tomorrow..

claydonkey, 2023-11-09:
@mantriyogesh 

ok solved that problem:

For some reason I had a horribly wrong routine in:

 ```C
int hosted_sdio_wait_slave_intr(uint32_t ticks_to_wait) 
```

 ```C
/* Blocking fn call. Returns when SDIO slave device generates a SDIO interupt */
int hosted_sdio_wait_slave_intr(uint32_t ticks_to_wait) {
	SDIO_FAIL_IF_NULL(hsd1->Instance);
	__SDMMC_CLEAR_FLAG(SDMMC1, SDMMC_IT_SDIOIT);
	__HAL_SD_DISABLE_IT(hsd1, SDMMC_IT_SDIOIT);
	// D1 if low, means SDIO interrupt
	if (!(g_h.funcs->_h_read_gpio(GPIOC, GPIO_PIN_9)))
		return ESP_OK;

	g_h.funcs->_h_get_semaphore(sdio_bus_lock, HOSTED_BLOCKING);
	_HAL_SD_ENABLE_IT(hsd1, SDMMC_IT_SDIOIT);
	return ESP_OK;
} 
```

changed to (but probably needs timeout and maybe needs to lock mutex
 
```C
int hosted_sdio_wait_slave_intr(uint32_t ticks_to_wait) {
	SDIO_FAIL_IF_NULL(hsd1->Instance);
	while (g_h.funcs->_h_read_gpio(GPIOC, GPIO_PIN_9)) {
	}
	return ESP_OK;
}```

mantriyogesh, 2023-11-10:
The current routine is polling, which will be very costly.
Worth to check if your MCU hal provides a better and reliable way to get SDIO interrupt..

claydonkey, 2023-11-10:
Indeed.
I forgot I wrote that routine based on the esp-idf routine but I am having probs...
The SDIOIT flag is never set and the app hangs at 

**I Starting SDIO process rx task**

```C
create new **sdio_trans_ready_sem** semaphore.

/* Blocking fn call. Returns when SDIO slave device generates a SDIO interrupt */
int hosted_sdio_wait_slave_intr(uint32_t ticks_to_wait) {
	SDIO_FAIL_IF_NULL(hsd1->Instance);

	__SDMMC_DISABLE_IT(hsd1->Instance, SDMMC_IT_SDIOIT);
	__SDMMC_CLEAR_FLAG(hsd1->Instance, SDMMC_IT_SDIOIT);
	if (!g_h.funcs->_h_read_gpio(GPIOC, GPIO_PIN_9))
		return ESP_OK;

	g_h.funcs->_h_get_semaphore(sdio_trans_ready_sem, 0);

	__SDMMC_ENABLE_IT(hsd1->Instance, SDMMC_IT_SDIOIT);

	if (g_h.funcs->_h_get_semaphore(sdio_trans_ready_sem, ticks_to_wait) == pdTRUE) {
		return ESP_OK;
	} else {
		return ESP_ERR_TIMEOUT;
	}
}
```
and in the ISR

```C
void SDMMC1_IRQHandler(void) {

	...

	if (__HAL_SD_GET_FLAG(card, SDMMC_IT_SDIOIT)) {
		__SDMMC_CLEAR_FLAG(SDMMC1, SDMMC_IT_SDIOIT);
		__HAL_SD_DISABLE_IT(hsd1, SDMMC_IT_SDIOIT);
		g_h.funcs->_h_post_semaphore_from_isr(sdio_trans_ready_sem);

	}

...
```

claydonkey, 2023-11-10:
@mantriyogesh 
I've ditched checking the D1 interrupt on:
 ```C
int hosted_sdio_wait_slave_intr(uint32_t ticks_to_wait);
```
As there seems to be a bug/design feature in the STM32 where the SDIOIT interrupt is garbled when using SDIO DMA transactions

I'm just using the routines :

```C
sdio_get_intr(&interrupts);
		sdio_clear_intr(interrupts); 
```
Do you think that's a bad idea?

mantriyogesh, 2023-11-11:
###

mantriyogesh, 2023-11-11:
No it should also be fine, but you have to check the CPU usage.
This would be polling mode.
So you kight have to add some sleep if the packet not arrived.

Obviously, your sleep duration would be affecting the  data throughout.
Also make sure the read packet counter correctness, before actually reading..

claydonkey, 2023-11-11:
I understand now that as I said in a previous  comment that the 'polling' /keep alive packet in the main branch of esp-hosted was doing exactly what I've resorted to. The keep alive was just querying interrupts...
I have got as far as I can get now. The iperf example does not connect to the wifi and I think this is because the new rpc calls have not been changed for RPC_ID__Req_WifiStart. I can't see any connection (ssid/pw) routine being processed. So I assume that @SohKamYung-Espressif will be adding these in the next commit in the feature branch...
I am pretty impatient so any advice getting this working (as always) would be appreciated and thanks again @mantriyogesh

claydonkey, 2023-11-11:
### PARTIALLY SOLVED

#### Solution
#define INCLUDE_xTimerPendFunctionCall		 1
which is not defined in sdkconfig.h

mantriyogesh, 2023-11-11:
Oh.. suggestion taken.

Although, I would be able to reply to queries next week due to festive holidays.
We will also try to push the current SDIO branch changes asap. They are into review already, just to let you know..

mantriyogesh, 2023-11-11:
Solution or application example at host is based on ESP-IDF wifi station / softap/ scan/iperf examples.

What I mean, without change to the esp IDF example, we run the wifi as remote instead of native wifi.
Essentially, those examples already are based on event group and based on FreeRTOS , so we missed focus on group event bits.

claydonkey, 2023-11-11:
Thanks @mantriyogesh 
Happy Diwali
Enjoy yourself (you seem to work v hard)
I will start another issue post for next week. There are problems which may be associated with STM32 and/or RTOS.
I'm gonna be grilling them this weekend.
Again
HAPPY DIWALI

Evlers, 2024-01-07:
## About branch for porting

- I have adapted the main branch into the wlan framework of rt-thread (indirectly adapted to lwip) and implemented the adaptation of STA functions.
- But reading your describe, I find that the [esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host) branch is more suitable for the MCU scheme!!!
- I think this branch supports more complete features (such as STA's RSSI and API for getting mac addresses), and the hierarchy is very clear

Do you recommend that I use the master branch or the [esp_as_mcu_host](https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host)  branch?

> 你好 ，
> 
> 我们还有另一个分支，https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host 我们使用 ESP（从站）- SPI/SDIO - ESP（主机），其中我们有：
> 
> * 改进了我们的代码（错误修复和功能）
> * 更好的 iperf 吞吐量
> * 清晰的端口层，可轻松适应任何 MCU。
> 
> #### STM32的局限性：
> 1. SDIO：由于STM32并没有真正提供纯粹的SDIO（我们实现时AFAIK），我们不得不手动进行SDIO HAL层初始化。
> 2. 这样复制的 HAL 层实际上是不可维护的，因为它在客户选择中从 STM32 到 STM32 不等。
> 3. 用户必须真正从一个 STM32 HAL 驱动程序迁移到所需的 STM32 HAL 驱动程序代码（因为我们必须将sdio_ll作为 HAL 端口代码导入），由于硬件依赖性，这有时会变得难以重现。
> 
> 这就是我们实现 ESP32 本身作为 Host 的原因。此外，使用 ESP 作为主机有助于使用完整的潜在功能集。
> 
> 虽然到目前为止，我们仍然支持特定的STM32的“SDIO”，但STM32F412ZGT6，我真的建议最初使用带有SPI的评估 https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host 作为传输。
> 
> SDIO（ESP 主机-SDIO-ESP 从属）传输也受支持，但在最新更改中中断，我们预计将在下周修复。
> 
> 如果您想尝试这个，请告诉我。我将解释（可能会创建小文档以使用此分支进行评估）端口层、引脚连接等。

mantriyogesh, 2024-01-08:
Yes, your observations are absolutely correct.
We already have solution as ESP as host and slave in branch, https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host with excellent throughput on SPI / SDIO (ESP as host)

In this branch,  ESP is used as host MCU. So we tried to re-use ESP-IDF as much as possible at host as well.
However, for your MCU, some of the components will need copying (for example, esp_events) etc.
This branch is definitely better than master, as already has easy port layer for most things.
Once port layer is implemented, new host should work as host.

Long term view is that,
This branch we are trying to take ahead to provide even more easier experience than master.
Master is little older way of doing the things but it is well tested solution.

We have **intermediate internal branch**, which @SohKamYung-Espressif  will share in some time.
This branch should be good to start for non-ESP MCU (feature/esp_as_mcu_host is much ahead than this branch).

I think @claydonkey already has ported much part of it and if his code is open, you can refer to.
Else you can start from intermediate branch. Once we integrate feature/esp_as_mcu_host branch or similar branch for non-ESP as Host MCU, we will notify you on this.

If you face any issues, you can raise separate issue and also refer to this issue in that.

Evlers, 2024-01-08:
> Yes, your observations are absolutely correct. We already have solution as ESP as host and slave in branch, https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host with excellent [...]

**Thank you very much for your reply**

I have finished porting the master branch, I will continue to port the feature/esp_as_mcu_host branch.
I found that this branch does not have the README.md file, so I would like to ask if this slave is built using the release/v5.2 branch of esp-idf?

mantriyogesh, 2024-01-08:
Yes. Supported with release/v5.0, release/v5.1, release/v5.2, master.
You need spi or sdio?

Evlers, 2024-01-08:
Ok, I'm going to use this branch as a submodule.
We are mainly SPI based, SDIO is a relatively scarce peripheral in the MCU.

This branch will eventually merge into the master branch, right?

mantriyogesh, 2024-01-08:
No. This branch will be independent.

[4 more replies]