espressif/esp-hosted #271: linux kernel crash which may be triggerd by esp32 module

IOTBaicheng, 2023-11-02:
We got a linux kernel crash which may be triggerd by esp32 module. the  kernel crash info is as follows. Plz refers to the attached file for details.  
It is hard  to reproduce this error. I got this error twice in total.
[image]
[kernel_error.txt](https://github.com/espressif/esp-hosted/files/13238143/kernel_error.txt)

kapilkedawat, 2023-11-02:
Hi @IOTBaicheng , what's the hosted driver version and linux version that you are using? 
Could you also please tell us regulatory_init+0x71/0xf0) points to which line in your kernel image?

IOTBaicheng, 2023-11-03:
[esp_log.txt](https://github.com/espressif/esp-hosted/files/13245799/esp_log.txt)
linux version:Linux version 4.9.84 (yao@ubuntu) (gcc version 8.2.1 20180802 (GNU Toolchain for the A-profile Architecture 8.2-2018-08 (arm-rel-8.23)) )
My question is : how to find code line of regulatory_init+0x71/0xf0) . I have not done this before. Can you give me some instructions.

mantriyogesh, 2023-11-03:
Simple internet search led me to https://stackoverflow.com/questions/38709275/how-to-debug-linux-kernel-module

#### Generic Kernel module backtrace debug.
It expects basic debugging at application is known.

Check https://www.dedoimedo.com/computers/crash-analyze.html This has many kernel debug techniques.
Just to quick check if esp32_spi.ko has triggered this issue or not,
check above page and search for string 'Step 1: Kernel module'

To add the debug symbol, please add '-g' in https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/Makefile#L43
and rebuild your module and reproduce the issue and note the crash (backtrace) top line.

and follow objdump steps to localise the line.
If this line is not in the esp32_spi.ko, it would not be from esp32 kernel module and something else might the problem lie.

kapilkedawat, 2023-11-03:
> [esp_log.txt](https://github.com/espressif/esp-hosted/files/13245799/esp_log.txt) linux version:Linux version 4.9.84 (yao@ubuntu) (gcc version 8.2.1 20180802 (GNU Toolchain for the A-profile [...]

Please compile your kernel image with -g without any cflags/configuration changes, that will help to embed symbols with code. Once the vmlinux image is available, you can use toolchain's objdump  on vmlinux to get the line number in source code.

crash address = (address of regulatory_init + 0x71)

IOTBaicheng, 2023-11-04:
Thanks, I will learn these to get more infomation.

IOTBaicheng, 2023-11-07:
hello the following pics show the code line infomation.
[image]

[image]

IOTBaicheng, 2023-11-07:
regulatory_init+0x71
[image]

mantriyogesh, 2023-11-07:
I am still not sure how is this linked with ESP-Hosted yet.

From your trace:
```
------------[ cut here ]------------
WARNING: CPU: 0 PID: 1 at net/wireless/reg.c:516 regulatory_init+0x71/0xf0
db.txt is empty, you should update it...Modules linked in:
CPU: 0 PID: 1 Comm: swapper/0 Not tainted 4.9.84 #72
Hardware name: SStar Soc (Flattened Device Tree)
[<c00125d1>] (unwind_backtrace) from [<c001001b>] (show_stack+0xb/0xc)
[<c001001b>] (show_stack) from [<c012cce3>] (dump_stack+0x57/0x70)
[<c012cce3>] (dump_stack) from [<c007230b>] (__warn+0x83/0xa0)
[<c007230b>] (__warn) from [<c001caef>] (warn_slowpath_fmt+0x1f/0x28)
[<c001caef>] (warn_slowpath_fmt) from [<c04c8cb1>] (regulatory_init+0x71/0xf0)
[<c04c8cb1>] (regulatory_init) from [<c04c8beb>] (cfg80211_init+0x37/0x8c)
[<c04c8beb>] (cfg80211_init) from [<c0009605>] (do_one_initcall+0xd1/0x104)
[<c0009605>] (do_one_initcall) from [<c04b6a7d>] (kernel_init_freeable+0x12d/0x180)
[<c04b6a7d>] (kernel_init_freeable) from [<c0308877>] (kernel_init+0x13/0xb4)
[<c0308877>] (kernel_init) from [<c000d5c1>] (ret_from_fork+0x11/0x30)
---[ end trace 3b3a15d2b63b913e ]---
```

Code mapping: https://elixir.bootlin.com/linux/v4.9.84/source/net/wireless/reg.c#L516

Possibly regulatory db is missing, you might have to dig in:
https://www.kernel.org/doc/html/next/networking/regulatory.html
https://www.marcusfolkesson.se/blog/linux-wireless-regulatory/

Also, Not sure what it means, but just an observation, that the backtrace 'WARN' you had received might be removed in later kernel versions.
For example, https://elixir.bootlin.com/linux/v5.2.21/source/net/wireless/reg.c#L3951

Possible that some config is missing for cfg80211 to run?

mantriyogesh, 2023-11-07:
[image]

Also, is it possible to test with `different Linux` / `different Linux Kernel version`?

mantriyogesh, 2023-11-07:
also, what happens if you run sample command like:
`iw reg get`

[image]

IOTBaicheng, 2023-11-07:
> also, what happens if you run sample command like: `iw reg get`
> 
> [image]

my system doesn't support this command.

mantriyogesh, 2023-11-07:
Either download iw in your system or use a similar command to query the regulatory database.
The intention is that you have the system which has correctly setup and has correct regulatory.db and its config.

Other alternative just to debug, (**not recommended** although in production), disable  CONFIG_CFG80211_INTERNAL_REGDB in KConfig and rebuild and reflash the Kernel.

⚠️ Please beware, this is just a suggestion and any incorrect steps might leave your system in non bootable format. (we would not be responsible! 😄  )

Also, just to localise, if you happen to have native Wi-Fi on your system, reboot and do not use esp32 kernel module. just use the default native Wi-Fi and see if this issue reproduces.

IOTBaicheng, 2023-11-07:
I don't think the crash is caused by regulatory. If this is the reason, the system will crash every time when it connect the AP.

IOTBaicheng, 2023-11-07:
There is a new crash may provide more infomation.
[gateway_test_two_error.txt](https://github.com/espressif/esp-hosted/files/13277195/gateway_test_two_error.txt)

mantriyogesh, 2023-11-07:
```
INFO: rcu_preempt self-detected stall on CPU
	0-...: (1 GPs behind) idle=301/140000000000002/0 softirq=1196/1199 fqs=8201 
	 (t=21009 jiffies g=198 c=197 q=496)
Task dump for CPU 0:
plc_test        R  running task        0   997      1 0x00000002
[<c00125d1>] (unwind_backtrace) from [<c001001b>] (show_stack+0xb/0xc)
[<c001001b>] (show_stack) from [<c0072cab>] (rcu_dump_cpu_stacks+0x65/0x72)
[<c0072cab>] (rcu_dump_cpu_stacks) from [<c004d9b3>] (rcu_check_callbacks+0x1df/0x5e4)
[<c004d9b3>] (rcu_check_callbacks) from [<c004fce7>] (update_process_times+0x1f/0x40)
[<c004fce7>] (update_process_times) from [<c005aa91>] (tick_nohz_handler+0x39/0x74)
[<c005aa91>] (tick_nohz_handler) from [<c020609b>] (arch_timer_handler_virt+0x1f/0x26)
[<c020609b>] (arch_timer_handler_virt) from [<c0048cf5>] (handle_percpu_devid_irq+0x41/0xc0)
[<c0048cf5>] (handle_percpu_devid_irq) from [<c004631d>] (generic_handle_irq+0xd/0x16)
[<c004631d>] (generic_handle_irq) from [<c0046693>] (__handle_domain_irq+0x63/0x74)
[<c0046693>] (__handle_domain_irq) from [<c00092bb>] (gic_handle_irq+0x3b/0x64)
[<c00092bb>] (gic_handle_irq) from [<c00106e5>] (__irq_svc+0x65/0xac)
Exception stack(0xc2d1feb0 to 0xc2d1fef8)
fea0:                                     00400000 c04e5900 00000000 00000000
fec0: ffffe000 00000002 00000000 c051ccc0 c3e08000 0000000a b6f5b000 c04e4080
fee0: 001bd60a c2d1ff00 c001fbc7 c001f8d4 40080133 ffffffff
[<c00106e5>] (__irq_svc) from [<c001f8d4>] (__do_softirq+0x48/0x1c4)
[<c001f8d4>] (__do_softirq) from [<c001fbc7>] (irq_exit+0x37/0x78)
[<c001fbc7>] (irq_exit) from [<c004667d>] (__handle_domain_irq+0x4d/0x74)
[<c004667d>] (__handle_domain_irq) from [<c00092bb>] (gic_handle_irq+0x3b/0x64)
[<c00092bb>] (gic_handle_irq) from [<c0010b01>] (__irq_usr+0x61/0x80)
Exception stack(0xc2d1ffb0 to 0xc2d1fff8)
ffa0:                                     00040158 00000000 00001df1 00046368
ffc0: 000243fc 00000000 00012760 00000000 00000000 00000000 b6f5b000 bea2dc6c
ffe0: 00000000 bea2dc40 b6d53eac b6ba036c 20080010 ffffffff
debug:
Soft IRQ_COUNTER:0 
ms_dump_irq_count
irq:000 count:       1
irq:002 count:19197845
irq:003 count:      13
irq:005 count:     366
irq:027 count:   56660
irq:098 count:   19914
irq:099 count:19206850
irq:120 count:     301
IRQ_COUNTER:0 
INFO: rcu_preempt self-detected stall on CPU
	0-...: (1 GPs behind) idle=301/140000000000002/0 softirq=1196/1199 fqs=10697 
	 (t=27312 jiffies g=198 c=197 q=504)
Task dump for CPU 0:
plc_test        R  running task        0   997      1 0x00000002
[<c00125d1>] (unwind_backtrace) from [<c001001b>] (show_stack+0xb/0xc)
[<c001001b>] (show_stack) from [<c0072cab>] (rcu_dump_cpu_stacks+0x65/0x72)
[<c0072cab>] (rcu_dump_cpu_stacks) from [<c004d9b3>] (rcu_check_callbacks+0x1df/0x5e4)
[<c004d9b3>] (rcu_check_callbacks) from [<c004fce7>] (update_process_times+0x1f/0x40)
[<c004fce7>] (update_process_times) from [<c005aa91>] (tick_nohz_handler+0x39/0x74)
[<c005aa91>] (tick_nohz_handler) from [<c020609b>] (arch_timer_handler_virt+0x1f/0x26)
[<c020609b>] (arch_timer_handler_virt) from [<c0048cf5>] (handle_percpu_devid_irq+0x41/0xc0)
[<c0048cf5>] (handle_percpu_devid_irq) from [<c004631d>] (generic_handle_irq+0xd/0x16)
[<c004631d>] (generic_handle_irq) from [<c0046693>] (__handle_domain_irq+0x63/0x74)
[<c0046693>] (__handle_domain_irq) from [<c00092bb>] (gic_handle_irq+0x3b/0x64)
[<c00092bb>] (gic_handle_irq) from [<c00106e5>] (__irq_svc+0x65/0xac)
Exception stack(0xc2d1feb0 to 0xc2d1fef8)
fea0:                                     00400000 c04e5900 00000000 00000000
fec0: ffffe000 00000002 00000000 c051ccc0 c3e08000 0000000a b6f5b000 c04e4080
fee0: 001bd60a c2d1ff00 c001fbc7 c001f8d4 40080133 ffffffff
[<c00106e5>] (__irq_svc) from [<c001f8d4>] (__do_softirq+0x48/0x1c4)
[<c001f8d4>] (__do_softirq) from [<c001fbc7>] (irq_exit+0x37/0x78)
[<c001fbc7>] (irq_exit) from [<c004667d>] (__handle_domain_irq+0x4d/0x74)
[<c004667d>] (__handle_domain_irq) from [<c00092bb>] (gic_handle_irq+0x3b/0x64)
[<c00092bb>] (gic_handle_irq) from [<c0010b01>] (__irq_usr+0x61/0x80)
Exception stack(0xc2d1ffb0 to 0xc2d1fff8)
ffa0:                                     00040158 00000000 00001df1 00046368
ffc0: 000243fc 00000000 00012760 00000000 00000000 00000000 b6f5b000 bea2dc6c
[... 26 lines cut ...]
[<c004631d>] (generic_handle_irq) from [<c0046693>] (__handle_domain_irq+0x63/0x74)
[<c0046693>] (__handle_domain_irq) from [<c00092bb>] (gic_handle_irq+0x3b/0x64)
[<c00092bb>] (gic_handle_irq) from [<c00106e5>] (__irq_svc+0x65/0xac)
Exception stack(0xc2d1feb0 to 0xc2d1fef8)
fea0:                                     00400000 c04e5900 00000000 00000000
fec0: ffffe000 00000002 00000000 c051ccc0 c3e08000 0000000a b6f5b000 c04e4080
fee0: 001bd60a c2d1ff00 c001fbc7 c001f8d4 40080133 ffffffff
[<c00106e5>] (__irq_svc) from [<c001f8d4>] (__do_softirq+0x48/0x1c4)
[<c001f8d4>] (__do_softirq) from [<c001fbc7>] (irq_exit+0x37/0x78)
[<c001fbc7>] (irq_exit) from [<c004667d>] (__handle_domain_irq+0x4d/0x74)
[<c004667d>] (__handle_domain_irq) from [<c00092bb>] (gic_handle_irq+0x3b/0x64)
[<c00092bb>] (gic_handle_irq) from [<c0010b01>] (__irq_usr+0x61/0x80)
Exception stack(0xc2d1ffb0 to 0xc2d1fff8)
ffa0:                                     00040158 00000000 00001df1 00046368
ffc0: 000243fc 00000000 00012760 00000000 00000000 00000000 b6f5b000 bea2dc6c
ffe0: 00000000 bea2dc40 b6d53eac b6ba036c 20080010 ffffffff
```

I still failed to see any direct ESP-Hosted function has triggered this new crash / old crash.

But anyway, to understand further,
1. Can you please monitor the top or some process manager to see which process has stalled the CPU?
2. Unhook(or stop and do not run) the user application you are running if any and just ping to some IP after AP
3. Any pattern when you see this crash (for example, reload the kernel module / first time without any data/ first time with data/ etc)
4. Where are the ESP logs when you see this issue.

IOTBaicheng, 2023-11-07:
[image]
this picture shows that the system reset when  the esp32 interface is intalling.

mantriyogesh, 2023-11-07:
1. Can you send the ESP side logs?
2. What is the
- git commit of ESP side code (If using release binary, let us know the release version)
- git commit of host side code (If using release code, let us know the link from which you had downloaded the host code)

mantriyogesh, 2023-11-07:
Also, provide any code change (in patch) you had done at either place (esp and/or host).

1. If you have build & flashed ESP from sources, please provide:
$ cd esp-hosted/esp_hosted_ng/esp/esp_driver/esp-idf
$ git status
$ git log --oneline |head -4
2. cd esp-hosted
$ git status
$ git diff
$ git log --oneline |head -4

2. On linux, 
$ cd esp-hosted
$ git status
$ git diff
$ git log --oneline |head -4

IOTBaicheng, 2023-11-08:
1. IDF: https://github.com/espressif/esp-idf/tree/91dc99a3ce09b66c51b8be884e2ed6b6e1558747   with no change.
2. ESP-Hosted-NG:
Administrator@fubc MINGW64 /e/code/ESP32/esp-hosted (master)
$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

Administrator@fubc MINGW64 /e/code/ESP32/esp-hosted (master)
$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean

Administrator@fubc MINGW64 /e/code/ESP32/esp-hosted (master)
$ git log --oneline |head -4
5910e16 Merge branch 'feature/p4_ci_deply_script_ym2_17aug' into 'master'
574ef06 Adding CI script for sync of branch 'feature/esp_as_mcu_host'
93d773d Merge branch 'ng_build_fix_ym2_10aug' into 'master'
b853bd4 NG: Build Fix for copying wifi lib

changes with patch file:
[esp.patch](https://github.com/espressif/esp-hosted/files/13293697/esp.patch)

IOTBaicheng, 2023-11-08:
> 1. IDF: https://github.com/espressif/esp-idf/tree/91dc99a3ce09b66c51b8be884e2ed6b6e1558747   with no change.
> 2. ESP-Hosted-NG:
>    Administrator@fubc MINGW64 /e/code/ESP32/esp-hosted (master)
>    $ [...]

Wait a moment. I will check ESP-Hosted-NG commit  one more time.

mantriyogesh, 2023-11-08:
Lower your SPI to 10.
/home/yao/ESP/esp_hosted_ng/host/spi/esp_spi.c -> SPI_INITIAL_CLK_MHZ

You had changed SPI mode to 0.
Possibly use same SPI mode as that of ESP side. (Note,  SPI mode 0 is not available at ESP slave)

You can optimize clk freq for SPI later.
Confirm with lower frequencies and same mode of SPI both sides.

Q. Do you use same code base for ESP and Host?

IOTBaicheng, 2023-11-08:
> Wait a moment. I will check ESP-Hosted-NG commit one more time.

The commit infomation is correct.

mantriyogesh, 2023-11-15:
any update here?

IOTBaicheng, 2023-11-15:
Not fixed yet. We are testing with new board. 
This problem has become less frequent.

mantriyogesh, 2023-11-15:
Please check the raw throughput test.
If you have issues in SPI timings, you will end up in unexpected failures.

So at first the underlying transport should be stable and reliably run raw throughput in both tx and Rx directions.

Do not attempt spi mode mismatch at first place. Check the porting_guide.md also for all porting related discussion.

IOTBaicheng, 2023-11-15:
OK. Thanks a lot.

mantriyogesh, 2023-11-28:
Considering #286, Shall we close this?

IOTBaicheng, 2023-12-08:
> Considering #286, Shall we close this?

Sure. Thanks.

closed 2023-12-08 by mantriyogesh