espressif/esp-hosted #272: Fail to creat GATT server

linchanghe123, 2023-11-03:
Hello @mantriyogesh
According to the guidance of docs/bluetooth.md, the following error was reported when using the bluetoothctl tool to create gatt servers. What is the reason for this?
[image]
[image]
[image]

thanks

mantriyogesh, 2023-11-03:
I see that HCIconfig is down.
I think the ESP-Hosted is not setup correctly.

Can you please check if raw throughput test is working fine first?

mantriyogesh, 2023-11-03:
Dmesg and esp logs can also point the issues

mantriyogesh, 2023-11-15:
Hello @linchanghe123 ,

Can you please let us know the above logs, if issue is still faced?

linchanghe123, 2023-11-19:
We have implemented the connection function by calling the functions of the bluez open-source library ourselves. Thank you!

closed 2023-11-19 by linchanghe123