espressif/esp-hosted #275: Raspberry Pi with ESP32-C6 in SDIO Mode failed with error -84

smart4energy, 2023-11-13:
I have external pull ups resistors on Data0-Data3 and CMD. But at the moment my cables are 20cm long. Could these error raise because of the cable length ? 

[ 9566.982105] mmc1: queuing unknown CIS tuple 0x01 [d9 01 ff] (3 bytes)
[ 9566.989917] mmc1: queuing unknown CIS tuple 0x1a [01 01 00 02 07] (5 bytes)
[ 9566.993292] mmc1: queuing unknown CIS tuple 0x1b [c1 41 30 30 ff ff ff ff] (8 bytes)
[ 9566.999682] mmc1: new SDIO card at address 0001
[ 9567.000247] esp_probe: ESP network device detected
[ 9567.000754] esp_sdio: probe of mmc1:0001:1 failed with error -84
[ 9567.001039] esp_sdio: probe of mmc1:0001:2 failed with error -22

mantriyogesh, 2023-11-14:
Hello @smart4energy ,

Yes, there is very high chance that is the reason for timeouts.
Ideally, you should not use the SDIO with jumper cables.
If cannot have alternative, cut and make them smaller with soldering to about <5 cm, all equal lengths.

Also, you can try other transport, SPI for evaluation, if jumper wires cannot be eliminated in testing.
For SDIO, with smaller wires, you can first test the 1-bit SDIO working, then try 4-bit mode.

For raspberry Pi, 1 bit mode can be configured as:
https://github.com/espressif/esp-hosted/issues/269#issuecomment-1780697351

mantriyogesh, 2023-11-16:
#### SDIO

What happens on raw throughput testing with 1 bit mode?
How long your jumper cables are? Use <5cm (smaller the better) for SDIO, also please check required pull-ups are in place.
All such hardware details, issues, and explanation can be referred from [proting_guide.md](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md)

#### SPI

Your observation wrt FSPICS0 is correct.
GPIO16 is not on devkit header.

But, you can change to any other GPIO, that is the reason we use GPIO10.
Another reason, just for your info, GPIO10 also helps us to keep pin-to-pin compatibility of ESP32-C3 and ESP32-C6 module pins.
It has drawback that SPI is routed through GPIO matrix instead of IO_MUX.
But that's okay. We have tested till 26MHz with PCB and ESP32-C6 works fine.

If you use wires, we would recommend not to go beyond 10MHz, as delays are dependent on the length, quality and type of jumper cables.
Again, even in case of SPI, we would suggest to use as small as possible jumper cables. < 10cm is recommended.

smart4energy, 2023-11-16:
Okay perfect, i didn't know that i can change the spi. I did it with success.

But i will try to create me a pcb to test it better.

But nerverless what you @mantriyogesh would suggest us to use on our own system on module. We want to use the esp32 as co-worker for wifi. Linux is our os. Nothing special is needed at the moment, just wifi and creating access point for installation.

Can we use the esp32-c6 for production or is this to early to use ?  

And my last question can i also flash the c6 via sdio or must be the esp32 preflashed ? 

Best regards and we can close this issue :)

mantriyogesh, 2023-11-16:
Hello @smart4energy ,
Yes, pcb would be much reliable performance.

For choosing ESP chipset, this link should be easy:
https://products.espressif.com/#/product-selector

Actually, ESP32 can be useful, if you have better performance needs. It comes with dual core options.
ESP32 allows you Classic -BT options like A2DT etc conventional bluetooth along with BLE 4.2 version.

Whereas ESP32-C6, although single core should be also quire fine. ESP32-C6 provides you additionally Wi-Fi 6 compatibility. It has only BLE, but with 5.x version.

If you plan to use SPI as your transport, ESP32-C6 has SPI CLK support till 26MHz, better than ESP32 (10Mhz).
ESP32-C2 can be winner with low cost.

Max SPI Freq supported (you would need to test although with your Linux host for interoperability and SPI timings):
ESP32: 10Mhz
S3/C3/C2: 40Mhz (30 is stable)
C6: 26Mhz

It depends upon what feature sets you use and how to optimize your mem/cost/feature-set.
For SDIO, ESP32 and ESP32-C6 are possible as slave. (Later C5 is also awaited).

#### OTA
OTA is supported on all the chipsets in FG over the transport. There are two alternative ways currently OTA supported: full binary available on host (C based OTA demo)
Full binary not required and chunk by chunk downloaded over Wi-Fi (python OTA demo)

Although this is just demo and not limitation as such. C based solution can be used with client http library for OTA. This just flexibility I would say. The slave reuses same transport for both, Wi-Fi and OTA!

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/ota_update.md

I hope the questions are answered..

smart4energy, 2023-11-17:
Perfect thank you for your answer :) 

We will take the c6 because of the possibility of coexistence of zigbee / matter in future and sdio communication.

Have a nice weekend

closed 2023-11-17 by smart4energy