espressif/esp-hosted #278: Under SoftAP mode, when station disconnected from SoftAP, the station mac address in the App Event response is incorrect

zhou88mang, 2023-11-16:
when i start softap ,and use my phone to connect to esp32, the log as follow
 wlan ap_start dbbz 12345678
[1062.498237] esp32 softAP started
[1062.498362] SoftAp : dbbz start success
[1153.280249] ap3 new lease, IP: 192.168.100.100, MAC: 42:6F:C1:66:95:DC
[1153.280520] station has connected, IP:192.168.100.100, MAC: 42:6F:C1:66:95:DC
the mac adress of my phone is 42:6F:C1:66:95:DC

when i closed wifi of my phone , the log as follow
[1166.941242] 1970-01-01 00:19:26 >  App EVENT: SoftAP mode: Disconnect MAC[bc:a7:40:3f:00:00]
[1166.941567] ap mode : a station: bc:a7:40:3f:00:00 has disc

i do not konow that  MAC[bc:a7:40:3f:00:00] comes from.
And the sourcecode  i used is esp-hosted-release-fg-v0.0.5

mantriyogesh, 2023-11-16:
As I understand,  that the mac address is not correct on the disconnect event.
Give us time to fix this (will reproduce this on master and give you a fix on master)

zhou88mang, 2023-11-16:
> As I understand, that the mac address is not correct on the disconnect event. Give us time to fix this (will reproduce this on master and give you a fix on master)

ok, thanks!

zhou88mang, 2023-12-14:
Is there any progress ？？

zhou88mang, 2023-12-18:
> As I understand, that the mac address is not correct on the disconnect event. Give us time to fix this (will reproduce this on master and give you a fix on master)

any progress ？？

mantriyogesh, 2023-12-18:
Hello @zhou88mang ,

We were off track on this.
I will check today and let you know..

mantriyogesh, 2023-12-18:
Hello @zhou88mang ,

I tried myself with the ESP softAP.

<details>
<summary> Linux commands fired and output </summary>
### Build and start softAP

#### Terminal 1
```
ym2@rpi2:~/h6/esp_hosted_fg/host/linux/host_control/c_support $ make
gcc -C -Wall --sysroot= -I/home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/protobuf-c -I/home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/include -I/home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/include -I/home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/src/include -I/home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../virtual_serial_if/include -I/home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../components/include -I/home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../port/include -I. /home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/protobuf-c/protobuf-c/protobuf-c.c /home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../../common/esp_hosted_config.pb-c.c /home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/src/ctrl_core.c /home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../control_lib/src/ctrl_api.c /home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../virtual_serial_if/src/serial_if.c /home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../../components/src/esp_queue.c /home/ym2/h6/esp_hosted_fg/host/linux/host_control/c_support/../../port/src/platform_wrapper.c ./test_utils.c -lpthread -lrt test.c -o test.out
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_TERMINAL = "iTerm2",
	LC_CTYPE = "UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
ym2@rpi2:~/h6/esp_hosted_fg/host/linux/host_control/c_support $ sudo ./test.out softap_start
Enable heartbeat with duration 20
Heartbeat operation successful
esp32 softAP started
ethap0 interface down
MAC address 10:97:bd:f0:e1:d5 set to ethap0 interface
ethap0 interface up

Requested operation complete
Sleeping for some time just to showcase heartbeat
^C
Clean-up and exit
Disable Heartbeat
Heartbeat operation successful
ym2@rpi2:~/h6/esp_hosted_fg/host/linux/host_control/c_support $
ym2@rpi2:~/h6/esp_hosted_fg/host/linux/host_control/c_support $ sudo ./test.out softap_start
Enable heartbeat with duration 20
Heartbeat operation successful
esp32 softAP started
ethap0 interface down
MAC address 10:97:bd:f0:e1:d5 set to ethap0 interface
ethap0 interface up

Requested operation complete
Sleeping for some time just to showcase heartbeat
^C
Clean-up and exit
Disable Heartbeat
Heartbeat operation successful
```

### Start DHCP server on Linux ethap0

```
ym2@rpi2:~/h6/esp_hosted_fg/host/linux/host_control/c_support $ sudo dnsmasq --no-daemon --no-resolv --no-poll --dhcp-script=/system/bin/dhcp_announce --dhcp-range=192.168.4.1,192.168.4.20,1h
dnsmasq: started, version 2.89 cachesize 150
dnsmasq: compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset nftset auth cryptohash DNSSEC loop-detect inotify dumpfile
dnsmasq: warning: no upstream servers configured
dnsmasq-dhcp: DHCP, IP range 192.168.4.1 -- 192.168.4.20, lease time 1h
dnsmasq: read /etc/hosts - 7 names

dnsmasq-dhcp: DHCPDISCOVER(ethap0) be:bd:5d:41:8d:48
dnsmasq-dhcp: DHCPOFFER(ethap0) 192.168.4.7 be:bd:5d:41:8d:48
dnsmasq-dhcp: DHCPDISCOVER(ethap0) be:bd:5d:41:8d:48
dnsmasq-dhcp: DHCPOFFER(ethap0) 192.168.4.7 be:bd:5d:41:8d:48
dnsmasq-dhcp: DHCPDISCOVER(ethap0) be:bd:5d:41:8d:48
dnsmasq-dhcp: DHCPOFFER(ethap0) 192.168.4.7 be:bd:5d:41:8d:48
dnsmasq-dhcp: DHCPREQUEST(ethap0) 192.168.4.7 be:bd:5d:41:8d:48
dnsmasq-dhcp: DHCPACK(ethap0) 192.168.4.7 be:bd:5d:41:8d:48 Xiaomi-11i
[... 297 lines cut ...]

```
ym2@rpi2:~/h6/esp_hosted_fg/host/linux/host_control/c_support $ sudo ./test.out softap_start
Enable heartbeat with duration 20
Heartbeat operation successful
esp32 softAP started
ethap0 interface down
MAC address 10:97:bd:f0:e1:d5 set to ethap0 interface
ethap0 interface up

Requested operation complete
Sleeping for some time just to showcase heartbeat
2023-12-18 17:43:26 >  App EVENT: SoftAP mode: Disconnect MAC[be:bd:5d:41:8d:48]
2023-12-18 17:43:34 >  App EVENT: Heartbeat event [0]

2023-12-18 17:43:50 >  App EVENT: SoftAP mode: Disconnect MAC[be:bd:5d:41:8d:48]
2023-12-18 17:43:51 >  App EVENT: SoftAP mode: Disconnect MAC[be:bd:5d:41:8d:48]
2023-12-18 17:43:54 >  App EVENT: Heartbeat event [1]
2023-12-18 17:43:55 >  App EVENT: SoftAP mode: Disconnect MAC[be:bd:5d:41:8d:48]
2023-12-18 17:44:03 >  App EVENT: SoftAP mode: Disconnect MAC[be:bd:5d:41:8d:48]
Disable Heartbeat
Heartbeat operation successful
ym2@rpi2:~/h6/esp_hosted_fg/host/linux/host_control/c_support $

```

mantriyogesh, 2023-12-18:
There might be some race condition in the code.
Give me some time to figure out.

mantriyogesh, 2023-12-20:
Over base current master, https://github.com/espressif/esp-hosted/commit/301e94a179c3a9a33d2cb90f90213b34e5e9dd4a , Can you please apply patch [GH_278.patch](https://github.com/espressif/esp-hosted/files/13722638/GH_278.patch) and re-flash ESP?