espressif/esp-hosted #280: any advice for migrating esp_wifi_init from wifi_drv.c > wifi_init.c for non esp32 host

claydonkey, 2023-11-17:
@mantriyogesh 
I'm sorry this is late in the week...
So, the features branch is compiling for stm32 but the hook esp_wifi_init is left as NULL
In the previous versions there was the wifi_drv.c file which called the test_wifi_..... functions to rpc's in addition to creating the esp_hosted_up_sem.

I understand on updating the features branch much of the component code was offloaded to the esp-idf functions but how does one hook another host into the process chain? rpc_api.c has disappeared and I am a bit lost as to how to proceed.

Before:
```C
void esp_hosted_up_cb(void)
{
	if (esp_hosted_up_sem)
		g_h.funcs->_h_post_semaphore_from_isr(esp_hosted_up_sem);
}

esp_err_t esp_wifi_init(const wifi_init_config_t *config)
{
	ESP_LOGV(TAG, "Trigger Wifi init");
    esp_hosted_up_sem = g_h.funcs->_h_create_semaphore(1);
	if (!esp_hosted_up_sem)
		printf("Unable to create sem %s:%u\n", __func__, __LINE__);
	g_h.funcs->_h_get_semaphore(esp_hosted_up_sem, portMAX_DELAY);

	ESP_ERROR_CHECK(esp_hosted_init(esp_hosted_up_cb));
	g_h.funcs->_h_get_semaphore(esp_hosted_up_sem, portMAX_DELAY);

#if 0//TEST_RAW_TP
	/* Just block the app main so that wifi will be stopped and only raw tp will be running */
	g_h.funcs->_h_get_semaphore(esp_hosted_up_sem, portMAX_DELAY);
#endif
	/*g_h.funcs->_h_get_semaphore(esp_hosted_up_sem, portMAX_DELAY); //Wait until slave is ready*/
	return test_wifi_init(config);
}

int test_wifi_init(const wifi_init_config_t *arg)
{
	/* implemented synchronous */
	ctrl_cmd_t req = RPC_DEFAULT_REQ();
	ctrl_cmd_t *resp = NULL;

	if (!arg)
		return FAILURE;

	g_h.funcs->_h_memcpy(&req.u.wifi_init_config, (void*)arg, sizeof(wifi_init_config_t));
	resp = wifi_init(req);

	return rpc_rsp_callback(resp);
}
```

closed 2023-11-20 by claydonkey

mantriyogesh, 2023-11-21:
yes, you can bypass esp-wifi-remote (and middle layer) if you think that is better.
Nevertheless, I took this as feedback and would improve the new branch for non esp MCU as well (with optional hooks wherever possible)..

claydonkey, 2023-11-22:
@mantriyogesh 
Thanks again for this.
As you can gather I have turned to the rpc version. It is nice and clean and I'm almost there...
Just the reset on the Slave... #283

mantriyogesh, 2023-12-20:
Lost the track on this. Anything blocked on @claydonkey ?