espressif/esp-hosted #281: Dror Gluska's STM32 LwIP port demo doesn't compile correctly

Tizio1234, 2023-11-18:
I'm getting some errors and warnings when trying to compile this [demo](https://github.com/drorgl/stm32-lwip-esp-hosted) mentioned [here](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/MCU_based_host/MCU_based_readme.md#113-things-worth-noting), here's the build log file:
[stm32-lwip.build.log](https://github.com/espressif/esp-hosted/files/13400778/stm32-lwip.build.log)
Any help? What am I doing wrong?

mantriyogesh, 2023-11-21:
Sorry, please check with implementer of this.
But I think you can cross check the lwipopts.h or lwip port layer.

alternatively can use https://github.com/espressif/esp-hosted/issues/186#issuecomment-1639235847
or 
https://github.com/espressif/esp-hosted/issues/270#issuecomment-1803576426 -> [2020.10.31__gh270_interim_branch_DoNOTUseInProduction.tgz](https://github.com/espressif/esp-hosted/files/13217485/2020.10.31__gh270_interim_branch_DoNOTUseInProduction.tgz)

claydonkey, 2023-11-26:
@Tizio1234 
I'm sorry it's taken so long to reply. I had some success with the STM32 and finally got it up and running @mantriyogesh 's features branch.
Please try [vectrx2020_F103_esp32_sdio](https://github.com/claydonkey/vectrx2020_F103_esp32_sdio.git)
Good luck

closed 2024-02-20 by mantriyogesh