espressif/esp-hosted #285: Missing guide for building NG firmware

sergey-suloev, 2023-11-22:
There is missing guide how to build NG firmware in different modes.
I.e. want to build it for esp32s3 
1) BT only + SPI
2) BT+WiFi + SPI 
but missing info on how to change config accordingly .
Could you update documentation, please?

mantriyogesh, 2023-11-22:
Hello @sergey-suloev ,

Thank you for your interest, but we could not understand clearly.

There are the following transport options available for ESP32-S3:
[image]

The hardware and software configurations to be done are described as part of https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng#2-hardware-and-software-setup

You can use
Bluetooth OR
Wi-Fi OR
Bluetooth + Wi-Fi at same time

Anything specific problem that is unclear from https://github.com/espressif/esp-hosted/tree/master/esp_hosted_ng ?
The documentation is in place, but if you feel something could be better or incomplete, we can definitely try to improve and simplify.

sergey-suloev, 2023-11-22:
@mantriyogesh  Yes, but in order to build firmware I need to have a correct config file in here esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter.  A config file with default options is created by "idf.py menuconfig" .  But the question is: will the default firmware work correctly in both scenarios ?

mantriyogesh, 2023-11-23:
So ESP-Hosted is targetted to by default provide the Wi-Fi and Bluetooth by default.
Wi-Fi & Bluetooth, both always considered as core features.
Rest other ESP-Hosted features are all configurable as user may want them.
I agree, that even Wi-Fi or Bluetooth can selectively enable could be better to control. We will think of adding those in future.

However, if you do not do Wi-Fi init & connect, the Wi-Fi would be effectively disabled.
If you need only Bluetooth, you can even do so, without changing anything.

sergey-suloev, 2023-11-23:
@mantriyogesh ok, clear, thanks

closed 2023-11-23 by sergey-suloev