espressif/esp-hosted #286: wifi disconnect detect issue

IOTBaicheng, 2023-11-22:
Hello,

When we disable the wifi AP, it seems that WPA can't get the disconnect event, and can not reconnect after enable the AP again.
We get  the following log info:

[image]

I found a similar issue in the github project:[The wifi will not auto reconnect after event espsta0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD · Issue #237 · espressif/esp-hosted · GitHub](https://github.com/espressif/esp-hosted/issues/237) 
Coud I fix my issue follow the patch in is link?

mantriyogesh, 2023-11-22:
@kapilkedawat

kapilkedawat, 2023-11-22:
Hi @IOTBaicheng, thanks for pointing out, we somehow missed to commit this patch, we will push it to repo.

Yes, this will fix your issue.

IOTBaicheng, 2023-11-23:
[image]
An error reported after patched.

IOTBaicheng, 2023-11-24:
> [image] An error reported after patched.

@kapilkedawat @mantriyogesh is any posible solution?

mantriyogesh, 2023-11-24:
sure, will check on this.

mantriyogesh, 2023-11-24:
can you please send `uname -a` output?

mantriyogesh, 2023-11-24:
Kapil is not available today,

I suppose your linux kernel is  < 5.9 ?
https://elixir.bootlin.com/linux/v5.9/source/net/wireless/scan.c#L715

I tried to port it to older kernel, but facing some build issue for it.
We will be able to get you this fine on Monday..

IOTBaicheng, 2023-11-25:
> Kapil is not available today,
> 
> I suppose your linux kernel is < 5.9 ? https://elixir.bootlin.com/linux/v5.9/source/net/wireless/scan.c#L715
> 
> I tried to port it to older kernel, but facing some build issue for it. We will be able to get you this fine on Monday..

uname -a :
Linux (none) 4.9.84 #73 SMP PREEMPT Tue Nov 14 14:28:53 CST 2023 armv7l GNU/Linux

mantriyogesh, 2023-11-27:
@kapilkedawat

IOTBaicheng, 2023-11-28:
> Kapil is not available today,
> 
> I suppose your linux kernel is < 5.9 ? https://elixir.bootlin.com/linux/v5.9/source/net/wireless/scan.c#L715
> 
> I tried to port it to older kernel, but facing some build issue for it. We will be able to get you this fine on Monday..

Are there any updates?

mantriyogesh, 2023-11-28:
@Shreyas0-7 ?

shreyassheth1, 2023-11-28:
Hey @IOTBaicheng 
Can you please try with this patch?
[cfg80211_bss_flush_unavailable_3.zip](https://github.com/espressif/esp-hosted/files/13488259/cfg80211_bss_flush_unavailable_3.zip)

IOTBaicheng, 2023-11-30:
> Hey @IOTBaicheng Can you please try with this patch? [cfg80211_bss_flush_unavailable_3.zip](https://github.com/espressif/esp-hosted/files/13488259/cfg80211_bss_flush_unavailable_3.zip)

Yes，It works OK. thanks