espressif/esp-hosted #290: esp32c2 for esp-hosted ng,connect ap sucessful then Automatic disconnect from ap

WindWard-atm, 2023-11-29:
[esp32c2 side.log](https://github.com/espressif/esp-hosted/files/13500020/esp32c2.side.log)
esp host on linux platform,esp32c2 as slave, after esp connect to ap ,it disconnect itself immediately

host log:
[    7.494642] Received ESP bootup event
[    7.495033] EVENT: 3
[    7.495190] EVENT: 2
[    7.495343] EVENT: 0
[    7.495491] EVENT: 4
[    7.495639] EVENT: 1
[    7.495803] ESP chipset's last reset cause:
[    7.496103] POWERON_RESET
[    7.496286] esp32: ESP Firmware version: 1.0.2
[    7.496582] ESP chipset detected [esp32-c2]
[    7.497109] 630 esp_wifi_device not initialized yet 
[    7.625368] ESP peripheral capabilities: 0xe8
[    7.831050] ESP Bluetooth init
[    7.831520] Capabilities: 0xe8. Features supported are:
[    7.831857]   * WLAN on SPI
[    7.832071]   * BT/BLE
[    7.832240]     - HCI over SPI
[    7.832453]     - BLE only
[   12.448584] ------------mf pm irq 73
[   12.453753] ------------mf pm irq 73
[   22.712603] adbd (953): /proc/953/oom_adj is deprecated, please use /proc/953/oom_score_adj instead.
[   47.475641] Authentication request: 8a:af:8f:ff:50:69 6 0 0 0
[   48.261120] Association request: 8a:af:8f:ff:50:69 6 27
[   48.432926] Connection status: 0
[   48.433789] Rx PACKET_TYPE_EAPOL!!!!
[   58.464336] Disconnect event for ssid skykeen [reason:204]
[   58.610579] cfg80211 regulatory domain callback for 00, current=
[   70.346350] Authentication request: 8a:af:8f:ff:50:69 6 0 0 0
[   71.119322] Association request: 8a:af:8f:ff:50:69 6 27
[   71.356785] Connection status: 0
[   71.489522] Disconnect event for ssid skykeen [reason:7]
[   71.609476] cfg80211 regulatory domain callback for 00, current=00
[  106.346252] Authentication request: 8a:af:8f:ff:50:69 6 0 0 0
[  107.136149] Association request: 8a:af:8f:ff:50:69 6 27
[  107.275427] Connection status: 0
[  107.396452] Disconnect event for ssid skykeen [reason:7]
[  107.516405] cfg80211 regulatory domain callback for 00, current=00
[  166.223714] Authentication request: 8a:af:8f:ff:50:69 6 0 0 0
[  167.013828] Association request: 8aaf:8f:ff:50:69 6 27
[  167.131681] Connection status: 0
[  167.132580] Rx PACKET_TYPE_EAPOL!!!!
[  167.132872] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[  167.181338] Rx PACKET_TYPE_EAPOL!!!!
[  167.181619] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!

shreyassheth1, 2023-11-29:
Hey @WindWard-atm 
Going through logs it seems there is no successful connection between ap and station, to debug this further can you please attach the following logs and sniffer capture?
1] Please enable the verbose logs in ESP 
- $ cd esp/esp_driver/network_adapter (directory where you build the firmware)
- $ idf.py menuconfig # Change as below
[image]
[image]
- exit, save and
- idf.py flash monitor

2] Capture sniffer.
- You will need another network interface to capture sniffer, lets say you another network interface is wlan0
- ifconfig wlan0 down
- iwconfig wlan0 mode monitor
- ifconfig wlan0 up
- iwconfig wlan0 channel 3
- wireshark

3] Please run wpa_supplicant command with **-dd** parameter to get the required debug logs

WindWard-atm, 2023-11-30:
hi 
    sorry, We have no other network interface to capture sniffer,There is no condition to do what you say
do you have any other easier method to do it?

***@***.***
 
From: shreyassheth1
Date: 2023-11-30 03:02
To: espressif/esp-hosted
CC: WindWard-atm; Mention
Subject: Re: [espressif/esp-hosted] esp32c2 for esp-hosted ng,connect ap sucessful then Automatic disconnect from ap (Issue #290)
Hey @WindWard-atm
Going through logs it seems there is no successful connection between ap and station, to debug this further can you please attach the following logs and sniffer capture?
1] Please enable the verbose logs in ESP
$ cd esp/esp_driver/network_adapter (directory where you build the firmware)
$ idf.py menuconfig # Change as below
image.png (view on web)
image.png (view on web)
exit, save and
idf.py flash monitor
2] Capture sniffer.
You will need another network interface to capture sniffer, lets say you another network interface is wlan0
ifconfig wlan0 down
iwconfig wlan0 mode monitor
ifconfig wlan0 up
iwconfig wlan0 channel 3
wireshark
3] Please run wpa_supplicant command with -dd parameter to get the required debug logs

mantriyogesh, 2023-11-30:
Generally, all desktop/laptops by default support sniffing.
You might have to check depending upon OS, distribution, how to start sniffer.

Just make sure that you start the sniffer before connection attempt.
So that the all (M1 to M4) encryption setup (EAPOL) frames are captured.
If any frame in M1 to M4 is missed, the wireshark capture is not fully decryptable.

I think the command wpa_supplicant with debug enabled (check earlier post by Shreyas), should also show much details, why the SSID was discarded.
If scenario is not straight forward (from ESP logs and wpa_supplicant logs), sniffer logs would be must.