espressif/esp-hosted #291: esp_sdio驱动加载失败，

liuyuaaa, 2023-11-30:
[image]
[image]
使用树莓派3b与esp32相连接，./rpi_init.sh sdio执行后，未见到主设备号为221的字符设备生成
[image]

mantriyogesh, 2023-11-30:
Hello @liuyuaaa 

Thank you for the logs and setup image.
I think the wires are very long still.

Can you please get ESP logs, Host logs (dmesg, rpi_init.sh command run and output) in textual form?
Can you please also try 1bit sdio mode : https://github.com/espressif/esp-hosted/issues/269#issuecomment-1780697351, which should face less signal integrity issues?

liuyuaaa, 2023-11-30:
这是ESP32的logs, 还有Host的logs (dmesg, rpi_init.sh）
[dmesg.log](https://github.com/espressif/esp-hosted/files/13507673/dmesg.log)
[ESP.log](https://github.com/espressif/esp-hosted/files/13507674/ESP.log)
[rpi_init.sh.log](https://github.com/espressif/esp-hosted/files/13507675/rpi_init.sh.log)

liuyuaaa, 2023-11-30:
我尝试了1bit sdio模式，在树莓派/boot/config.txt文件添加了dtoverlay=sdio,poll_once=off,bus_width=1，现象是一样的，还需要修改其他的吗

mantriyogesh, 2023-11-30:
I assume this is 4 bit mode logs you had shared.
Checked the logs. Logs as such are clean.

The jumper wire length is very big, need to make them smaller, else you would face such issues.

[image]

liuyuaaa, 2023-11-30:
我们尝试了以下几点：
1，我们又重新飞线，缩短至3cm，上拉电阻为20k
2，Linux host 版本切换到了0.0.5
3，sdio的时钟频率已降到了5MHz

 rpi_init.sh执行后，结果没有变化。

liuyuaaa, 2023-11-30:
我们测量发现data2管脚（esp32 gpio12）的电压为2.4v，其他data管脚为3.3v，这一点有没有影响。
[image]
是不是因为内部下拉影响了这个管脚

mantriyogesh, 2023-11-30:
Worth checking:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html

liuyuaaa, 2023-11-30:
感谢，问题解决了。
我们测试了sdio的1 bit mode，ESP32是可以正常工作的。这样可以看出data2和data3出问题了，后来发现data2管脚接错了，重新焊接测试下就行了。

mantriyogesh, 2023-12-18:
@liuyuaaa Can we close this issue?

closed 2023-12-19 by mantriyogesh