espressif/esp-hosted #292: RF testing and upgrade

linchanghe123, 2023-11-30:
Hello @mantriyogesh

Is there currently a method for RF testing using hosted_ng mode? In addition, does the current code include an upgrade channel for esp32c3 firmware?

thanks

mantriyogesh, 2023-11-30:
Hello @linchanghe123 ,

The ESP-Hosted-NG relies in ESP-IDF Wi-Fi and RF.
You can use the ESP-IDF independent example for RF at: https://github.com/espressif/esp-idf/tree/release/v5.1/examples/phy/cert_test

> In addition, does the current code include an upgrade channel for esp32c3 firmware?
I did not understand, do you mean OTA update?

linchanghe123, 2023-11-30:
Hello @mantriyogesh，

[image]
We want to upgrade the firmware of esp32c3 through SPI，not OTA. Is there any relevant mechanism currently?

mantriyogesh, 2023-11-30:
That is what we say OTA.
Firmware upgrade over ESP-Hosted transport is supported in ESP-Hosted-FG. but NG doesn't have that capability yet.

We are planning to add such support, but may take some time..

linchanghe123, 2023-12-25:
Hello @mantriyogesh，
When is the NG mode expected to support firmware upgrade functionality? Can we also port the upgrade process of FG to NG? Is this approach feasible and does it require a significant amount of work?

mantriyogesh, 2023-12-26:
FG actually works over serial interface and handles all the host requests as RPC over this serial (character) driver.

Such character driver is not present by design in NG. NG might take little different approach, still unclear. If nothing else effective or easy, we will fall back to FG way of OTA for NG.

If you need urgently, you can try FG upgrade and satisfy you, you can add youself, as NG upgrade may take some time.
If you thing direct serial flashing (not OTA) works by the time, You can also stick with that for now.

NG OTA will definitely be supported, but might take considerable time.