espressif/esp-hosted #293: esp32c6 linux driver

HanJiec, 2023-12-01:
esp32c6 as slave ,  mcu run linux kernel 5.10 , using esp-hosted-fg as codebase

HanJiec, 2023-12-01:
Dear Sir,
When I compile host driver ， Need to distinguish between the esp32 and the esp32c6 ？
When I compile firmware in esp , how to  config it for ESP32C6?

mantriyogesh, 2023-12-01:
Hello @HanJiec ,

Assuming that you are using SPI over FG Linux.
You can differentiate using the firmware_chip_id sent from esp to host.
https://github.com/espressif/esp-hosted/blob/ce3c50a33fa4bc562a1b6cbcee292c1ae0b0a404/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L208-L217

Here, if you wish to print which ESP chipset being used, can add code like:

```c
		printk(KERN_INFO "ESP chipset detected [%s]\n",
				hardware_type == ESP_FIRMWARE_CHIP_ESP32 ? "esp32" :
				hardware_type == ESP_FIRMWARE_CHIP_ESP32S2 ? "esp32-s2" :
				hardware_type == ESP_FIRMWARE_CHIP_ESP32C3 ? "esp32-c3" :
				hardware_type == ESP_FIRMWARE_CHIP_ESP32C2 ? "esp32-c2" :
				hardware_type == ESP_FIRMWARE_CHIP_ESP32S3 ? "esp32-s3" :
				"unknown");
```

Host driver generally should work for all the ESP similar way.
For ESP firmware building from source, you can refer ESP-Hosted-FG [documentation](https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg). Once you follow, you should be landed on page like 
[2. ESP-Hosted Comprehensive Guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Linux_based_readme.md#2-esp-hosted-comprehensive-guide). Need to follow 2.2, 2.3, 2.4.
In 2.4, depending upon the transport (for example, SPI), you will see 'Source Compilation' step, where all supported chipsets are covered.

The document is structured the way that you will be pointed to specific links as you go deep for the specific transport (for example, SPI)

HanJiec, 2023-12-07:
In my product ， need no BT. And no bluetooth.ko . How to remove BT in host driver ?

mantriyogesh, 2023-12-07:
yes. this is not yet resolved.
Selective bluetooth disabling, is not yet pushed.

For now, check https://github.com/espressif/esp-hosted/issues/150#issuecomment-1309860142

HanJiec, 2023-12-12:
esp stops at somewhere.

ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x4b (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd48
load:0x4086e610,len:0x2d64
load:0x40875720,len:0x17e8
entry 0x4086c410
I (23) boot: ESP-IDF v5.1 2nd stage bootloader
I (24) boot: compile time Dec  8 2023 15:08:58
I (24) boot: chip revision: v0.0
I (26) boot.esp32c6: SPI Speed      : 40MHz
I (31) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (64) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (72) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  3 factory          factory app      00 00 00010000 00100000
I (87) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (94) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (102) boot: End of partition table
I (106) boot: Defaulting to factory image
I (111) esp_image: segment 0: paddr=00010020 vaddr=420b0020 size=27e70h (163440) map
I (152) esp_image: segment 1: paddr=00037e98 vaddr=40800000 size=08180h ( 33152) load
I (161) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=a1dfch (663036) map
I (296) esp_image: segment 3: paddr=000e1e24 vaddr=40808180 size=0c894h ( 51348) load
I (309) esp_image: segment 4: paddr=000ee6c0 vaddr=40814a20 size=0334ch ( 13132) load
I (317) boot: Loaded app from partition at offset 0x10000
I (318) boot: Disabling RNG early entropy source...
I (329) cpu_start: Unicore app
I (329) cpu_start: Pro cpu up.
W (338) clk: esp_perip_clk_init() has not been implemented yet
I (345) cpu_start: Pro cpu start user code
I (345) cpu_start: cpu freq: 160000000 Hz
I (345) cpu_start: Application information:
I (348) cpu_start: Project name:     network_adapter
I (354) cpu_start: App version:      release/ng-v1.0.2-106-g48405288
I (361) cpu_start: Compile time:     Dec  8 2023 15:11:23
I (367) cpu_start: ELF file SHA256:  7be87e9a17dc6329...
I (373) cpu_start: ESP-IDF:          v5.1
I (377) cpu_start: Min chip rev:     v0.0
I (382) cpu_start: Max chip rev:     v0.99
I (387) cpu_start: Chip rev:         v0.0
I (392) heap_init: Initializing. RAM available for dynamic allocation:
I (399) heap_init: At 40821030 len 0005B5E0 (365 KiB): D/IRAM
I (405) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (412) heap_init: At 50000010 len 00003FF0 (15 KiB): RTCRAM
I (419) spi_flash: detected chip: generic
I (423) spi_flash: flash io: dio
I (427) sleep: Configure to isolate all GPIO pins in sleep state
I (434) sleep: Enable automatic switching of GPIO sleep configuration
I (441) coexist: coex firmware version: ebddf30
I (446) coexist: coexist rom version 5b8dcfa
I (451) app_start: Starting scheduler on CPU0
I (456) main_task: Started on CPU0
I (456) main_task: Calling app_main()
I (456) NETWORK_ADAPTER: *********************************************************************
I (466) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5
I (476) NETWORK_ADAPTER:                 Transport used :: SDIO only
I (486) NETWORK_ADAPTER: *********************************************************************
I (496) NETWORK_ADAPTER: Supported features are:
I (506) NETWORK_ADAPTER: - WLAN over SDIO
I (506) ESP_BT: - BT/BLE
I (506) ESP_BT:    - HCI Over SDIO
I (516) ESP_BT:    - BLE only
I (516) NETWORK_ADAPTER: capabilities: 0xd
I (526) phy_init: phy_version 202,b4b3263,May 17 2023,20:14:14
controller lib commit: [5cacafa]
I (566) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 40:4c:ca:45:85:d2
I (566) SDIO_SLAVE: Using SDIO interface
I (576) SDIO_SLAVE: sdio_init: ESP32-C6 SDIO timing: 1

I (576) pp: pp rom version: 5b8dcfa
I (586) net80211: net80211 rom version: 5b8dcfa
I (596) wifi:wifi driver task: 408459d0, prio:23, stack:6656, core=0
I (596) wifi:wifi firmware version: b2f1f86
I (596) wifi:wifi certification version: v7.0
I (606) wifi:config NVS flash: disabled
I (606) wifi:config nano formating: disabled
I (606) wifi:mac_version:HAL_MAC_ESP32AX_761,ut_version:N
I (616) wifi:Init data frame dynamic rx buffer num: 32
I (616) wifi:Init management frame dynamic rx buffer num: 32
I (626) wifi:Init management short buffer num: 32
I (626) wifi:Init dynamic tx buffer num: 32
I (636) wifi:Init static tx FG buffer num: 2
I (636) wifi:Init static rx buffer size: 1700
I (646) wifi:Init static rx buffer num: 10
I (646) wifi:Init dynamic rx buffer num: 32
I (646) wifi_init: rx ba win: 6
I (656) wifi_init: tcpip mbox: 32
I (656) wifi_init: udp mbox: 6
I (666) wifi_init: tcp mbox: 6
I (666) wifi_init: tcp tx win: 5744
I (666) wifi_init: tcp rx win: 5744
I (676) wifi_init: tcp mss: 1440
I (676) wifi_init: WiFi IRAM OP enabled
I (686) wifi_init: WiFi RX IRAM OP enabled
W (686) esp_adapter: wifi_reset_mac_wrapper() has not been implemented yet
W (696) wifi:(bf)761:0x600a7cac:0x01b4b4b0
W (696) wifi:(agc)0x600a7128:0xd210a800, min.avgNF:0xce->0xd2(dB), RCalCount:0x10a, min.RRssi:0x800(-128.00)
W (706) wifi:(TB)WDEV_PWR_TB_MCS0:19
W (706) wifi:(TB)WDEV_PWR_TB_MCS1:19
W (716) wifi:(TB)WDEV_PWR_TB_MCS2:19
W (716) wifi:(TB)WDEV_PWR_TB_MCS3:19
W (716) wifi:(TB)WDEV_PWR_TB_MCS4:19
W (726) wifi:(TB)WDEV_PWR_TB_MCS5:19
W (726) wifi:(TB)WDEV_PWR_TB_MCS6:18
W (726) wifi:(TB)WDEV_PWR_TB_MCS7:18
W (736) wifi:(TB)WDEV_PWR_TB_MCS8:17
W (736) wifi:(TB)WDEV_PWR_TB_MCS9:15
W (736) wifi:(TB)WDEV_PWR_TB_MCS10:15
W (746) wifi:(TB)WDEV_PWR_TB_MCS11:15
I (746) wifi:11ax coex: WDEVAX_PTI0(0x55777555), WDEVAX_PTI1(0x00003377).

I (756) wifi:mode : null

mantriyogesh, 2023-12-12:
What was the scenario? 
Also provide logs:
1. ESP logs
2. logs at host
- dmesg from boot
- Commands run and their outputs
3. Base code `git commits` used at **ESP** and **Host**
4. Setup image
5. Confirm if you have followed the [porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md) for SDIO?

HanJiec, 2023-12-12:
1&2 host and esp log are mcukmsg.log and  esp32c6.log attached here.
[esp32c6.log](https://github.com/espressif/esp-hosted/files/13645775/esp32c6.log)
[mcukmsg.log](https://github.com/espressif/esp-hosted/files/13645776/mcukmsg.log)
3 host and esp Git commit is  4840528810457f393e0e65fe2bb1442dcb6dbc10 , while esp-idf commit is cbce221e88d52665523093b2b6dd0ebe3f1243f1 (HEAD, tag: v5.1)
4  cross compile host driver and the ko is packaged in kernel img , once mcu is booted , insmod esp32_sdio.ko is doned.
    In host driver main.c  I  have changed the code to : static int resetpin = 42;  
5 pull-ups is added. But sdio timing use default ，I have not try others

mantriyogesh, 2023-12-12:
I think you had done insmod directly on boot.
You need to use `rpi_init.sh` to do additional things than that just of loading module.

check 
https://github.com/espressif/esp-hosted/blob/ce3c50a33fa4bc562a1b6cbcee292c1ae0b0a404/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L100-L101

These lines 'SERIAL' driver is created, also other things being done in rpi_init.sh, need to be ported.
Check point (5) in https://github.com/espressif/esp-hosted/issues/293#issuecomment-1851230779

NG and FG running procedure differs, Please read the [documentation](https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg), how to load the kernel module.
We always try to make the documentation as small as possible, so that it should be easy to follow.

I also spot there are kernel warnings in your run, with: 
https://github.com/espressif/esp-hosted/blob/4840528810457f393e0e65fe2bb1442dcb6dbc10/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio_api.c#L80

I suggest to run ported rpi_init.sh in separate shell/terminal session first, get it running first and then hook rpi_init.sh with system startup.

HanJiec, 2023-12-13:
In [mcukmsg.log](https://github.com/espressif/esp-hosted/files/13645776/mcukmsg.log) , there are :
>> [sdmmc_0] Err: #Cmd_52 (0x00000C00)=>(E: 0x0008)(S: 0x00000000)__(L:1047)
>> [sdmmc_0] Err: #Cmd_52 (0x80000C08)=>(E: 0x0008)(S: 0x00000000)__(L:1047)
means sdio_reset fails , so sdio connect fails.
In esp32c6 firmware , which code process the Cmd 52 ?

Kernel code shows ：
/*
	 * sdio_reset sends CMD52 to reset card.  Since we do not know
	 * if the card is being re-initialized, just send it.  CMD52
	 * should be ignored by SD/eMMC cards.
	 * Skip it if we already know that we do not support SDIO commands
	 */
	if (!(host->caps2 & MMC_CAP2_NO_SDIO))
		sdio_reset(host);

mantriyogesh, 2023-12-14:
I understand that you use ESP32-C6 SDIO.

Yes, SDIO_ID looks correct for esp32-c6:
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio_decl.h#L81-L86

Have you selected correct `ESP_SLAVE_CHIPSET="esp32-c6"` in `rpi_init.sh` ?
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L23-L25

BTW, we use func1 only.

mantriyogesh, 2023-12-14:
I will ammend the troubleshoot guide with ESP32-C6, as you pointed.
But anyway, you can expect the 
`SDIO_ID=0092:6666` for ESP32-C6, which code already aware and should work.

Thank you for checking all the documents, btw.

Do you see CIS populated correctly in Linux dmesg?
Also, have you unhooked insmod from startup?
Also, can you please share setup image ?

HanJiec, 2023-12-14:
Are you Chinese? Are  you in China ？ Could we  talk in phone ？

HanJiec, 2023-12-14:
> I will ammend the troubleshoot guide with ESP32-C6, as you pointed. But anyway, you can expect the `SDIO_ID=0092:6666` for ESP32-C6, which code already aware and should work.
> 
> Thank you for checking all the documents, btw.
> 
> Do you see CIS populated correctly in Linux dmesg?
You means the following lines shows correct communication between esp and mcu.
mmc0: queuing unknown CIS tuple 0x81 (1 bytes)
mmc0: queuing unknown CIS tuple 0x82 (1 bytes)
mmc0: queuing unknown CIS tuple 0x80 (1 bytes)
mmc0: queuing unknown CIS tuple 0x81 (1 bytes)
mmc0: queuing unknown CIS tuple 0x82 (1 bytes)
mmc0: new SDIO card at address 0001

 Also, have you unhooked insmod from startup? Also, can you please share setup image ?

mantriyogesh, 2023-12-14:
Sorry we don't understand the mandarin. Not in China.
Phone calls will be difficult, I hope you understand.

You can get formal support from ESP support if need so.
But anyway, we will support the GitHub channel.

Can you please help answer above questions?
1. Have you removed the kernel module loading from system startup?
2. Can you share the picture of hardware connections for ESP and Host?
3. Commands output of :
$ uname -a
$ ls /dev/spidev*

If you wish to type in Chinese script,  it is okay. We will try to use translator to help you as much as possible, and respond in English.

What happens in dmesg when you run rpi_init.sh manually?

HanJiec, 2023-12-14:
1 Yes I have removed the kernel module loading from sysstem startup
   But question still exist. ko is crashed. log is attached
[mculog1214.txt](https://github.com/espressif/esp-hosted/files/13668368/mculog1214.txt)
2 wifi schematic is here :
[image]
3 /dev # ls /dev/spidev*
ls: /dev/spidev*: No such file or directory
/dev # uname -a
Linux (none) 5.10.61 #80 PREEMPT Thu Dec 14 09:38:38 CST 2023 armv7lNU/Linux
4  answers for What happens in dmesg when you run rpi_init.sh manually? is in item 1 answer .

HanJiec, 2023-12-14:
[mcukmsg.log](https://github.com/espressif/esp-hosted/files/13669148/mcukmsg.log)
This is more clear.  Boot ( without insmod esp32 ko ) , then tftp ko into /tmp , then run rpi_init.sh , and crash information dump to console . ls show /dev/esps0 is created successfully.

mantriyogesh, 2023-12-14:
Do you use PCB or jumper cables?

mantriyogesh, 2023-12-14:
Also, there are some bug fixes added after your current commit in sdio.
Can you please test using master, at both **ESP** and **Linux**?

HanJiec, 2023-12-15:
PCB,
Kindly please help locate the problem .  For example , tell me where in firmware catch event sent from host ?  SDIO_ID is from esp chip , or just a fake value . I can see same SDIO_ID without insmod esp32 host driver.

mantriyogesh, 2023-12-15:
Caution: Little bit technical stuff, just for understanding the flow ahead.

### Technical details 
- First, host should enter the [esp_probe()](https://github.com/espressif/esp-hosted/blob/ce3c50a33fa4bc562a1b6cbcee292c1ae0b0a404/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c#L693)
- Before it enters, if any other thread starts, will become a problem.
- [SDIO interrupt](https://github.com/espressif/esp-hosted/blob/ce3c50a33fa4bc562a1b6cbcee292c1ae0b0a404/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c#L109) should **not** be available or triggered before esp_probe() is successful.
- Unless probe is successful, host will not send the [ESP_OPEN_DATA_PATH](https://github.com/espressif/esp-hosted/blob/ce3c50a33fa4bc562a1b6cbcee292c1ae0b0a404/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c#L743) to ESP.
- Unless Firmware doesn't receive [ESP_OPEN_DATA_PATH](https://github.com/espressif/esp-hosted/blob/ce3c50a33fa4bc562a1b6cbcee292c1ae0b0a404/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c#L556), the ESP will not send the INIT event on ESP_PRIV_IF.
- you are receiving the INIT event from slave to host. As you are not able to see ["Start Data Path"](https://github.com/espressif/esp-hosted/blob/ce3c50a33fa4bc562a1b6cbcee292c1ae0b0a404/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c#L560) trace in ESP32-C6.

It's okay, you do not understand above details.

To keep it simple,
Can you please try the master branch on ESP and host and test again?
Some of the fixes in reader tasks to wait untill the sdio is not up, already present in master.

The difference between your commit and master commit are minimal, at both ESP and host, so it should be easy to move to master.
Also please make sure always to use exact same commit at ESP flashing and Linux host kernel driver building.

If you are not able to get it working on master, we will provide a debug patch with lot of debug logs, to understand the issue.

HanJiec, 2023-12-15:
For your suggestion "Can you please try the master branch on ESP and host " . I get code as red block in the following picture. Is it right ?
[image]

mantriyogesh, 2023-12-15:
yes. this is right.

For ESP-IDF, you can save all your changes. and then use the cmake script as shown in:
[image]

HanJiec, 2023-12-15:
You means:
In esp_hosted_fg/esp/esp_driver   cmake . will update esp-idf correctly ?

mantriyogesh, 2023-12-15:
Yes.

mantriyogesh, 2023-12-20:
Were you able to proceed ahead?

HanJiec, 2023-12-20:
I download firmware compiled from the newest code. But mcu cannot config the the pins ( 6 pins ) to sdio mode.  may Esp 's  pin enter incorrect state , so MCU cannot enter sdio mode , stay GPIO mode. Do you have any suggestions?

mantriyogesh, 2023-12-20:
1. As I remember, you were able to get the CIS populated, but the 'Start data path' was not seen on ESP32-C6 SDIO. 
If CIS is populated, that means the SDIO was detected.
- Have you checked the Technical Details in https://github.com/espressif/esp-hosted/issues/293#issuecomment-1857148359?
- Can you verify with ESP-Hosted-FG master, if you get the 'Start Data Path' print in ESP32-C6?
- 

2. Are you using Device Tree for SDIO in Linux? If so, can you post the Device Tree blob for your SDIO?
Try to use `broken-cd;` to try to probe in polling if anytime it failed to probe.

3. Have you gone through [Porting Guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md)?
- Pull-Up (Not able to download schematic, but I suppose you have added external pull-ups)
- PCB : I assume you have PCB created with above external pull-ups (> 10
- 

4. Can you please post the commands/output and configuration you had used?
We are trying to understand what exact phase the SDIO is not working for you.

5. Have you tried **one bit SDIO mode**?
For raspberry Pi, 1 bit mode can be configured as:
https://github.com/espressif/esp-hosted/issues/269#issuecomment-1780697351
For Linux with Device tree, may be configured as `bus-width` device tree param. equal to <1>
One bit mode should be the easiest to get the SDIO verified quickly.

6. Are you facing low voltage on any of the SDIO pins?

HanJiec, 2023-12-20:
using a new board ,  setting mcu sdio mode is OK . using the newest firmware and host driver ko.
after insmod ( using rpi_init.sh ) , there are following   err:
>> [sdmmc_0] Err: #Cmd_52 (0x80022000)=>(E: 0x0008)(S: 0x000010FF)__(L:1047)
esp_sdio: probe of mmc0:0001:1 failed with error 8
>> [sdmmc_0] Err: #Cmd_52 (0x80042000)=>(E: 0x0008)(S: 0x000010FF)__(L:1047)
esp_sdio: probe of mmc0:0001:2 failed with error 8
Please find detail log attached
[mcukmsg.log](https://github.com/espressif/esp-hosted/files/13722656/mcukmsg.log)
[esp32c6.log](https://github.com/espressif/esp-hosted/files/13722659/esp32c6.log)

If success , after rpi_init.sh , net cars espsta0 will list vi ifconfig -a ?

mantriyogesh, 2023-12-20:
Still the start data path is not set at esp. meaning register read not working yet from host.

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#242-sdio

Change to sdio timing help?

mantriyogesh, 2023-12-20:
Can you show your setup photo?

HanJiec, 2023-12-20:
> Still the start data path is not set at esp. meaning register read not working yet from host.
> 
> https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md#242-sdio
> 
> Change to sdio timing help?

Last month have tried each value of 4 , no successs. I would it on this version . I should only change it in Firmware in esp32c6 , right ?

HanJiec, 2023-12-20:
> Can you show your setup photo?

[image]

red words are esp chip and mcu chip

mantriyogesh, 2023-12-20:
ESP I can't see. Is it connected to MCU? By wires?

> I would it on this version 

You mean master?

>  I should only change it in Firmware in esp32c6 , right ?

yes. Generally one of PP or NP should work.

HanJiec, 2023-12-20:
> ESP I can't see. Is it connected to MCU? By wires?
> 
> > I would it on this version
> 
> You mean master?
> 
> > I should only change it in Firmware in esp32c6 , right ?
> 
[image]
MCU( run linux) and esp32  both are in the same pcb, i marked them red . can you see them?
> yes. Generally one of PP or NP should work.
When i tried timing = 2 (PN) , MCU sdio mode fail to set , enter gpio mode. It is very troublesome,

mantriyogesh, 2023-12-20:
Do you have logic analyzer?

Can you please capture what happens on the SDIO bus?

mantriyogesh, 2023-12-20:
Have you tested on master, BTW?

Many questions / alternatives discussed in prior posts.
Can you check them?

HanJiec, 2023-12-20:
> Have you tested on master, BTW?
> what means master ?
> Many questions / alternatives discussed in prior posts. Can you check them?

now my board cannot re-back to normal state?

HanJiec, 2023-12-20:
dead loop

HanJiec, 2023-12-20:
if you said only try NP and PP first , maybe can keep my board to workable state

mantriyogesh, 2023-12-20:
CIs was okay, which is typically read at 400KHz.
So can you try to uncomment:

https://github.com/espressif/esp-hosted/blob/301e94a179c3a9a33d2cb90f90213b34e5e9dd4a/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c#L714-L718
 and change frequency to 1 MHz?

mantriyogesh, 2023-12-20:
But unless you run rpi_init.sh, the esp32 module will not be usable.
So just loading/reloading esp32 module cannot put the Linux in bootloop.

Try 1 Bit SDIO (might need changes in Device Tree / Linux config) with lower freq like 1MHz.

HanJiec, 2023-12-20:
> CIs was okay, which is typically read at 400KHz. So can you try to uncomment:
> means " mmc0: queuing unknown CIS tuple 0x01 (3 bytes) " ? 3bytes is received from esp32c6 slave ?
> https://github.com/espressif/esp-hosted/blob/301e94a179c3a9a33d2cb90f90213b34e5e9dd4a/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c#L714-L718
> 
> 
> and change frequency to 1 MHz?

mantriyogesh, 2023-12-20:
Yes. You can refer the SDIO spec. But basically, this communication is also happening over SDIO register.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/protocols/esp_sdio_slave_protocol.html#esp-sdio-slave-initialization

This is enumeration from ESP32-C6 slave. CIS is successful.
You should try lower sdio clock.

HanJiec, 2023-12-20:
Just i try 1 Mhz clk and fail, can i set it to 400000 (400Khz) ?

mantriyogesh, 2023-12-20:
Yes. log exact same for 1MHz?

Can you please check one bit SDIO mode?

HanJiec, 2023-12-21:
> CIs was okay, which is typically read at 400KHz. So can you try to uncomment:
> 
> https://github.com/espressif/esp-hosted/blob/301e94a179c3a9a33d2cb90f90213b34e5e9dd4a/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c#L714-L718
> 
> 
> and change frequency to 1 MHz?

Sir , above #if 0, there is a line   printk(KERN_INFO "%s: ESP network device detected\n", __func__);
but "ESP network device detected" doesn't in mcu log . so uncomment  should not run . 
I think problen is not here.

mantriyogesh, 2023-12-21:
1. In your case, the probe is failing as:
```
esp_sdio: probe of mmc0:0001:1 failed with error 8
```
which is part of esp_probe. If you **test with master** and **share the expected logs**, we will get the place and reason of failure.
the error logs are added in master.

Is the [esp_probe()](https://github.com/espressif/esp-hosted/blob/301e94a179c3a9a33d2cb90f90213b34e5e9dd4a/esp_hosted_fg/host/linux/host_driver/esp32/sdio/esp_sdio.c#L693) being called?

2. Additionally, can you send the ESP & Linux side code diff?
3. you can add the logs manually in esp_probe() to understand where the failure is returned?
Why we are asking, **the return code '8' is not recognizable for esp_sdio module.**
- It could be because of your base Linux driver problem
- Could be configuration problem in Device Tree (something not supported by ESP chipset?)
- Physical connection issue : Incorrect GPIO lines connected / Voltage is low for some pins for SDIO operation

4. Post the Device Tree of your Linux SoC?

Now the questions are being repeated, I think.

mantriyogesh, 2023-12-21:
Check if your kernel allows the KERN_INFO logs: https://linuxconfig.org/introduction-to-the-linux-kernel-log-levels

HanJiec, 2023-12-21:
> If you **test with master**

Sorry , how to test with master , here what does master mean to ?

mantriyogesh, 2023-12-21:
https://github.com/espressif/esp-hosted/tree/master

HanJiec, 2023-12-21:
> For your suggestion "Can you please try the master branch on ESP and host " . I get code as red block in the following picture. Is it right ? ![git-master](https://private-user-images.githubusercontent.com/132651638/290716517-c635f4df-20cc-4d27-980f-448efcf7d94b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMxMjc4NzgsIm5iZiI6MTcwMzEyNzU3OCwicGF0aCI6Ii8xMzI2NTE2MzgvMjkwNzE2NTE3LWM2MzVmNGRmLTIwY2MtNGQyNy05ODBmLTQ0OGVmY2Y3ZDk0Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIyMVQwMjU5MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNjMxNjcyNmE3Yzc1MWFjNjQ4M2E1ZGFlMDczNjVmNDVhNmI1MGU1NWEyZTM2NTdjNzllMGEwNjNiOWE2ZDRkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Iett6JRLUNXj1kKkuIpeZ68uf7tX6qjx3y5qIfuT3sc) Could you please show me echo commit of the component ，such as fg esp-idf and so on

I get code  follow your answer , the code got using this method , is  master , right ?

HanJiec, 2023-12-21:
I modify esp_sdio.c   in esp_probe  start , printk(KERN_DEBUG "%s enter" , __func__ ) ; not in log 
in esp_init_interface_layer ,  	printk(KERN_INFO "%s enter" , __func__ ) ;
	printk(KERN_DEBUG "%s enter" , __func__ ) ;   2  line are in log
esp_init_interface_layer is called really ， seems esp_probe not be called.
please see  attached source modified and log
[mcukmsg.log](https://github.com/espressif/esp-hosted/files/13734848/mcukmsg.log)
[log.txt](https://github.com/espressif/esp-hosted/files/13734849/log.txt)

esp_sdio: probe of mmc0:0001:1 failed with error 8 can been seen.

mantriyogesh, 2023-12-21:
1. What are the command outputs for :

```sh
$ cat /sys/devices/platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1\:0001/mmc1\:0001\:1/uevent

SDIO_CLASS=00
SDIO_ID=0092:6666
MODALIAS=sdio:c00v0092d6666

$ sudo cat /sys/kernel/debug/mmc1/ios
```
Above command will not exactly match.
Please check your matching mmc instance (mmc0) and change path according to your linux and post output

2. what is the code diff at Linux? Source file without base cimit will be hard to understand the change.
run:

```
$ cd esp-hosted
$ git log --oneline |head -4
$ git status
$ git diff
```
and get the full textual output at both **ESP** and **Linux**

3. Debug patch
Please apply
[sdio_debug_gh293.patch](https://github.com/espressif/esp-hosted/files/13736724/sdio_debug_gh293.patch)
 over clean repo with current git master (301e94a179c3a9a33d2cb90f90213b34e5e9dd4a)

If don't want to use patch, use 
[esp_sdio.c.zip](https://github.com/espressif/esp-hosted/files/13736670/esp_sdio.c.zip)
[esp_sdio_api.c.zip](https://github.com/espressif/esp-hosted/files/13736672/esp_sdio_api.c.zip)
[esp_sdio_decl.h.zip](https://github.com/espressif/esp-hosted/files/13736704/esp_sdio_decl.h.zip)

Note:
a. Changes are **not compiled**. **Please correct if any issues in compilation** either side ESP or Host.

HanJiec, 2023-12-21:
For Esp_hosted_fg ：
h@hvB:~/esp_hosted_202312152053/esp-hosted

$  git log --oneline |head -4

301e94a Merge branch 'fix/build_mac2str_error' into 'master'

20af007 fix(esp_hosted_ng): Fix build error for MAC2STR

ce3c50a Merge branch 'fix/spi_sdio_uart_c2_documentation' into 'master'

c4843f1 fix(esp_hosted_ng): Fix error in esp32c2 spi/sdio + uart documentation

h@hvB:~/esp_hosted_202312152053/esp-hosted

$ git status

On branch master

Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
	modified:   esp_hosted_fg/esp/esp_driver/esp-idf (untracked content)
	modified:   esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c
	modified:   esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c
Untracked files:
  (use "git add <file>..." to include in what will be committed)
	esp-driver-cmake-dot.log
	esp_hosted_fg/esp/esp_driver/gitcloneHis.txt
	esp_hosted_fg/esp/esp_driver/network_adapter/cpbuild.sh
	esp_hosted_fg/host/gitcloneHis.txt
	gitcloneHis.txt

no changes added to commit (use "git add" and/or "git commit -a")
h@hvB:~/esp_hosted_202312152053/esp-hosted

$ git diff

diff --git a/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c b/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c

index aa44039..ac0923d 100644

--- a/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c

+++ b/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c

@@ -778,6 +778,7 @@ void app_main()

 

        while (!datapath) {

                sleep(1);

+               ESP_LOGI(TAG,"wait datapath true");

        }

 

        /* send capabilities to host */

diff --git a/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c b/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c

index 5cc3f61..688756c 100644

--- a/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c

+++ b/esp_hosted_fg/esp/esp_driver/network_adapter/main/sdio_slave_api.c

@@ -205,6 +205,13 @@ static interface_handle_t * sdio_init(void)

                 * */

 #if defined(CONFIG_IDF_TARGET_ESP32C6)

                .timing             = SDIO_SLAVE_TIMING_NSEND_PSAMPLE,

+

+/*  SDIO_SLAVE_TIMING_PSEND_PSAMPLE

+    SDIO_SLAVE_TIMING_NSEND_PSAMPLE

+    SDIO_SLAVE_TIMING_PSEND_NSAMPLE

+    SDIO_SLAVE_TIMING_NSEND_NSAMPLE  */

+

+

 #endif

        };

        config.flags |= SDIO_SLAVE_FLAG_DEFAULT_SPEED;

HanJiec, 2023-12-21:
> 1. What are the command outputs for :
> 
> ```shell
> $ cat /sys/devices/platform/soc/3f300000.mmc/mmc_host/mmc1/mmc1\:0001/mmc1\:0001\:1/uevent
> 
> SDIO_CLASS=00
> SDIO_ID=0092:6666
> MODALIAS=sdio:c00v0092d6666
> 
Answer：
/sys/devices/soc0/soc/soc:sdmmc/mmc_host/mmc0/mmc0:0001/mmc0:0001:1 # cat uevent
SDIO_CLASS=00
SDIO_ID=0092:6666
SDIO_REVISION=0.0
MODALIAS=sdio:c00v0092d6666

> $ sudo cat /sys/kernel/debug/mmc1/ios
Answer：
/sys/kernel/debug/mmc0 # cat ios
clock:          25000000 Hz
vdd:            20 (3.2 ~ 3.3 V)
bus mode:       2 (push-pull)
chip select:    0 (don't care)
power mode:     2 (on)
bus width:      2 (4 bits)
timing spec:    0 (legacy)
signal voltage: 0 (3.30 V)
driver type:    0 (driver type B)

> Above command will not exactly match. Please check your matching mmc instance (mmc0) and change path according to your linux and post output 2. what is the code diff at Linux? Source file without [...]

Linux source code is got by zip , no Git  repo
answer for esp_hosted_fg  has provided in last comment

> 
> 3. Debug patch
>    Please apply
>    [sdio_debug_gh293.patch](https://github.com/espressif/esp-hosted/files/13736724/sdio_debug_gh293.patch)
>    over clean repo with current git master ([301e94a](https://github.com/espressif/esp-hosted/commit/301e94a179c3a9a33d2cb90f90213b34e5e9dd4a))
>    
>    If don't want to use patch, use
>    [esp_sdio.c.zip](https://github.com/espressif/esp-hosted/files/13736670/esp_sdio.c.zip)
>    [esp_sdio_api.c.zip](https://github.com/espressif/esp-hosted/files/13736672/esp_sdio_api.c.zip)
>    [esp_sdio_decl.h.zip](https://github.com/espressif/esp-hosted/files/13736704/esp_sdio_decl.h.zip)
>    
>    Note:
>    a. Changes are **not compiled**. **Please correct if any issues in compilation** either side ESP or Host.

mantriyogesh, 2023-12-21:
In Raspberry Pi, I could see output as:

root@rpi-zhf:/home/pi# cat /sys/kernel/debug/mmc1/ios
clock:        25000000 Hz
actual clock:    25000000 Hz
**vdd:        21 (3.3 ~ 3.4 V)**
bus mode:    2 (push-pull)
chip select:    0 (don't care)
power mode:    2 (on)
bus width:    2 (4 bits)
timing spec:    0 (legacy)
signal voltage:    0 (3.30 V)
driver type:    0 (driver type B)

> Linux source code is got by zip , no Git repo

While debugging like this, git repo will be useful to understand nay difference. Otherwise, if something is changed or wrong, will not be able to know diff/change as no tracking information.

mantriyogesh, 2023-12-21:
1. https://github.com/espressif/esp-hosted/issues/293#issuecomment-1865636657 has debug patch, can be applied easilt if you have git repo. anyway the Documentation always will ask to clone, not to download zip.
Apply this code and find where you have reached and failure.
Although I strongly suggest to check your Linux Driver for SDIO / SDHOST / SDHCI
error code '8' is not from esp_hosted.

2. SPI solution is easier to implement. Can thing of this altrenative

3. Try with some other Linux. To test the solution, can use Raspberry Pi also.

4. Ship the board with necessary linux setup details, we will try to check the issue, but may take some time to ship & debug the issue

mantriyogesh, 2023-12-21:
Have you checked the voltage at the pins?

Have you tried one bit sdio mode?

HanJiec, 2023-12-21:
> Have you checked the voltage at the pins?
> 
> Have you tried one bit sdio mode?

esp  power in my board is 3.293 v , so 3.2-3.3v is right ?

HanJiec, 2023-12-21:
> 1. [esp32c6 linux driver ，via SDIO transport #293 (comment)](https://github.com/espressif/esp-hosted/issues/293#issuecomment-1865636657) has debug patch, can be applied easily if you have git repo. anyway, the Documentation will always ask to _git clone_, not to download zip.
>    Apply this code patch and find where you have reached in code and where the failure starts.
> 
> Although I strongly suggest checking your Linux Driver for SDIO / SDHOST / SDHCI.
can I send the source code to you?
> 
> **error code '8' is not from esp_hosted.**
> 
> 2. SPI solution is easier to implement. Can think of this as an alternative.
> 3. Try with some other Linux. To test the solution, can also use Raspberry Pi.
> 4. Ship the board with necessary Linux setup details, we will try to check the issue, but may take some time to ship & debug the issue

[34 more replies]