espressif/esp-hosted #296: It does not work !

AyoubCoding21, 2023-12-08:
I did the installation steps and flashed the esp_hosted_ng to my ESP32 (not C2 not S2) but it did not got recongised by my system or able to do ```sudo iw dev esp32sta0 scan``` I don't know what to do (Bad luck perhaps ?)

AyoubCoding21, 2023-12-08:
2: esp-log from main task:
mI (612) main_task: Started on CPU0
I (620) main_task: Calling app_main()
I (622) stats: *********************************************************************
I (628) stats:                 ESP-Hosted Firmware version :: 1.0.2                        
I (636) stats:                 Transport used :: SDIO only                     
I (642) stats: *********************************************************************
I (648) FW_MAIN: Supported features are:
I (652) FW_MAIN: - WLAN over SDIO
I (656) FW_BT: - BT/BLE
I (658) FW_BT:    - HCI Over SDIO
I (662) FW_BT:    - BT/BLE dual mode
I (664) FW_MAIN: Capabilities: 0x1d
I (678) wifi:wifi driver task: 3ffbd8b4, prio:23, stack:6656, core=1
I (680) wifi:wifi firmware version: fddc5e5
I (680) wifi:wifi certification version: v7.0
I (682) wifi:config NVS flash: disabled
I (684) wifi:config nano formating: disabled
I (688) wifi:Init data frame dynamic rx buffer num: 32
I (692) wifi:Init management frame dynamic rx buffer num: 32
I (696) wifi:Init management short buffer num: 32
I (698) wifi:Init dynamic tx buffer num: 64
I (702) wifi:Init static rx buffer size: 1600
I (706) wifi:Init static rx buffer num: 16
I (708) wifi:Init dynamic rx buffer num: 32
I (712) wifi_init: rx ba win: 6
I (714) wifi_init: tcpip mbox: 32
I (718) wifi_init: udp mbox: 6
I (720) wifi_init: tcp mbox: 6
I (724) wifi_init: tcp tx win: 5744
I (728) wifi_init: tcp rx win: 5744
I (730) wifi_init: tcp mss: 1440
I (734) wifi_init: WiFi IRAM OP enabled
I (738) wifi_init: WiFi RX IRAM OP enabled
I (742) BTDM_INIT: BT controller compile version [946b762]
I (746) BTDM_INIT: Bluetooth MAC: 34:86:5d:fc:10:42
I (752) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1070) FW_MAIN: ESP Bluetooth MAC addr: 34-86-5d-fc-10-42

I (1070) FW_SDIO_SLAVE: Using SDIO interface
I (1072) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0

AyoubCoding21, 2023-12-08:
esp-idf version (used with esp/esp-driver) : v5.1
PS : i haven't used host i went to esp/esp-driver ran ```cmake .``` then to esp-idf ```. ./export.sh``` then i go to network-adapter run ```idf.py -p [port] flash monitor``` it builds all and runs all

AyoubCoding21, 2023-12-08:
here is the output of when the flash is ongoing :
esptool.py v4.7.dev3
Serial port /dev/ttyUSB0
Connecting........................
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 34:86:5d:fc:10:40
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x000e0fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000d000 to 0x0000efff...
Compressed 26656 bytes to 16691...
Writing at 0x00001000... (50 %)
Writing at 0x0000768e... (100 %)
Wrote 26656 bytes (16691 compressed) at 0x00001000 in 0.7 seconds (effective 313.6 kbit/s)...
Hash of data verified.
Compressed 854144 bytes to 547147...
Writing at 0x00010000... (2 %)
Writing at 0x0001c22b... (5 %)
Writing at 0x0002466b... (8 %)
Writing at 0x00029b9d... (11 %)
Writing at 0x0003519d... (14 %)
Writing at 0x0003b011... (17 %)
Writing at 0x00040804... (20 %)
Writing at 0x00046163... (23 %)
Writing at 0x0004beb1... (26 %)
Writing at 0x00051cdb... (29 %)
Writing at 0x0005808f... (32 %)
Writing at 0x0005ddad... (35 %)
Writing at 0x00063aca... (38 %)
Writing at 0x000696a5... (41 %)
Writing at 0x0006f421... (44 %)
Writing at 0x00074d7f... (47 %)
Writing at 0x00079fce... (50 %)
Writing at 0x0007f513... (52 %)
Writing at 0x00084749... (55 %)
Writing at 0x00089ecd... (58 %)
Writing at 0x0008f42e... (61 %)
Writing at 0x00094b10... (64 %)
Writing at 0x0009a758... (67 %)
Writing at 0x0009ff32... (70 %)
Writing at 0x000a5704... (73 %)
Writing at 0x000ab398... (76 %)
Writing at 0x000b0d17... (79 %)
Writing at 0x000b7332... (82 %)
Writing at 0x000bf468... (85 %)
Writing at 0x000c78ad... (88 %)
Writing at 0x000cd971... (91 %)
Writing at 0x000d3488... (94 %)
Writing at 0x000d9011... (97 %)
Writing at 0x000de27a... (100 %)
Wrote 854144 bytes (547147 compressed) at 0x00010000 in 12.4 seconds (effective 551.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 138...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (138 compressed) at 0x00008000 in 0.0 seconds (effective 545.9 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Writing at 0x0000d000... (100 %)
Wrote 8192 bytes (31 compressed) at 0x0000d000 in 0.1 seconds (effective 949.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

AyoubCoding21, 2023-12-08:
dmesg output :
[  758.511461] perf: interrupt took too long (2503 > 2500), lowering kernel.perf_event_max_sample_rate to 79750
[ 1430.092316] perf: interrupt took too long (3131 > 3128), lowering kernel.perf_event_max_sample_rate to 63750
[ 4018.091156] usb 6-2: USB disconnect, device number 4
[ 4018.091477] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[ 4018.091505] cp210x 6-2:1.0: device disconnected
[ 4021.007154] usb 6-2: new full-speed USB device number 5 using uhci_hcd
[ 4021.219555] usb 6-2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 4021.219562] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4021.219565] usb 6-2: Product: CP2102 USB to UART Bridge Controller
[ 4021.219568] usb 6-2: Manufacturer: Silicon Labs
[ 4021.219570] usb 6-2: SerialNumber: 0001
[ 4021.232589] cp210x 6-2:1.0: cp210x converter detected
[ 4021.234653] usb 6-2: cp210x converter now attached to ttyUSB0
[ 5039.558538] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[ 5039.559235] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'

AyoubCoding21, 2023-12-08:
that is all that i was able to get @mantriyogesh

AyoubCoding21, 2023-12-08:
ps i didn't use rpis
i have a computer
and i connected esp32 by usb
specs :
core 2 duo e8400
2 gb ram
no gpu (intel q45)

AyoubCoding21, 2023-12-08:
ok
[    0.000000] microcode: microcode updated early to revision 0xa0b, date = 2010-09-28
[    0.000000] Linux version 6.2.0-37-generic (buildd@bos03-amd64-055) (x86_64-linux-gnu-gcc-11 (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #38~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov  2 18:01:13 UTC 2 (Ubuntu 6.2.0-37.38~22.04.1-generic 6.2.16)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-6.2.0-37-generic root=UUID=8159721d-3e00-4ec7-91bc-dfde7f05d4c4 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Hygon HygonGenuine
[    0.000000]   Centaur CentaurHauls
[    0.000000]   zhaoxin   Shanghai  
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009f7ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f800-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e8000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007b9a163f] usable
[    0.000000] BIOS-e820: [mem 0x000000007b9a1640-0x000000007b9a369f] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007b9a36a0-0x000000007dffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f4000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fed3ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed45000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.6 present.
[    0.000000] DMI: Hewlett-Packard HP Compaq 8000 Elite SFF PC/3646h, BIOS 786G7 v01.02 10/22/2009
[    0.000000] tsc: Fast TSC calibration failed
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] last_pfn = 0x7b9a1 max_arch_pfn = 0x400000000
[    0.000000] total RAM covered: 1978M
[    0.000000] Found optimal setting for mtrr clean up
[    0.000000]  gran_size: 64K 	chunk_size: 128M 	num_reg: 4  	lose cover RAM: 0G
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000000] found SMP MP-table at [mem 0x000f9bf0-0x000f9bff]
[    0.000000] RAMDISK: [mem 0x2fb63000-0x33da8fff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000000E5810 000014 (v00 COMPAQ)
[    0.000000] ACPI: RSDT 0x000000007B9C5740 000044 (v01 HPQOEM SLIC-BPC 20091022      00000000)
[    0.000000] ACPI: FACP 0x000000007B9C57E8 000074 (v01 COMPAQ EAGLLAKE 00000001      00000000)
[    0.000000] ACPI BIOS Warning (bug): Optional FADT field Pm2ControlBlock has valid Address but zero Length: 0x0000000000000050/0x0 (20221020/tbfadt-615)
[    0.000000] ACPI BIOS Warning (bug): Invalid length for FADT/Pm2ControlBlock: 0, using default 8 (20221020/tbfadt-669)
[    0.000000] ACPI: DSDT 0x000000007B9C5D27 00A5F1 (v01 COMPAQ DSDT_PRJ 00000001 MSFT 0100000E)
[    0.000000] ACPI: FACS 0x000000007B9C5700 000040
[    0.000000] ACPI: APIC 0x000000007B9C585C 000084 (v01 COMPAQ EAGLLAKE 00000001      00000000)
[    0.000000] ACPI: ASF! 0x000000007B9C58E0 000063 (v32 COMPAQ EAGLLAKE 00000001      00000000)
[    0.000000] ACPI: MCFG 0x000000007B9C5943 00003C (v01 COMPAQ EAGLLAKE 00000001      00000000)
[    0.000000] ACPI: TCPA 0x000000007B9C597F 000032 (v01 COMPAQ EAGLLAKE 00000001      00000000)
[    0.000000] ACPI: SLIC 0x000000007B9C59B1 000176 (v01 HPQOEM SLIC-BPC 00000001      00000000)
[    0.000000] ACPI: HPET 0x000000007B9C5B27 000038 (v01 COMPAQ EAGLLAKE 00000001      00000000)
[    0.000000] ACPI: DMAR 0x000000007B9C5B5F 0001C8 (v01 COMPAQ EAGLLAKE 00000001      00000000)
[    0.000000] ACPI: Reserving FACP table memory at [mem 0x7b9c57e8-0x7b9c585b]
[    0.000000] ACPI: Reserving DSDT table memory at [mem 0x7b9c5d27-0x7b9d0317]
[    0.000000] ACPI: Reserving FACS table memory at [mem 0x7b9c5700-0x7b9c573f]
[    0.000000] ACPI: Reserving APIC table memory at [mem 0x7b9c585c-0x7b9c58df]
[    0.000000] ACPI: Reserving ASF! table memory at [mem 0x7b9c58e0-0x7b9c5942]
[    0.000000] ACPI: Reserving MCFG table memory at [mem 0x7b9c5943-0x7b9c597e]
[    0.000000] ACPI: Reserving TCPA table memory at [mem 0x7b9c597f-0x7b9c59b0]
[    0.000000] ACPI: Reserving SLIC table memory at [mem 0x7b9c59b1-0x7b9c5b26]
[    0.000000] ACPI: Reserving HPET table memory at [mem 0x7b9c5b27-0x7b9c5b5e]
[    0.000000] ACPI: Reserving DMAR table memory at [mem 0x7b9c5b5f-0x7b9c5d26]
[    0.000000] No NUMA configuration found
[    0.000000] Faking a node at [mem 0x0000000000000000-0x000000007b9a0fff]
[    0.000000] NODE_DATA(0) allocated [mem 0x7b976000-0x7b9a0fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.000000]   DMA32    [mem 0x0000000001000000-0x000000007b9a0fff]
[    0.000000]   Normal   empty
[    0.000000]   Device   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.000000]   node   0: [mem 0x0000000000100000-0x000000007b9a0fff]
[... 848 lines cut ...]
[  153.698236] e1000e 0000:00:19.0 enp0s25: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[  154.105472] e1000e 0000:00:19.0 enp0s25: NIC Link is Down
[  156.733985] e1000e 0000:00:19.0 enp0s25: NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[... cut ...]
[ 1982.295803] usb 5-2: Manufacturer: Silicon Labs
[ 1982.295805] usb 5-2: SerialNumber: 0001
[ 1986.148331] usbcore: registered new interface driver usbserial_generic
[ 1986.148352] usbserial: USB Serial support registered for generic
[ 1986.339146] usbcore: registered new interface driver cp210x
[ 1986.339665] usbserial: USB Serial support registered for cp210x
[ 1986.339696] cp210x 5-2:1.0: cp210x converter detected
[ 1988.332185] usb 5-2: cp210x converter now attached to ttyUSB0
ayoub@quanxion:~/ElectroBench$ iwconfig

AyoubCoding21, 2023-12-08:
setup : my pc connected to esp32 by usb cable ( i don't have camera )

mantriyogesh, 2023-12-08:
I must say you are expecting a lot just by connecting ESP just by usb.

Check a image of typical spi setup looks like: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SPI_setup.md

You need to enable the GPIO for specific transport using device tree and then use the ESP-Hosted when platform is posted and ready in that transport (for example spi)

AyoubCoding21, 2023-12-08:
i don't have pins to connect my pc with the esp32

AyoubCoding21, 2023-12-08:
i don't have rpi
i use a real pc

mantriyogesh, 2023-12-08:
You can't use it without GPIOs.
Yet something else, ESP-Hosted can't fulfill your requirement.

AyoubCoding21, 2023-12-08:
like what

mantriyogesh, 2023-12-08:
Yes, software world is a little complex than we think.
Well, I can't help you at this moment my friend.

If you can't use p pp proposals

closed 2023-12-08 by mantriyogesh