espressif/esp-hosted #297: NG: Binary protocol does not compatible with last release

andelf, 2023-12-08:
In https://github.com/espressif/esp-hosted/commit/04ac48b2b8548a674018f5ea74e58d90a3ce0f1c :

It is recommended to append 'ESP_INVALID_IF' to the tail of `ESP_INTERFACE_TYPE` instead of adding it at the beginning.

```diff
typedef enum {
--	ESP_INVALID_IF,
	ESP_STA_IF,
	ESP_AP_IF,
	ESP_SERIAL_IF,
	ESP_HCI_IF,
	ESP_PRIV_IF,
	ESP_TEST_IF,
++	ESP_INVALID_IF,
	ESP_MAX_IF,
} ESP_INTERFACE_TYPE;
```

See-also: https://github.com/embassy-rs/embassy/pull/2269

mantriyogesh, 2023-12-09:
This change was done to avoid any dummy or blank packet to identify itself as 'station' interface.
If you wish to move it to end (or comment it for that matter), is okay as long as you use same change at both, ESP and host.

andelf, 2023-12-10:
@mantriyogesh 
The matter is, the official esp-hosted should maintain its backward compatibility.
Not me. I can use whatever protocol I like.

Users will download the firmware from the release page and asking why 0.0.5 works while 0.x.y not.
This also makes third-party driver builders (other than Linux or stm32 driver provided officially) impossible to support both versions.

Blank packets can be detected in other ways, instead of changing the protocol header types.

mantriyogesh, 2023-12-11:
> Blank packets can be detected in other ways, instead of changing the protocol header types.

Yes, we are have safeguard checks against this, in multiple ways.
- (A) received payload length, (header->len) > 0
- (B) Checksum checked for validity of packet

Okay. This should be just one liner, just to remove `ESP_INVALID_IF` from the enum.
I will change this in the master.
We also accept the code submissions (with credits to author) with pull-request, if you wish to.

We generally take multiple changes in batch (easy to implement and test together for sanity), we will take in next slot.

mantriyogesh, 2023-12-19:
This is taking some time, please bear with us.

mantriyogesh, 2024-04-02:
Fix added as part of fabe8725

closed 2024-04-02 by mantriyogesh