espressif/esp-hosted #298: Esp

felix-lhp, 2023-12-09:
This  is the log of eps32c3::

I (388) stats: *********************************************************************
I (388) stats:                 ESP-Hosted Firmware version :: 1.0.2                        
I (398) stats:                 Transport used :: SPI + UART                    
I (408) stats: *********************************************************************
I (418) FW_MAIN: Supported features are:
I (418) FW_MAIN: - WLAN over SPI
I (428) FW_BT: - BT/BLE
I (428) FW_BT:    - HCI Over UART
I (428) FW_BT:    - BLE only
I (438) FW_MAIN: Capabilities: 0xaa
I (448) pp: pp rom version: 9387209
I (448) net80211: net80211 rom version: 9387209
I (458) wifi:wifi driver task: 3fc9e6ec, prio:23, stack:6656, core=0
I (458) system_api: Base MAC address is not set
I (458) system_api: read default base MAC address from EFUSE
I (468) wifi:wifi firmware version: 9f5e450
I (468) wifi:wifi certification version: v7.0
I (478) wifi:config NVS flash: disabled
I (478) wifi:config nano formating: disabled
I (478) wifi:Init data frame dynamic rx buffer num: 32
I (488) wifi:Init management frame dynamic rx buffer num: 32
I (488) wifi:Init management short buffer num: 32
I (498) wifi:Init dynamic tx buffer num: 32
I (498) wifi:Init static tx FG buffer num: 2
I (508) wifi:Init static rx buffer size: 1600
I (508) wifi:Init static rx buffer num: 10
I (518) wifi:Init dynamic rx buffer num: 32
I (518) wifi_init: rx ba win: 6
I (518) wifi_init: tcpip mbox: 32
I (528) wifi_init: udp mbox: 6
I (528) wifi_init: tcp mbox: 6
I (528) wifi_init: tcp tx win: 5744
I (538) wifi_init: tcp rx win: 5744
I (538) wifi_init: tcp mss: 1440
I (548) wifi_init: WiFi IRAM OP enabled
I (548) wifi_init: WiFi RX IRAM OP enabled
I (558) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (568) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (568) gpio: GPIO[8]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (578) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (588) FW_BT: baud rate for HCI uart :: 921600 

I (598) BTDM_INIT: BT controller compile version [05195c9]
I (598) phy_init: phy_version 912,d001756,Jun  2 2022,16:28:07
I (638) BTDM_INIT: Bluetooth MAC: c0:4e:30:84:98:b6

I (638) FW_MAIN: ESP Bluetooth MAC addr: c0-4e-30-84-98-b6

I (638) FW_SPI: Using SPI interface
I (648) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (658) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1668) FW_MAIN: Initial set up done
I (1678) FW_MAIN: INIT Interface command

I (1678) wifi:mode : sta (c0:4e:30:84:98:b4)
I (1678) wifi:enable tsf

May I ask: Are wif mac and bluethoot mac unique? Or are them randomly generated?This relates to the production and use of products.

mantriyogesh, 2023-12-09:
Unique.

Although, if you wish, you can set the Mac address for the interface.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_set_mac16wifi_interface_tAL6E_K7uint8_t

Need to check fr Bluetooth, but should have support fr that too

mantriyogesh, 2023-12-11:
For bluetooth, you can use:
```c
esp_err_t esp_base_mac_addr_set(const uint8_t *mac) 
```

closed 2023-12-11 by mantriyogesh