espressif/esp-hosted #299: why spi clk select 30Mhz

felix-lhp, 2023-12-09:
@esp32c3 

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x16cc
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d28
entry 0x403ce000
I (30) boot: ESP-IDF v4.4.1-584-g91dc99a3ce-dirty 2nd stage bootloader
I (30) boot: compile time 17:17:27
I (30) boot: chip revision: 4
I (34) boot_comm: chip revision: 4, min. bootloader chip revision: 3
I (41) boot.esp32c3: SPI Speed      **: 80MHz**
I (46) boot.esp32c3: SPI Mode       : DIO
I (50) boot.esp32c3: SPI Flash Size : 4MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (79) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (86) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (94) boot:  3 factory          factory app      00 00 00010000 00100000
I (101) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (109) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (116) boot: End of partition table
I (121) boot: Defaulting to factory image
I (125) boot_comm: chip revision: 4, min. application chip revision: 3
I (132) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=17368h ( 95080) map
I (156) esp_image: segment 1: paddr=00027390 vaddr=3fc90c00 size=032f0h ( 13040) load
I (158) esp_image: segment 2: paddr=0002a688 vaddr=40380000 size=05990h ( 22928) load
I (166) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=8bdd8h (572888) map
I (260) esp_image: segment 4: paddr=000bbe00 vaddr=40385990 size=0b1c8h ( 45512) load
I (268) esp_image: segment 5: paddr=000c6fd0 vaddr=50000010 size=00010h (    16) load
I (274) boot: Loaded app from partition at offset 0x10000
I (274) boot: Disabling RNG early entropy source...
I (288) cpu_start: Pro cpu up.
I (297) cpu_start: Pro cpu start user code
I (297) cpu_start: cpu freq: 160000000
I (297) cpu_start: Application information:
I (300) cpu_start: Project name:     network_adapter
I (305) cpu_start: App version:      2027593
I (310) cpu_start: Compile time:     Dec 16 2022 17:20:30
I (316) cpu_start: ELF file SHA256:  9f0bf326bb4e3f75...
I (322) cpu_start: ESP-IDF:          v4.4.1-584-g91dc99a3ce-dirty
I (329) heap_init: Initializing. RAM available for dynamic allocation:
I (336) heap_init: At 3FC980D0 len 00027F30 (159 KiB): DRAM
I (343) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (349) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (356) spi_flash: detected chip: generic
I (361) spi_flash: flash io: dio
I (365) sleep: Configure to isolate all GPIO pins in sleep state
I (371) sleep: Enable automatic switching of GPIO sleep configuration
I (378) coexist: coexist rom version 9387209
I (383) cpu_start: Starting scheduler.
I (388) stats: *********************************************************************
I (388) stats:                 ESP-Hosted Firmware version :: 1.0.2                        
I (398) stats:                 Transport used :: SPI + UART                    
I (408) stats: *********************************************************************

@arm 

[    3.522300] 270: send header for set cfg change rc_mode/fps/gop
[    3.524460] mpp_vcodec: 162: destroy chan 0 hnd b4b88100 online 0 combo -1 mst -1
[    3.524851] mpp_vcodec: 188: destroy chan 0 done
[    3.528203] vsys_release, 26
[    3.528218] vrgn_release, 122
[    3.528356] vi_release 98
[    3.528481] venc_release 50
[    4.483203] esp32c3:process_esp_bootup_event: 
[    4.483203] Received ESP bootup event
[    4.483266] esp32c3:process_event_esp_bootup: EVENT: 3
[    4.483285] esp32c3:process_event_esp_bootup: EVENT: 2
[    4.483307] esp32c3:adjust_spi_clock: **ESP Reconfigure SPI CLK to 30 MHz**
[    4.483324] esp32c3:process_event_esp_bootup: EVENT: 0
[    4.483342] esp32c3:process_event_esp_bootup: EVENT: 4
[    4.483359] esp32c3:process_event_esp_bootup: EVENT: 1
[    4.483380] esp32c3:process_fw_data: ESP chipset's last reset cause:
[    4.483397] esp32c3:print_reset_reason: POWERON_RESET
[    4.483418] esp32c3:check_esp_version: esp32: ESP Firmware version: 1.0.2
[    4.483439] esp32c3:process_event_esp_bootup: ESP chipset detected [esp32-c3]
[    4.483512] esp32c3:add_network_iface: add_network_iface
[    4.484010] esp32c3:esp_reg_notifier: 629 esp_wifi_device not initialized yet 
[    4.484282] esp32c3:add_network_iface: add_network_iface before esp_cfg80211_add_iface
[    4.523672] esp32c3:add_network_iface: add_network_iface after esp_cfg80211_add_iface
[    4.523720] esp32c3:process_capabilities: ESP peripheral capabilities: 0xaa
[    4.523744] esp32c3:print_capabilities: Capabilities: 0xaa. Features supported are:
[    4.523763] esp32c3:print_capabilities:       * WLAN on SPI
[    4.523778] esp32c3:print_capabilities:       * BT/BLE
[    4.523795] esp32c3:print_capabilities:         - HCI over UART
[    4.523812] esp32c3:print_capabilities:         - BLE only
[   14.910752] esp32c3:cmd_auth_request: Authentication request: f4:84:8d:fb:5a:17 6 0 0 0
[   15.143528] esp32c3:cmd_assoc_request: Association request: f4:84:8d:fb:5a:17 6 22
[   15.175222] esp32c3:process_assoc_event: Connection status: 0
[   15.176401] esp32c3:process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   15.176818] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[   15.202544] esp32c3:process_rx_packet: Rx PACKET_TYPE_EAPOL!!!!
[   15.202678] NOHZ tick-stop error: Non-RCU local softirq work is pending, handler #08!!!
[   36.292255] esp32c3:esp_inetaddr_event: NETDEV_UP interface espsta0 ip changed to  192.168.001.120

The SPI SPEED @c3 is 80MHz, why send to host driver clk sel 30MHz. How to select higher clk speed: 60MHz or higher CLk?

mantriyogesh, 2023-12-09:
Default max is 30MHz considering jumper cables.
If you have PCB, you can try higher frequency.

We have tested till max 40MHz.
If your Linux supports QSPI, you can try QSPI(yet not tested for ESP-Hosted, but anyway knowing, ESP-IDF supports it). This should to increase further speed.

Also you can instead switch to SDIO (using PCB with external pull-ups), to increase the performance.

felix-lhp, 2023-12-11:
esp32c3 not support sdioWhat is the WIFI rate of SPI at 30HMhz?

mantriyogesh, 2023-12-11:
1. If you use FG:
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L102

2. If you use NG: 
- Preferred & easy:
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/network_adapter/main/spi_slave_api.c#L88
OR
- use moduleparam: 
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/host/main.c#L35
but need to comment line, 
https://github.com/espressif/esp-hosted/blob/ce3c50a33fa4bc562a1b6cbcee292c1ae0b0a404/esp_hosted_ng/host/spi/esp_spi.c#L204

mantriyogesh, 2023-12-11:
This is just information, to keep in mind while tuning the SPI clock:

1. SPI slave peripheral max supported clock for ESP32-C3: 60MHz
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/spi_slave.html#sclk-frequency-requirements

2. Where clocks would be limited when:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/spi_slave.html#restrictions-and-known-issues

You would need to optimize the spi speed at much possible depending upon the timing of the SPI transaction timings.
[image]

mantriyogesh, 2023-12-11:
> esp32c3 not support sdio

Yes, this is correct.
You can use ESP32-C6 or ESP32 for SDIO slave.
SDIO is half duplex (by nature of protocol), but should achieve better speeds with 4-bit SDIO (with ESP32-C2 or ESP32)

mantriyogesh, 2023-12-18:
Does above answer your questions?
Can we close this issue?

closed 2023-12-19 by mantriyogesh