espressif/esp-hosted #300: STM32MP151C with ESP32C6mini over SPI5

vardan, 2023-12-12:
Hello,

A question more on a higher level as I am still struggling to get the device tree right: what ultimately should be a driver in ```compatible``` field of device tree?
Right now, the DT looks like following:

```
&spi5{
    pinctrl-names = "default", "sleep";
    pinctrl-0 = <&spi5_pins_mx>;
    pinctrl-1 = <&spi5_sleep_pins_mx>;
    cs-gpios = <&gpioj 1>;
    num-cs = <1>;
    status = "okay";
 
    esp32spi@0 {
        **compatible = "espressif,esp32-spi";**
        reg = <0>; /* CS #0 */
        spi-max-frequency = <24000000>;
		spi-cs-high;
        reset-gpios = <&gpiof 15 GPIO_ACTIVE_HIGH>;
    };
};
```
Compatible field still confuses me because when I try to put something not from linux supported drivers (including esp32-spi) the device is simply missing.

mantriyogesh, 2023-12-13:
Hello @vardan ,

We do not have any formal compatible string yet.
Getting the spi node working is irrespective of compatible.

Also, we can guide to get your linux up, but anyway you would need to make spi node available, as prerequisite of ESP-Hosted..

We cannot test you device tree as such and every linux may have different customisation possible as per their data sheet.
However, from general background, comments:

I could see that you are going to use SPI bus 5, CS 1.
SO WITHOUT spidev disabling, you should get output in 
$ ls /dev/spidev5.1

Once confirmed this, you can disable spidev so as kernel module would be able to access this device.

you wish to also cross check if you want to use active high CS. (spi-cs-high)
in general we use active low CS in ESP side, same needed in host side.

Also please take a look at Porting guide. https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md

vardan, 2023-12-13:
Thanks for quick reply, @mantriyogesh. Now it starts to take shape and I was able to get esp32-spi.ko loaded:
`esp32_spi:spi_dev_init: ESP32 peripheral is registered to SPI bus [5],chip select [0], SPI Clock [10] `

It looks however that I don't get any INIT events from ESP32, nothing in the dmesg

mantriyogesh, 2023-12-14:
Hello @vardan ,

Before disabling spidev, you observed that 
```
$ ls /dev/spidevX.Y 
```
Was visible and disappears on disabling spidev?
X I assume is 5 and Y is 1 (or 0)?

Also, have you added and tested the additional pins required for spi communication? Porting guide already details it..

mantriyogesh, 2023-12-14:
Okay.
Make sure you use exactly same code base for flashing ESP and host kernel module building.

vardan, 2023-12-14:
Oh, okay. That is tricky... I think environments are very similar, but they are different linux machines

vardan, 2023-12-15:
@mantriyogesh , now I have flashed ESP-AT firmware on to ESP32C6 device, however, I still don't get any INIT events. All the additional pins were configured in host/spi/esp_spi.h:
```
#define HANDSHAKE_PIN           156
#define SPI_IRQ                 gpio_to_irq(HANDSHAKE_PIN)
#define SPI_DATA_READY_PIN      162
#define SPI_DATA_READY_IRQ      gpio_to_irq(SPI_DATA_READY_PIN)
#define SPI_BUF_SIZE            1600
```
And in esp_spi.c:
```
esp_board.max_speed_hz = spi_clk_mhz * NUMBER_1M;
esp_board.bus_num = 5;
esp_board.chip_select = 0;
```
Kernel module is loaded and is using SPI5 with clock 10MHz, but I am not sure what frequency is used on the other end

vardan, 2023-12-15:
Here is the bootlog from ESP peripheral
```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd54
load:0x4086e610,len:0x2de0
load:0x40875728,len:0x17d8
entry 0x4086c410
I (23) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (24) boot: compile time Dec 13 2023 17:37:11
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 otadata          OTA data         01 00 0000d000 00002000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 nvs              WiFi data        01 02 00010000 0000e000
I (80) boot:  3 at_customize     unknown          40 00 0001e000 00042000
I (87) boot:  4 ota_0            OTA app          00 10 00060000 001d0000
I (95) boot:  5 ota_1            OTA app          00 11 00230000 001d0000
I (102) boot: End of partition table
I (107) boot: No factory image, trying OTA 0
I (111) esp_image: segment 0: paddr=00060020 vaddr=42150020 size=30b88h (199560)                                                                                                                                                                                                 map
I (161) esp_image: segment 1: paddr=00090bb0 vaddr=40800000 size=0f468h ( 62568)                                                                                                                                                                                                 load
I (176) esp_image: segment 2: paddr=000a0020 vaddr=42000020 size=14651ch (133660                                                                                                                                                                                                4) map
I (453) esp_image: segment 3: paddr=001e6544 vaddr=4080f468 size=0a57ch ( 42364)                                                                                                                                                                                                 load
I (463) esp_image: segment 4: paddr=001f0ac8 vaddr=408199f0 size=03c80h ( 15488)                                                                                                                                                                                                 load
I (468) esp_image: segment 5: paddr=001f4750 vaddr=50000000 size=00068h (   104)                                                                                                                                                                                                 load
I (475) boot: Loaded app from partition at offset 0x60000
I (509) boot: Set actual ota_seq=1 in otadata[0]
I (509) boot: Disabling RNG early entropy source...
no external 32k oscillator, disable it now.
at param mode: 1
AT cmd port:uart1 tx:7 rx:6 cts:5 rts:4 baudrate:115200
module_name: ESP32C6-4MB
max tx power=78, ret=0
v4.0.0.0-dev
```

mantriyogesh, 2023-12-15:
> now I have flashed ESP-AT firmware

Are you trying ESP-Hosted or ESP-AT?

mantriyogesh, 2023-12-15:
Also, Have you verified the GPIOs to be working?
Above log is not familiar ESP-Hosted log.

vardan, 2023-12-15:
Ok, now I get it - I need to build and use ESP-hosted firmware, right?

mantriyogesh, 2023-12-15:
Yes please.
Also, please ensure the ESP-Hosted porting guide is followed correctly.

If you fail to follow that stepwise, you will likely face the typical set of issues and verification techniques that [porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/porting_guide.md) already explains.

We will help as much as possible, but please bear in mind, getting the platform up (Device tree, disabling spidev , etc, etc) is prerequisite for ESP-Hosted to work and is user's responsibility to do.

vardan, 2023-12-15:
Again a high level question: is it possible to use esp-hosted-fg as a firmware and esp-hosted-ng as a host driver? The reason I am asking is we are using ESP32C6 which is not supported by esp-hosted-ng atm

mantriyogesh, 2023-12-15:
Not possible that way.
But NG support is coming soon.

vardan, 2023-12-15:
How soon? :) 
Not sure how to proceed then. If I try to flash esp32c3 firmware, ESP peripheral of course complains that chip is wrong. I might try to use hosted-fg even for Linux host, but it feels like there will be obstacles

mantriyogesh, 2023-12-15:
FG and NG transport is same.
So, As you have got NG running, you will also get FG running easily.

Check a nice demo, https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/python_demo.md with GIF created, to make you aware. Demo in C is also available.

vardan, 2023-12-18:
Hi again @mantriyogesh , another question maybe to community in general on how clock looks like in device tree after NG is running? I still see, even when kernel module is loaded, that clocks are disabled in `/sys/kernel/debug/clk/clk_summary`. Enable count is 0

mantriyogesh, 2023-12-18:
I suppose it should be dependent on the Clock derived from for your SoC, for your SPI instance.

For Raspberry Pi, I could see thread: https://forums.raspberrypi.com/viewtopic.php?t=277596

For my current Raspberry Pi, values look like:

```
ym2@rpi2:~ $ sudo cat /sys/kernel/debug/clk/clk_summary
                                 enable  prepare  protect                                duty  hardware
   clock                          count    count    count        rate   accuracy phase  cycle    enable
-------------------------------------------------------------------------------------------------------
 fw-clk-vec                           0        0        0           0          0     0  50000         Y
 fw-clk-pixel-bvb                     0        0        0    75000000          0     0  50000         Y
 fw-clk-m2mc                          0        0        0   120000000          0     0  50000         Y
 fw-clk-hevc                          0        0        0   250000000          0     0  50000         Y
 fw-clk-pixel                         0        0        0           0          0     0  50000         Y
 fw-clk-isp                           0        0        0   250000000          0     0  50000         Y
 fw-clk-v3d                           0        0        0   250000000          0     0  50000         Y
 fw-clk-core                          2        2        0   250000000          0     0  50000         Y
 fw-clk-arm                           0        0        0   800000000          0     0  50000         Y
 108MHz-clock                         0        0        0   108000000          0     0  50000         Y
    hdmi1-108MHz                      0        0        0   108000000          0     0  50000         N
    hdmi0-108MHz                      0        0        0   108000000          0     0  50000         N
 27MHz-clock                          0        0        0    27000000          0     0  50000         Y
 otg                                  0        0        0   480000000          0     0  50000         Y
 osc                                  4        4        0    54000000          0     0  50000         Y
    tsens                             0        0        0     3375000          0     0  50000         Y
    otp                               0        0        0    13500000          0     0  50000         Y
    timer                             0        0        0     1000000          0     0  50000         Y
    plld                              5        5        0  3000000091          0     0  50000         Y
       plld_dsi1                      1        1        0    11718751          0     0  50000         Y
       plld_dsi0                      1        1        0    11718751          0     0  50000         Y
       plld_per                       2        2        0   750000023          0     0  50000         Y
          emmc2                       1        1        0   100000003          0     0  50000         Y
          emmc                        0        0        0   250000007          0     0  50000         Y
          uart                        0        0        0    48000001          0     0  50000         Y
       plld_core                      1        1        0   600000019          0     0  50000         Y
    pllc                              5        5        0  2592000000          0     0  50000         Y
       pllc_per                       1        1        0   648000000          0     0  50000         Y
       pllc_core2                     1        1        0    10125000          0     0  50000         Y
       pllc_core1                     1        1        0    10125000          0     0  50000         Y
       pllc_core0                     1        1        0    10125000          0     0  50000         Y
    pllb                              2        2        0  1600000024          0     0  50000         Y
       pllb_arm                       1        1        0   800000012          0     0  50000         Y
    plla                              4        4        0  2999999988          0     0  50000         Y
       plla_ccp2                      1        1        0    11718750          0     0  50000         Y
       plla_dsi0                      1        1        0    11718750          0     0  50000         Y
       plla_core                      2        2        0   499999998          0     0  50000         Y
          h264                        0        0        0   249999999          0     0  50000         Y
          isp                         0        0        0   249999999          0     0  50000         Y
          vpu                         3        3        0   250000000          0     0  50000         Y
             aux_spi2                 0        0        0   250000000          0     0  50000         N
             aux_spi1                 0        0        0   250000000          0     0  50000         N
             aux_uart                 0        0        0   250000000          0     0  50000         N
             peri_image               0        0        0   250000000          0     0  50000         Y
 dsi1p                                0        0        0           0          0     0  50000         Y
 dsi0p                                0        0        0           0          0     0  50000         Y
 dsi1e                                0        0        0           0          0     0  50000         Y
[... cut ...]
 hsm                                  0        0        0           0          0     0  50000         Y
 
 ```
 
 As can see vpu is 250M for my RasPi, As I have 

```
 ym2@rpi2:~ $ cat /boot/config.txt  | tail -5
[all]
dtparam=spi=on
dtoverlay=disable-bt
core_freq=250           # Reduce and Fix clock to 250M - SPI workaround for Clock
core_freq_min=250
```

You can state your SoC if anyone can answer from community.
Also you can refer to your SoC datasheet or Reference manual, to understand which clock the SPI is derived from.

mantriyogesh, 2023-12-18:
```
ym2@rpi2:~ $ sudo cat /sys/kernel/debug/clk/aux_spi2/clk_rate
250000000
ym2@rpi2:~ $ sudo cat /sys/kernel/debug/clk/aux_spi1/clk_rate
250000000
```

mantriyogesh, 2023-12-18:
```
spi-cs-high
```

might not work with ESP chipsets. ESP chipsets assume **Chip select is active low**, always.
If you need **active high**, you might have to add such change in spi_slave_api.c in slave initialization.

**At slave**, Something like
```
spi_dev_t *hw = SPI_LL_GET_HW(GPSPI2/3);
assert(hw != NULL);
hw->misc.slave_cs_pol = 1;
```

**Please note:**
1. This is just for reference. you might need to adjust as per your SPI instance **at slave**.
2. We have only tested with active low CS always at ESP slave and do not intend to use Active high CS.

vardan, 2023-12-18:
Thanks @mantriyogesh! I've already removed spi-cs-high setting from DTS

mantriyogesh, 2023-12-18:
Following comments might be helpful for device tree:

https://github.com/espressif/esp-hosted/issues/128#issuecomment-1178572863
https://github.com/espressif/esp-hosted/issues/165#issuecomment-1275877729

vardan, 2023-12-19:
Hi @mantriyogesh,

a question about pin mux on ESP32C6 side - where is it configured in firmware? I see that on our board we are using IO4 as DATA_READY and IO5 is floating, but according to datasheet those are strapping pins for the clock polarity settings. Is it possible to change pin function from firmware?

vardan, 2023-12-19:
There is a macro defined in host_reg.h in esp-idf SLCHOST_FRC_POS_SAMP, but I do not see it being used anywhere in firmware code

mantriyogesh, 2023-12-20:
Pin connections with C6 are tested with: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/SPI_setup.md#111-pin-connections

ESP32-C6 TRM: https://www.espressif.com/sites/default/files/documentation/esp32-c6_technical_reference_manual_en.pdf

The strapping pins for ESP32-C6 should be 
[image]

In general, you can choose any GPIO for SPI pins for ESP32-C6, unless they are being used for something else.
IT will follow GPIO matrix signal routing.

https://github.com/espressif/esp-hosted/issues/300#issuecomment-1857425426 was showing incorrect firmware flashed for ESP-Hosted. Have you corrected it?
You can follow https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg, keeping in mind ESP32-C6 as ESP chipset as slave and transport as SPI.

mantriyogesh, 2023-12-20:
I assume you using FG from earlier discussion.

In general, you can choose GPIO can be changed:
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L118-L124
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L133-L134

Change and flash using esp32-c6 idf.py (refer documentation)

Make sure you have exactly same `git commit` used at slave and host building.
Also, please check [Porting_guide](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md). You can step by step things to be done for testing pins and interrupts.

Please post both side textual logs:
ESP: full minicom log from flashing
Linux Host: dmesg from boot, commands run and their outputs
Both: git commit (prefer to use master)
Linux Host: uname -a

Without  current logs, will be hard to understand current problems.

vardan, 2023-12-22:
I checked the clock in oscilloscope and what I get out to SPI master clock is 50KHz, 20 times lower than configured (1 MHz). There are no dividers I can see in the code, the clock SPI is using is an external HSE clock of 24 MHz. STM32CP151 has a OSPEEDR register for GPIO max speed settings but even lowest settings are well above the frequency I am trying to achieve. Are there any dividers one shoud be aware of in esp-hosted solution?

mantriyogesh, 2023-12-23:
That's good that you tested with analyser.

There are no div as such known to ESP-Hosted.
The init frequency is controlled at host: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L34

And readjusted after receiving clock freq change (depending on ESP chipset in slave used) at 
https://github.com/espressif/esp-hosted/blob/ea8b0aae76813d663fed1a4d3e8b6a6f96ac43eb/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L197

However, irrespective of frequency, **Important** is that, you able to get init event (even on small freq, this is must). Check dmesg for some 'EVENT' traces from esp32_spi.

Regarding spi clock, kernel module just assume the spi is working fine.
So prerequisite is that spi pins, mode, the clock etc is stable working.
You might need to check your device driver for spi, how the frequency is released for your SoC.

You can also run raw throughout test, just as marker, to understand max TX and RX speed at pure transport. https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/Raw_TP_Testing.md.
This throughout should work before you try any iperf etc tests.

vardan, 2024-01-04:
Hi @mantriyogesh, now I was able to sort out clock issues, so the clock is correct now. I still however not able to get any INIT event from ESP32 in to host. I have actually tweaked hosted-ng to work on ESP32C6 target. At least I am able to flash it and it looks like it is sending and bootup event to host. What I see on the host side however is all zeros.

ESP32C& bootup log:

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x1c (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd10
load:0x4086e610,len:0x2dc4
load:0x40875728,len:0x17d8
entry 0x4086c410
I (23) boot: ESP-IDF d3c99ed3-dirty 2nd stage bootloader
I (24) boot: compile time Jan  3 2024 13:36:18
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 4MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (65) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (73) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  3 factory          factory app      00 00 00010000 00100000
I (88) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (95) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (103) boot: End of partition table
I (107) boot: Defaulting to factory image
I (112) esp_image: segment 0: paddr=00010020 vaddr=420b0020 size=21cb8h (138424) map
I (148) esp_image: segment 1: paddr=00031ce0 vaddr=40800000 size=0e338h ( 58168) load
I (162) esp_image: segment 2: paddr=00040020 vaddr=42000020 size=a051ch (656668) map
I (296) esp_image: segment 3: paddr=000e0544 vaddr=4080e338 size=0b090h ( 45200) load
I (313) boot: Loaded app from partition at offset 0x10000
I (313) boot: Disabling RNG early entropy source...
I (324) cpu_start: Unicore app
I (325) cpu_start: Pro cpu up.
W (333) clk: esp_perip_clk_init() has not been implemented yet
I (340) cpu_start: Pro cpu start user code
I (340) cpu_start: cpu freq: 160000000 Hz
I (340) cpu_start: Application information:
I (343) cpu_start: Project name:     network_adapter
I (349) cpu_start: App version:      release/ng-v1.0.2-136-g301e94a-
I (356) cpu_start: Compile time:     Jan  3 2024 22:39:55
I (362) cpu_start: ELF file SHA256:  5b723fbda8352c6d...
I (368) cpu_start: ESP-IDF:          d3c99ed3-dirty
I (373) cpu_start: Min chip rev:     v0.0
I (378) cpu_start: Max chip rev:     v0.99
I (383) cpu_start: Chip rev:         v0.0
I (388) heap_init: Initializing. RAM available for dynamic allocation:
I (395) heap_init: At 4081D8D0 len 0005ED40 (379 KiB): D/IRAM
I (401) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (408) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (415) spi_flash: detected chip: generic
I (419) spi_flash: flash io: dio
I (423) sleep: Configure to isolate all GPIO pins in sleep state
I (430) sleep: Enable automatic switching of GPIO sleep configuration
I (437) coexist: coex firmware version: 80b0d89
I (442) coexist: coexist rom version 5b8dcfa
I (447) app_start: Starting scheduler on CPU0
I (452) main_task: Started on CPU0
I (452) main_task: Calling app_main()
I (452) stats: *********************************************************************
I (462) stats:                 ESP-Hosted Firmware version :: 1.0.2             
I (472) stats:                 Transport used :: SPI only
I (482) stats: *********************************************************************
I (492) FW_MAIN: Supported features are:
I (492) FW_MAIN: - WLAN over SPI
[... 170 lines cut ...]
[ 1671.689180] spi_rx: 6228bbf9: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.699976] spi_rx: c87e8c88: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.708546] spi_rx: 85d44e2e: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.717174] spi_rx: dd0a4475: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.725767] spi_rx: c79b865f: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.734678] spi_rx: 70884064: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.743207] spi_rx: 4d046af5: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.750612] spi_rx: ccd4b5ab: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.761724] spi_rx: 7dbd7abb: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.771323] spi_rx: 1b66e7c8: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.780221] spi_rx: 63dc1d59: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.788826] spi_rx: e2a4c399: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.797657] spi_rx: 85a4080f: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.806212] spi_rx: 56c298a6: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.814782] spi_rx: 48b2f4f4: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.823661] spi_rx: 8ed8f034: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.830966] spi_rx: a79af693: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.842010] spi_rx: 680a35bc: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.849316] spi_rx: 7cc9dd83: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1671.860161] if_num 0
[ 1671.860905] seq_num 0
[ 1671.865261] flags 0
```

The problem can of course lie in my tweaked solution to support C6, but at least I should be able to get some event into host

vardan, 2024-01-04:
On the analyzer I see MISO gets pulled low, there are some very short pulses happening, but they are way too short to be detected on 2MHz frequency I am running. See picture:
[image]

vardan, 2024-01-04:
Another thing is if I understand this correctly SPI controller should be 2, which is a GP-SPI controller SPI0/1 are used for communication with memory. Is this right or I am misunderstanding it completely?

mantriyogesh, 2024-01-04:
If you use ESP32-C6 with dev kit, by default you use the GPIO matrix (Not the IO-Mux).
Can you collect the logic analyser capture?

I strongly suspect you might have to work with the spi clock, spi modes and spi timings.
Timings strongly change depending upon the length of jumper cables.
You should use smaller jumper cables for stability.

Also please consider many factors detailed in Porting guide once.

Unless the lower transport timings are corrected, you will face issues.
So in general, once you move to specific mode, which looks working, it is worth to test raw throughout test. These all.details are captured in Porting guide.. 

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/porting_guide.md

Also do not jump to high frequency, try smaller frequency and step wise increase, till you find optimal frequency. Logic analyser will help a lot to understand the issues very clearly..

mantriyogesh, 2024-01-04:
It might seem difficult or complex in start, but it should not be.
We can also assess  and help at each step..

But it is really suggested to carefully slowly follow Porting guide..
Feel free to ping us for any confusion..

vardan, 2024-01-04:
No, it is not a dev kit. It is a ready-made PCB, so I cannot change any wiring. With regard to frequency , I am running now at 1-2 MHz already, I cannot go any lower than that

mantriyogesh, 2024-01-05:
I also see you use C6 strapping pin GPIO16 at C6. Can you please change it with something else?

[image]

If you have GPIO header, you can add direct jumper and change in C6 firmware and flash.

vardan, 2024-01-08:
Hi @mantriyogesh, I got it working now - changed to latest version that features esp32c6 support. But also discovered a bug in my device tree - CS was wrongly configured. Communication is working now and espsta0 interface is up and running. Thanks for your support! Will try to write "porting done" style ticket and describe my fundings for STM32MP151C

mantriyogesh, 2024-01-08:
That would be super helpful..
If can also let us know what issue you identified in the device tree while be helpful for someone later visiting this page..

vardan, 2024-01-10:
Hi again @mantriyogesh , sorry, another question. I have esp32_spi module configured as autoload in bb recipe, but for some reason it gets probed before spi_stm32 kernel driver has initialized, which means by the time esp32_spi is loaded device is still not available
```
[   18.273007] esp32_spi: loading out-of-tree module taints kernel.
[   18.494925] esp32_spi:spi_dev_init: Failed to obtain SPI master handle
[   18.932031] esp32_spi:spi_init: Failed Init SPI device
[   21.147883] spi_stm32 44004000.spi: driver initialized
[   21.161186] esp32spi@0 enforce active low on chipselect handle
[   21.199236] spi_stm32 44009000.spi: driver initialized
```
Is there a way around this issue?

mantriyogesh, 2024-01-10:
Instead of hooking in kernel module, possible to load bash rpi_init.sh after system startup?
Possible places would be 
rc.local
or systemd daemon or some other way

If these not possible, you can make esp32_spi dependent on spi_stm32
But need to check how the kernel module execution is done by kernel.

mantriyogesh, 2024-01-10:
There were several layers how kernel loads the modules.
Compile with kernel will always load. 
1. Is there entry of esp32_spi in newly built modules.dep after depmod?
2. Is modules.dep copied to /lib/modules/$(uname -r)/modules.dep ?

vardan, 2024-01-10:
Adding `MODULE_SOFTDEP("pre: spi_stm32");` in host/main.c and re-runing depmod did the trick

vardan, 2024-01-17:
Hi @mantriyogesh, for spi_slave is `gpio_set_drive_capability(GPIO_SCLK, GPIO_DRIVE_CAP_3);` necessary? We experience that with higher frequencies (currently running with 16MHz) clock signal form is not a square wave anymore, even when it is driven hard from spi master with slew rate 3

mantriyogesh, 2024-01-17:
Okay. No it is needed if you face singnal issues.
Typically we observed with long wires, which generally is case for users, with normal drive strength, some times signals are lagged.

You can remove if you have PCB connection already. In fact with no drive strength, you can additionally save some power.

vardan, 2024-02-20:
Hi again @mantriyogesh,

Can you please explain how txpower is set in hosted_ng? I went through other similar Issues here, but I still did not get how power is set.

Here's my example:

```
root@stm32:~# iw wlan0 info
Interface wlan0
        ifindex 9
        wdev 0x200000001
        addr 40:4c:ca:46:fd:e0
        type managed
        wiphy 2
        txpower 0.00 dBm
root@stm32:~# iw wlan0 set txpower limit 2000
root@stm32:~# iw wlan0 info
Interface wlan0
        ifindex 9
        wdev 0x200000001
        addr 40:4c:ca:46:fd:e0
        type managed
        wiphy 2
        txpower 30.00 dBm
```
wlan0 is the same as espsta0, I added a link basically. As you can see even if I try to set power to 20dBm it looks like txpower is set to 30dBm. Looks like something is hardcoded in esp32c6 fw

mantriyogesh, 2024-02-20:
I will check and get back to you.