espressif/esp-hosted #301: complile error

hustzju, 2023-12-14:
when I compile esp-drive ，I have error below.please help me, I'll appreciate it 
/main/sdio_slave_api.c: In function 'sdio_init':
../main/sdio_slave_api.c:119:18: error: 'SDIO_SLAVE_FLAG_DEFAULT_SPEED' undeclared (first use in this function); did you mean 'SDIO_SLAVE_FLAG_DAT2_DISABLED'?
  config.flags |= SDIO_SLAVE_FLAG_DEFAULT_SPEED;

mantriyogesh, 2023-12-14:
What's the idf version being used?

hustzju, 2023-12-15:
idf4.3

mantriyogesh, 2023-12-15:
Hello @hustzju ,

You have three options:
1. Follow the documentation, which will automatically use ESP-IDF release/v5.1 branch.
[image]
:warning: Save the local changes if you have any.

2. Update your IDF release to branch, release/v5.1, and set up IDF again using ./install.sh etc
3. Comment the line , for release/v4.3, this line doesn't support (base IDF change available for release/v5.1).

If you know how to do, we recommend option (2).
If unsure, follow option (1), seamless and automated.

Please follow documentation as is for correct expected behaviour.

hustzju, 2023-12-15:
Thanks for your reply,  I follow your  option(2) and  I update IDF to V5.1, then recompile,  A new  error occur：

hustzju, 2023-12-15:
es/__idf_main.dir/cmd.c.obj   -c ../main/cmd.c
../main/cmd.c: In function 'sta_connected_cb':
../main/cmd.c:1876:2: error: unterminated argument list invoking macro "ESP_LOGI"
 1876 | }
      |  ^
../main/cmd.c:232:9: error: 'ESP_LOGI' undeclared (first use in this function)
  232 |         ESP_LOGI(TAG, "STA connected with:" MACSTR "\n", MAC2STR(bssid);

mantriyogesh, 2023-12-15:
Do you want to use FG or NG?
From earlier comments, I inferred that you use FG.

Last comment however is NG.

mantriyogesh, 2023-12-15:
Complete trace with command and output, what you are trying to do, will be useful.
If ESP-IDF is not clean, you will face problems like this.

Goto option (1).
⚠️ Please be aware, any of local changes, will be lost!
So save your work somewhere and try option (1).

hustzju, 2023-12-15:
I use  NG.

hustzju, 2023-12-15:
I use NG.  and I try to follow your suggestion option(1).

mantriyogesh, 2023-12-15:
In case of ng, follow,

https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/setup.md#13-esp-comprehensive-guide

Also check https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng fully..

hustzju, 2023-12-15:
I use NG. and I try to follow your suggestion option(1), and my steps is
1.git clone --recursive https://github.com/espressif/esp-hosted.git (before this, I use ./jihu-mirror.sh set, in China )
2. cd esp-hosted/esp_hosted_ng/esp/esp_driver
3. cmake .  
 everything is OK,  I receive:
 All done! You can now run:
. ./export.sh)
4. cd cd esp-idf/ and  . ./export.sh

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
  
5.  cd ../network_adapter/  
6. idf.py set-target esp32
7. idf.py build

BUT  the error still occurs just like above

hustzju, 2023-12-15:
I'm feeling a bit overwhelmed by this question， Thanks for your reply again

shreyassheth1, 2023-12-15:
Hey @hustzju thanks for reporting the issue. We have fixed the issue internally and fix will be available on repository soon.

hustzju, 2023-12-15:
OK, So  Can I  compile esp-hosted-ng  firmware   by myself  or download  your release  asset  directly  at this moment ?

mantriyogesh, 2023-12-15:
yes, try now. the GitHub is updated with the change.
Get us the complete log if you fail..

hustzju, 2023-12-15:
It works now

closed 2023-12-15 by hustzju