espressif/esp-hosted #302: ESP32-C3 esp_ble_gattc_write_char, The connection  not created

monkey-qiu, 2023-12-15:
ESP32-C3 IDF5.1 
 蓝牙5.0主从一体模式，ESP 设备作为client连接一个server，然后ESP设备的server被app主机连接，当app和server的连接断开是，会上报ESP_GATTS_DISCONNECT_EVT 和 ESP_GATTC_DISCONNECT_EVT两个事件，然后ESP设备作为client向server发送数据时报未建立连接
[0m
[0;32mI (112345) GATTC_COEX: Gattc_Read_Time =  112346ms, client_s
end_cnt = 16, client_recv_cnt = 0,client_sendlose_cnt = 0[0m
[0;33mW (116605) BT_APPL: gattc_conn_cb: if=4 st=0 id=260 rsn=0x13[0m
[0;33mW (116605) BT_HCI: hcif disc complete: hdl 0x1, 
rsn 0x13[0m
[0;32mI (116605) GATTC_GATTS_COEX: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13[0m
[0;32mI (116615) GATTC_GATTS_COEX: ESP_GATTC_DISCONNECT_EVT, reason = 19
[0m
[0;32mI (116615) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m
[0;32mI (117345) GATTC_COEX: start write =  117345ms
[0m
[0;33mW (117345) BT_LOG: esp_ble_gattc_write_char, The connection 
not created.[0m
[0;32mI (117345) GATTC_COEX: Gattc_Read_Time =  117347ms, client_send_cnt = 16, client_recv_cnt = 0,client_sendlose_cnt = 1[0m
[0;32mI (122355) GATTC_COEX: start write =  122355ms
[0m
[0;33mW (122355) BT_LOG: esp_ble_gattc_write_char, The connection 
not created.[0m
[0;32mI (122355) GATTC_COEX: Gattc_Read_Time =  122357ms, client_send_cnt = 16, client_recv_cnt = 0,client_sendlose_cnt = 2[0m

mantriyogesh, 2023-12-15:
1. Full logs please.

ESP : full log from flashing.
Host: dmesg from boot, commands fired with their output, **btmon** log
If you have Bluetooth sniffer capture, will be fast to locate the issue.

2. What is the host: MCU or Linux? version?

monkey-qiu, 2023-12-15:
> 1. Full logs please.
> 
> ESP : full log from flashing. Host: dmesg from boot, commands fired with their output, **btmon** log If you have Bluetooth sniffer capture, will be fast to locate the issue.
> 
> 2. What is the host: MCU or Linux? version?

[0;31mE (6525) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:[0m
[0;31
mE (6525) task_wdt:  - IDLE (CPU 0)[0m
[0;31mE (6525) task_wdt: Tasks currently running:[0m
[0;31mE (6525) task_wdt: CPU 0: throughput_clie[0m
[0;31mE (6525) task_wdt: Print CPU 0 (current core) registers[0m
Core  0 register dump:
MEPC    : 0x4200913c  RA      : 0x42008f28  SP      : 0x3fcaa170  GP      : 0x3fc91000  
TP      : 0x3fc879c4  T0      : 0x00000000  T1      : 0x00000000  T2      : 0x00000000  
S0/FP   : 0x00000000  S1      : 0x00000000  A0      : 0x00000001  A1      : 0x00000001  
A2      : 0x00000074  A3      : 0x00000004  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00000000  MTVEC   : 0x00000000  MCAUSE  : 0x42008f12  MTVAL   : 0x00000000  
MHARTID : 0x00000000  
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xf (SPI_FAST_FLASH_BO
OT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x170c
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f68
entry 0x403cc710
[0;32mI (30) boot: ESP-IDF v5.1.1-dirty 2nd stage bootloader[0m
[0;32mI (30) boot: compile time Nov 23 2023 09:12:32[0m
[0;32mI (30) boot: chip revision: v0.3[0m
[0;32mI (34) boot.esp32c3: SPI Speed      : 80MHz[0m
[0;32mI (39) boot.esp32c3: SPI Mode       : DIO[0m
[0;32mI (43) boot.esp32c3: SPI Flash Size : 2MB[0m
[0;32mI (48) boot: Enabling RNG early entropy source...[0m
[0;32mI (53) boot: Partition Table:[0m
[0;32mI (57) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (79) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (87) boot: End of partition table[0m
[0;32mI (91) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=23110h (143632) map[0m
[0;32mI (122) esp_image: segment 1: paddr=00033138 vaddr=3fc90800 size=02124h (  8484) load[0m
[0;32mI (124) esp_image: segment 2: paddr=00035264 vaddr=40380000 size=0adb4h ( 44468) load[0m
[0;32mI (136) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=91ea8h (597672) map[0m
[0;32mI (232) esp_image: segment 4: paddr=000d1ed0 vaddr=4038adb4 size=05a18h ( 23064) load[0m
[0;32mI (242) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (242) boot: Disabling RNG early entropy source...[0m
[0;32mI (253) cpu_start: Unicore app[0m
[0;32mI (253) cpu_start: Pro cpu up.[0m
[0;32mI (262) cpu_start: Pro cpu start user code[0m
[0;32mI (262) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (262) cpu_start: Application information:[0m
[0;32mI (265) cpu_start: Project name:     gattc_gatts_coex[0m
[0;32mI (271) cpu_start: App version:      v5.1.1-dirty[0m
[0;32mI (276) cpu_start: Compile time:     Dec 15 2023 14:56:04[0m
[0;32mI (282) cpu_start: ELF file SHA256:  4570852ab15bfa94...[0m
[0;32mI (288) cpu_start: ESP-IDF:          v5.1.1-dirty[0m
[0;32mI (294) cpu_start: Min chip rev:     v0.3[0m
[0;32mI (298) cpu_start: Max chip rev:     v0.99 [0m
[0;32mI (303) cpu_start: Chip rev:         v0.3[0m
[0;32mI (308) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (315) heap_init: At 3FC97060 len 000456B0 (277 KiB): DRAM[0m
[0;32mI (321) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM[0m
[0;32mI (328) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM[0m
[... 284 lines cut ...]
[0;32mI (87825) GATTC_COEX: start write =  87825ms
[0m
[0;3
2mI (87825) GATTC_COEX: Gattc_Read_Time =  87825ms, client_send_cnt = 10, client_recv_cnt = 0,client_sendlose_cnt = 0[0m
[0;33mW (87945) BT_APPL: gattc_conn_cb: if=4 st=0 id=260 rsn=0x13[0m
[0;33mW (87945) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13[0m
[0;32mI (87945) GATTC_GATTS_COEX: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x13[0m
[0;32mI (87955) GATTC_GATTS_COEX: ESP_GATTC_DISCONNECT_EVT, reason = 19
[0m
[0;32mI (87965) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m
[0;32mI (92825) GATTC_COEX: start write =  92825ms
[0m
[0;3
3mW (92825) BT_LOG: esp_ble_gattc_write_char, The connection not created.[0m
[0;32mI (92825) GATTC_COEX: Gattc_Read_Time =  92827ms, client_send_cnt = 10, client_recv_cnt = 0,client_sendlose_cnt = 1[0m
[0;32mI (97835) GATTC_COEX: start write =  97835ms
[0m
[0;3
3mW (97835) BT_LOG: esp_ble_gattc_write_char, The connection not created.[0m
[0;32mI (97835) GATTC_COEX: Gattc_Read_Time =  97837ms, client_send_cnt = 10, client_recv_cnt = 0,client_sendlose_cnt = 2[0m

host: one MCU with ESP32-C3 and one ESP32-C3-DevKitC-02v1.1 and one APP
ESP32-C3-DevKitC-02v1.1 is ble50_gattc_gattx_coex 
MCU  ESP32-C3 is server
APP is client

mantriyogesh, 2023-12-18:
This doesn't look like ESP-Hosted issue, as you are experiencing the problem in ESP-IDF API.

I suspect you have issue with ESP-IDF example and also able to see the issue is already created/open on https://github.com/espressif/esp-idf/issues/12803

If nothing is to be done from ESP-Hosted perspective, we can close this issue?

monkey-qiu, 2023-12-18:
> This doesn't look like ESP-Hosted issue, as you are experiencing the problem in ESP-IDF API.
> 
> I suspect you have issue with ESP-IDF example and also able to see the issue is already created/open on [espressif/esp-idf#12803](https://github.com/espressif/esp-idf/issues/12803)
> 
> If nothing is to be done from ESP-Hosted perspective, we can close this issue?

那我应该用那个版本的IDF，我的idf版本是5.1

mantriyogesh, 2023-12-18:
Are you using ESP-Hosted?

Can you please explain, in block diagram, how you are using ESP-Hosted?

monkey-qiu, 2023-12-18:
> Are you using ESP-Hosted?
> 
> Can you please explain, in block diagram, how you are using ESP-Hosted?

windows 版本，离线模式安装的IDF
下载地址：https://dl.espressif.cn/dl/esp-idf/
[image]

mantriyogesh, 2023-12-18:
Which firmware do you flash?

Which **directory** or **folder** you run idf.py flash monitor?

monkey-qiu, 2023-12-18:
> Which firmware do you flash?
> 
> Which **directory** or **folder** you run idf.py flash monitor?

目录：\Espressif\frameworks\esp-idf-v5.1.1\examples\bluetooth\bluedroid\ble_50\ble50_gattc_gattx_coex
demo是乐鑫同事给的一个，因为官方没有蓝牙5.0主从一体的demo

mantriyogesh, 2023-12-18:
Yes. Okay.

ESP-Hosted supports master and slave.
If you wish to try ESP as slave and ESP as master, we are already developing ESP-Hosted based solution, but it is in development yet.
https://github.com/espressif/esp-hosted/tree/feature/esp_as_mcu_host

Here for now, Wi-Fi is only yet supported.
Bluetooth is expected to support in a month or so.

But the issue you had raised is ESP-IDF (code) specific, which is already open. So for now, we cannot do much on this.

monkey-qiu, 2023-12-18:
> Yes. Okay.
>
> ESP-Hosted supports master and slave. If you wish to try ESP as slave and ESP as master, we are already developing ESP-Hosted based solution, but it is in development yet. [...]

好的，谢谢你

closed 2023-12-18 by mantriyogesh