espressif/esp-hosted #303: Can I use esp32h2?

teleger, 2023-12-20:
Can I use esp32h2?
Now the cpu is needed to connect the esp32h2, can I use it? Can a linux cpu use this mode

mantriyogesh, 2023-12-20:
ESP32-H2 doesn't support Wi-Fi. So yet not added as ESP-Hosted **slave**.

teleger, 2023-12-21:
> ESP32-H2 doesn't support Wi-Fi. So yet not added as ESP-Hosted **slave**.

yes，i need zigbee functions, so , i used esp32h2,but cpu connected ,now use esp32s3 + esp32h2 is ok, can i use it?
I don't know if I made myself clear, but I want to use a cpu instead of my esp32s3 device.
Thank you for your quick response. Thank you~

mantriyogesh, 2023-12-21:
So you need to offload zigbee, instead of conventional WiFi and Bluetooth?
ESP-Hosted is currently supported for Wi-Fi and Bluetooth offload.

But RPC Design being generic, It can be extended for Zigbee.
We have never ported Zigbee on ESP-Hosted. (and not planned in early future). If you wish, you can extend it and we can guide how to do it.
In fact it should not be much difficult (more of mechanical work).
You can figure out essential zigbee APIs and just port them in RPC , just similar to other RPC calls.

If you need out of the box code you will have to long wait..

closed 2023-12-21 by teleger