espressif/esp-hosted #304: esp_reset  esp_EN is low

hustzju, 2023-12-21:
Hi, I have a question,

mantriyogesh, 2023-12-21:
High-> delay-> low-> delay->high should also have same effect

hustzju, 2023-12-22:
I have followed your suggestion,High-> delay-> low-> delay->high. ESP32 is enabled ， but have not got
any event，the espsta0 doesn't appear.

host side log :
[50226.809747] esp32_spi:esp_reset: Triggering ESP reset.
[50227.046728] esp32_spi:spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]

ESP32 side log:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7096
load:0x40078000,len:15584
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
[0;32mI (31) boot: ESP-IDF v5.1.1-1-gd3c99ed-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time Dec 15 2023 16:58:37[0m
[0;32mI (33) boot: Multicore bootloader[0m
[0;32mI (37) boot: chip revision: v1.0[0m
[0;32mI (41) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (45) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (50) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (54) boot: Enabling RNG early entropy source...[0m
[0;32mI (60) boot: Partition Table:[0m
[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (71) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (93) boot:  3 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (100) boot:  4 ota_0            OTA app          00 10 00110000 00100000[0m
[0;32mI (108) boot:  5 ota_1            OTA app          00 11 00210000 00100000[0m
[0;32mI (116) boot: End of partition table[0m
[0;32mI (120) boot: Defaulting to factory image[0m
[0;32mI (124) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=20860h (133216) map[0m
[0;32mI (181) esp_image: segment 1: paddr=00030888 vaddr=3ffbdb60 size=05430h ( 21552) load[0m
[0;32mI (190) esp_image: segment 2: paddr=00035cc0 vaddr=40080000 size=0a358h ( 41816) load[0m
[0;32mI (207) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=8c878h (575608) map[0m
[0;32mI (415) esp_image: segment 4: paddr=000cc8a0 vaddr=4008a358 size=14cc0h ( 85184) load[0m
[0;32mI (467) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (467) boot: Disabling RNG early entropy source...[0m
[0;32mI (478) cpu_start: Multicore app[0m
[0;32mI (479) cpu_start: Pro cpu up.[0m
[0;32mI (479) cpu_start: Starting app cpu, entry point is 0x40081448[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (497) cpu_start: Pro cpu start user code[0m
[0;32mI (497) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (497) cpu_start: Application information:[0m
[0;32mI (501) cpu_start: Project name:     network_adapter[0m
[0;32mI (507) cpu_start: App version:      release/ng-v1.0.2-134-gce3c50a-[0m
[0;32mI (514) cpu_start: Compile time:     Dec 15 2023 16:58:28[0m
[0;32mI (520) cpu_start: ELF file SHA256:  b1d8f6e4525c260a...[0m
[0;32mI (526) cpu_start: ESP-IDF:          v5.1.1-1-gd3c99ed-dirty[0m
[0;32mI (532) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (537) cpu_start: Max chip rev:     v3.99 [0m
[0;32mI (542) cpu_start: Chip rev:         v1.0[0m
[0;32mI (547) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (554) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (560) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (566) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (572) heap_init: At 3FFC72D0 len 00018D30 (99 KiB): DRAM[0m
[0;32mI (578) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (585) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (591) heap_init: At 4009F018 len 00000FE8 (3 KiB): IRAM[0m
[0;32mI (599) spi_flash: detected chip: generic[0m
[0;32mI (602) spi_flash: flash io: dio[0m
[... 23 lines cut ...]
I (702) wifi:Init dynamic tx buffer num: 64
I (706) wifi:Init static rx buffer size: 1600
I (710) wifi:Init static rx buffer num: 16
I (712) wifi:Init dynamic rx buffer num: 32
[0;32mI (716) wifi_init: rx ba win: 6[0m
[0;32mI (718) wifi_init: tcpip mbox: 32[0m
[0;32mI (722) wifi_init: udp mbox: 6[0m
[0;32mI (724) wifi_init: tcp mbox: 6[0m
[0;32mI (728) wifi_init: tcp tx win: 5744[0m
[0;32mI (730) wifi_init: tcp rx win: 5744[0m
[0;32mI (734) wifi_init: tcp mss: 1440[0m
[0;32mI (738) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (742) wifi_init: WiFi RX IRAM OP enabled[0m
[0;32mI (746) BTDM_INIT: BT controller compile version [946b762][0m
[0;32mI (750) BTDM_INIT: Bluetooth MAC: 84:cc:a8:63:13:1a[0m
[0;32mI (756) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07[0m
[0;32mI (1074) FW_MAIN: ESP Bluetooth MAC addr: 84-cc-a8-63-13-1a
[0m
[0;32mI (1076) FW_SPI: Using SPI interface[0m
[0;32mI (1076) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (1082) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m
[0;32mI (2092) FW_MAIN: Initial set up done[0m
[0;32mI (2092) main_task: Returned from app_main()[0m

Could you give me some advice?

mantriyogesh, 2024-01-10:
@kapilkedawat