espressif/esp-hosted #307: Is there any api could let the master get the firmware version from slave ?

LuciaferKelvin, 2024-01-02:
As tile.

mantriyogesh, 2024-01-02:
Hello @LuciaferKelvin ,

This is something we had done before, but on older commit, so would not be applicable on master.
If you want it as reference, let us know, will attach that.

The logic is  farly simple, use existing 'INIT' event from slave to host.
Add the firmware version using 3 additional bytes in TLV format.
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L316

and parse them init event in host at
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_driver/esp32/spi/esp_spi.c#L210

LuciaferKelvin, 2024-01-02:
Hello @mantriyogesh  ,
Sorry, my bad.
Is the release V0.4 has any the api which could get the firmware version form slave?

mantriyogesh, 2024-01-02:
Hello @LuciaferKelvin 

No release/v0.4 doesn't have such API.
But if you wish, we can add such API for you.

So the app will request firmware and get response as firmware version.
Is this what you need or something else?

If yes, let me know:
1. base firmware binary or base git commits you use at slave and host

We are blocked on other high priority issues right now, this might get delayed by week or so.

LuciaferKelvin, 2024-01-02:
Oh, Thank you.
I just want to confirm that if it has any api could get version detail from the slave.
And I will tell you , if I need the api like this.

closed 2024-01-02 by LuciaferKelvin