espressif/esp-hosted #309: udhcpc -i espsta0 kernel panic

xmlhp, 2024-01-08:
[1106_esp_panic.zip](https://github.com/espressif/esp-hosted/files/13855787/1106_esp_panic.zip)
Please unzip the attachment to view the operation diary file.
1. First, I shut down udhcpc
2. I kill wpa_supplicant
3.ifconfig espsta0 down
4.ifconfig espsta0 up
Enable wpa_supplicant
6. udhcp-i espsta0, to reacquire the IP, this time the driver will panic

Please help to see, why the crash?
How to down espsta0, and then enable espsta0 normally

mantriyogesh, 2024-01-08:
@kapilkedawat

kapilkedawat, 2024-01-08:
@xmlhp Please recompile the module using following changes in Makefile.

--- a/esp_hosted_ng/host/Makefile
+++ b/esp_hosted_ng/host/Makefile
@@ -32,7 +32,7 @@ ifeq ($(CONFIG_INFO_LOGS), y)
 endif
 
 ifeq ($(CONFIG_DEBUG_LOGS), y)
-       EXTRA_CFLAGS += -DCONFIG_INFO_LOGS
+       EXTRA_CFLAGS += -DCONFIG_INFO_LOGS -g
        EXTRA_CFLAGS += -DCONFIG_DEBUG_LOGS
 endif
 
Please reproduce the crash after that and share *.ko file and panic logs.

Please also provide following info:
1. kernel version
2. arch (arm64/arm)

xmlhp, 2024-01-09:
[esp32c3.zip](https://github.com/espressif/esp-hosted/files/13867693/esp32c3.zip)
esp32c3.zip, is the driver file in my kernel. The.ko file is there, too.
1.kernel version 5.10
2.arch: arm
3. kernel panic log  see [1106_esp_panic.zip](https://github.com/espressif/esp-hosted/files/13855787/1106_esp_panic.zip)