espressif/esp-hosted #310: C3 Can't it reconnect automatically?

xmlhp, 2024-01-10:
1.C3 connection hotspot successfully obtains IP and can PING external network, such as www.baidu.com
2. I turn off the hotspot, at this time, ifconfig -a; espsta0 still has IP, but ping www.baidu.com can't work.
3. I open the hotspot and ping www.baidu.com, but it still doesn't work.
Question: Can't c3 automatically determine the network condition and change the connection state? Won't it reconnect automatically?

Rainbow-Cai, 2024-01-10:
Hi @kapilkedawat  Could you please help this questions?

kapilkedawat, 2024-01-10:
@xmlhp what's the commit version that you are using?

xmlhp, 2024-01-10:
This is the git log info:
commit ce3c50a33fa4bc562a1b6cbcee292c1ae0b0a404 (HEAD -> master, origin/master, origin/HEAD)
Merge: d29c683 c4843f1
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Thu Nov 23 16:47:44 2023 +0800

    Merge branch 'fix/spi_sdio_uart_c2_documentation' into 'master'

    Fix/spi sdio uart c2 documentation

    See merge request app-frameworks/esp_hosted!362

kapilkedawat, 2024-01-10:
Could you please pull latest changes and retry?

xmlhp, 2024-01-10:
commit 504210623ec632efae1d39ad2636e235485ab3d4 (HEAD -> master, origin/master, origin/HEAD)
Merge: 0d434c5 58c0e50
Author: Yogesh Mantri <yogesh.mantri@espressif.com>
Date:   Tue Jan 9 00:27:28 2024 +0800

    Merge branch 'bugfix/hosted_ng_cleanup' into 'master'

    esp_hosted_ng: Host side code cleanup

    See merge request app-frameworks/esp_hosted!378

I get the last version from github, but compile failure:

drivers/net/wireless/esp32c3/esp_utils.c:9:10: fatal error: esp_wpa_utils.h: No such file or directory
 #include "esp_wpa_utils.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[6]: *** [scripts/Makefile.build:273: drivers/net/wireless/esp32c3/esp_utils.o] Error 1
make[6]: *** Waiting for unfinished jobs....
drivers/net/wireless/esp32c3/main.c: In function 'add_network_iface':
drivers/net/wireless/esp32c3/main.c:411:8: error: implicit declaration of function 'esp_cfg80211_register'; did you mean 'esp_cfg80211_add_iface'? [-Werror=implicit-function-declaration]
  ret = esp_cfg80211_register(adapter);
        ^~~~~~~~~~~~~~~~~~~~~
        esp_cfg80211_add_iface
drivers/net/wireless/esp32c3/main.c: In function 'process_esp_bootup_event':
drivers/net/wireless/esp32c3/main.c:542:2: error: implicit declaration of function 'process_event_esp_bootup'; did you mean 'process_esp_bootup_event'? [-Werror=implicit-function-declaration]
  process_event_esp_bootup(adapter, evt->data, evt->len);
  ^~~~~~~~~~~~~~~~~~~~~~~~
  process_esp_bootup_event

kapilkedawat, 2024-01-10:
Apologies for this, we will push this fix in sometime. you can use following changes for now

`
diff --git a/esp_hosted_ng/host/esp_utils.c b/esp_hosted_ng/host/esp_utils.c
index 71898c1..56e2b64 100644
--- a/esp_hosted_ng/host/esp_utils.c
+++ b/esp_hosted_ng/host/esp_utils.c
@@ -6,7 +6,7 @@
  *
  */
 #include "utils.h"
-#include "esp_wpa_utils.h"
+#include "esp_utils.h"
 
 
 int wpa_cipher_to_alg(int cipher)
`
Rest of the errors doesn't relate to repo, please make sure your main.c file has been updated correctly.

mantriyogesh, 2024-02-20:
Can we close this @xmlhp ?