espressif/esp-hosted #312: imx8mm use ESP32-WROOM-32E module, Driver loading error

mengleia, 2024-01-10:
cpu: imx8mm
system: linux kernel 6.1.36
esp driver: ESP-Hosted-NG_release_v1.0.2
esp-hosted:  latest commit 504210623ec632efae1d39ad2636e235485ab3d4

insmod esp32_sdio.ko error
[image]
mmc log
[image]

[esp_log.txt](https://github.com/espressif/esp-hosted/files/13885874/esp_log.txt)
[kernel_log.txt](https://github.com/espressif/esp-hosted/files/13885877/kernel_log.txt)

mantriyogesh, 2024-01-10:
+ @kapilkedawat

kapilkedawat, 2024-01-10:
@mengleia Please make sure the connections are correct, and that the pull-up requirements are met.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html

mengleia, 2024-01-11:
> @mengleia Please make sure the connections are correct, and that the pull-up requirements are met. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html

Thank you for your reply
I have Burn the flash voltage selection eFuses and pull-up resistor to GPIO12, Our hardware has already met the requirements pull-up, But it's still the same error. 
[image]
I see an example of using imx8mm. Do you have any relevant configuration references？

mengleia, 2024-01-11:
@mantriyogesh 
mmc device log
[image]
Please tell me how to analyze the problem？

mantriyogesh, 2024-01-11:
Setup schematic and picture please?

mengleia, 2024-01-11:
@mantriyogesh 
thank you for your reply, I have successfully loaded the driver.  
the dts cannot use "non-removable" options.

mantriyogesh, 2024-01-11:
Try `broken-cd`