espressif/esp-hosted #313: NG: ESP32c3 used in OpenWRT v21 - error in cfg80211 during insmod

Philipansari, 2024-01-11:
I ported the esp-hosted driver (v.1.0.2) to OpenWRT-Linux System (v21) and tried to insert the kernel module. Unfortunately i get the following dmesg traceback errors (see at the end). 

Three warnings are displayed and the /net/wireless/core.c source codes are referenced in lines 420 (1.), 432 (2.) and 771 (3.). Here are the corresponding source codes that trigger the warnings:

### 1. `line 420`
```{}
WARN_ON(ops->auth && (!ops->assoc || !ops->deauth || !ops->disassoc));
```

### 2. `line 432`
```{}
WARN_ON(ops->remain_on_channel && !ops->cancel_remain_on_channel);
```

### 3. `line 771` 
```{}
if (WARN_ON(!(rdev->wiphy.flags & WIPHY_FLAG_SUPPORTS_FW_ROAM) &&
		    rdev->ops->update_connect_params))
```

Here is the source-file of `/net/wireless/core.c` shown (<https://elixir.bootlin.com/linux/v5.4.154/source/net/wireless/core.c>) 

I don't understand the warnings because looking at the source code at the function `int esp_cfg80211_register(struct esp_adapter *adapter)` at <https://github.com/espressif/esp-hosted/blob/release/ng-v1.0.2/esp_hosted_ng/host/esp_cfg80211.c> 

1. the cfg80211_ops `auth`, `assoc`, `deauth` and `disassoc` are defined.
2. ops `remain_on_channel` is not defined
3. either the flag `WIPHY_FLAG_SUPPORTS_FW_ROAM` nor the ops `update_connect_params` is defined. 

Maybe im on the wrong track here?

I've already tried to find a similar problem in other issues, but I haven't found anything like it.

Here is the log of dmesg regarding the kernel module

```{}
...
[    2.958796] esp_reset, ESP32: Triggering ESP reset.
[    3.174913] ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
[    3.444112] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
[    4.567653] Received ESP bootup event
[    4.572818] EVENT: 3
[    4.575069] EVENT: 2
[    4.577261] ESP Reconfigure SPI CLK to 30 MHz
[    4.581619] EVENT: 0
[    4.583808] EVENT: 4
[    4.586022] EVENT: 1
[    4.588216] esp32: process_fw_data ESP chipset's last reset cause: 
[    4.588220] POWERON_RESET
[    4.597130] esp32: ESP Firmware version: 1.0.2
[    4.601577] ESP chipset detected [esp32-c3]
[    4.605825] ------------[ cut here ]------------
[    4.610597] WARNING: CPU: 0 PID: 989 at backports-5.10.68-1/net/wireless/core.c:420 wiphy_new_nm+0x624/0x66c [cfg80211]
[    4.621419] Modules linked in: usb_storage mclh_extension_boards esp_interface_ng cfg80211 compat leds_gpio ledtrig_transient fsl_mph_dr_of ehci_platform ehci_fsl sd_mod mc100_extension_boards mct_km_lib gpio_button_hotplug mii
[    4.641671] CPU: 0 PID: 989 Comm: kworker/0:1H Not tainted 5.4.154 #0
[    4.648137] Hardware name: Freescale i.MX6 Ultralite (Device Tree)
[    4.654361] Workqueue: ESP_EVENTS_WORKQUEUE esp_events_work [esp_interface_ng]
[    4.661640] [<8010d9a4>] (unwind_backtrace) from [<8010abec>] (show_stack+0x10/0x14)
[    4.669426] [<8010abec>] (show_stack) from [<801293ec>] (__warn+0xbc/0xd4)
[    4.676341] [<801293ec>] (__warn) from [<80129454>] (warn_slowpath_fmt+0x50/0x94)
[    4.683938] [<80129454>] (warn_slowpath_fmt) from [<7f04f874>] (wiphy_new_nm+0x624/0x66c [cfg80211])
[    4.693227] [<7f04f874>] (wiphy_new_nm [cfg80211]) from [<7f09c2b8>] (esp_cfg80211_register+0x1c/0xb0 [esp_interface_ng])
[    4.704262] [<7f09c2b8>] (esp_cfg80211_register [esp_interface_ng]) from [<7f099aec>] (esp_add_card+0x24/0xb8 [esp_interface_ng])
[    4.715985] [<7f099aec>] (esp_add_card [esp_interface_ng]) from [<7f09cd20>] (process_event_esp_bootup+0x19c/0x32c [esp_interface_ng])
[    4.728142] [<7f09cd20>] (process_event_esp_bootup [esp_interface_ng]) from [<7f099a48>] (esp_events_work+0xa0/0xb4 [esp_interface_ng])
[    4.740367] [<7f099a48>] (esp_events_work [esp_interface_ng]) from [<8013e1bc>] (process_one_work+0x1d4/0x43c)
[    4.750406] [<8013e1bc>] (process_one_work) from [<8013e47c>] (worker_thread+0x58/0x59c)
[    4.758538] [<8013e47c>] (worker_thread) from [<80142948>] (kthread+0x144/0x148)
[    4.765972] [<80142948>] (kthread) from [<801010e8>] (ret_from_fork+0x14/0x2c)
[    4.773201] Exception stack(0x9eb71fb0 to 0x9eb71ff8)
[    4.778292] 1fa0:                                     00000000 00000000 00000000 00000000
[... 37 lines cut ...]
[    5.106838] [<7f099aec>] (esp_add_card [esp_interface_ng]) from [<7f09cd20>] (process_event_esp_bootup+0x19c/0x32c [esp_interface_ng])
[    5.118991] [<7f09cd20>] (process_event_esp_bootup [esp_interface_ng]) from [<7f099a48>] (esp_events_work+0xa0/0xb4 [esp_interface_ng])
[    5.131215] [<7f099a48>] (esp_events_work [esp_interface_ng]) from [<8013e1bc>] (process_one_work+0x1d4/0x43c)
[    5.141252] [<8013e1bc>] (process_one_work) from [<8013e47c>] (worker_thread+0x58/0x59c)
[    5.149379] [<8013e47c>] (worker_thread) from [<80142948>] (kthread+0x144/0x148)
[    5.156810] [<80142948>] (kthread) from [<801010e8>] (ret_from_fork+0x14/0x2c)
[    5.164040] Exception stack(0x9eb71fb0 to 0x9eb71ff8)
[    5.169121] 1fa0:                                     00000000 00000000 00000000 00000000
[    5.177328] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    5.185534] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[    5.192155] ---[ end trace cceddc1ca24c8671 ]---
[    5.196793] Failed to register with cfg80211 (err code 0xffffffea)
[    5.202977] esp32: add_network_iface() failure, ret: -22
[    5.208313] network iterface init failed
[    5.212239] ESP peripheral capabilities: 0xe8
[    5.434509] ESP Bluetooth init
[    5.437943] Capabilities: 0xe8. Features supported are:
[    5.443179]   * WLAN on SPI
[    5.446035]   * BT/BLE
[    5.448398]     - HCI over SPI
[    5.451456]     - BLE only
[    6.720436] urngd: v1.0.2 started.
[    6.768395] kmodloader: loading kernel modules from /etc/modules.d/*
...
```

mantriyogesh, 2024-01-11:
@kapilkedawat ++

kapilkedawat, 2024-01-11:
@Philipansari any reason for not using latest driver? Could you please rebase your changes on latest driver and check once. Also please share the changes if you have made any in the wifi driver.

Philipansari, 2024-01-11:
You mean latest esp-hosted driver? thought i do so, or should i checkout to master branch? Yesterday there were some compilation errors so i preferred to use the release version. 

I just tried to cross-compile the kernel module.

There were some compilation errors for my specific kernel version (5.4.154)

- `wiphy_lock` and `wiphy_unlock` are first available in kernel version 5.12, i commented them out in main.c line 504, 506. Now i am able to build the module. 

However on the esp32 side there is a problem during the compilation of `network_adapter`: 

```
/home/philip/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(cmd.c.obj): in function `sta_connection':
/home/philip/git/dpm/es/eli/esp-hosted-ng/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:128: undefined reference to `esp_wifi_connect_internal'
/home/philip/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(cmd.c.obj): in function `process_auth_request':
/home/philip/git/dpm/es/eli/esp-hosted-ng/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:1110: undefined reference to `esp_wifi_issue_auth_internal'
/home/philip/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: esp-idf/main/libmain.a(cmd.c.obj): in function `process_assoc_request':
/home/philip/git/dpm/es/eli/esp-hosted-ng/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:1234: undefined reference to `esp_wifi_issue_assoc_internal'
collect2: error: ld returned 1 exit status
```

Running the latest driver on the linux-side and the release version v1.0.2 on the esp-side is resulting in basically the same errors, see dmesg log:
```
...
[    3.165210] esp_interface_latest:spi_dev_init: ESP32 peripheral is registered to SPI bus [0],chip select [0], SPI Clock [10]
[    3.444287] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
[    3.462213] init: - preinit -
[    3.754067] random: procd: uninitialized urandom read (4 bytes read)
[    3.808356] mount_root: mounting /dev/root
[    3.832216] urandom-seed: Seeding with /etc/urandom.seed
[    3.876826] procd: - early -
[    3.879918] procd: - watchdog -
[    4.556155] esp_interface_latest:process_esp_bootup_event: Received ESP bootup event
[    4.563948] esp_interface_latest:process_event_esp_bootup: Bootup Event tag: 3
[    4.571226] esp_interface_latest:esp_validate_chipset: Chipset=ESP32-C3 ID=05 detected over SPI
[    4.579951] esp_interface_latest:process_event_esp_bootup: Bootup Event tag: 2
[    4.587199] esp_interface_latest:adjust_spi_clock: ESP Reconfigure SPI CLK to 30 MHz
[    4.594963] esp_interface_latest:process_event_esp_bootup: Bootup Event tag: 0
[    4.602189] esp_interface_latest:process_event_esp_bootup: Bootup Event tag: 4
[    4.609434] esp_interface_latest:process_event_esp_bootup: Unsupported tag=4 in bootup event
[    4.617891] esp_interface_latest:process_event_esp_bootup: Bootup Event tag: 1
[    4.625141] esp_interface_latest:process_fw_data: ESP chipset's last reset cause:
[    4.632627] esp_interface_latest:print_reset_reason: POWERON_RESET
[    4.638836] esp_interface_latest:check_esp_version: ESP Firmware version: 1.0.2
[    4.646209] ------------[ cut here ]------------
[    4.650981] WARNING: CPU: 0 PID: 989 at backports-5.10.68-1/net/wireless/core.c:420 wiphy_new_nm+0x624/0x66c [cfg80211]
[    4.661800] Modules linked in: usb_storage mclh_extension_boards esp_interface_latest cfg80211 compat leds_gpio ledtrig_transient fsl_mph_dr_of ehci_platform ehci_fsl sd_mod mc100_extension_boards mct_km_lib gpio_button_hotplug mii
[    4.682389] CPU: 0 PID: 989 Comm: kworker/0:1H Not tainted 5.4.154 #0
[    4.688859] Hardware name: Freescale i.MX6 Ultralite (Device Tree)
[    4.695104] Workqueue: ESP_EVENTS_WORKQUEUE esp_events_work [esp_interface_latest]
[    4.702710] [<8010d9a4>] (unwind_backtrace) from [<8010abec>] (show_stack+0x10/0x14)
[    4.710497] [<8010abec>] (show_stack) from [<801293ec>] (__warn+0xbc/0xd4)
[    4.717410] [<801293ec>] (__warn) from [<80129454>] (warn_slowpath_fmt+0x50/0x94)
[    4.725021] [<80129454>] (warn_slowpath_fmt) from [<7f04e874>] (wiphy_new_nm+0x624/0x66c [cfg80211])
[    4.734287] [<7f04e874>] (wiphy_new_nm [cfg80211]) from [<7f098f4c>] (esp_add_wiphy+0x20/0xf0 [esp_interface_latest])
[    4.744991] [<7f098f4c>] (esp_add_wiphy [esp_interface_latest]) from [<7f09ce14>] (esp_add_card+0x20/0xe8 [esp_interface_latest])
[    4.756716] [<7f09ce14>] (esp_add_card [esp_interface_latest]) from [<7f09cf68>] (process_event_esp_bootup+0x8c/0x438 [esp_interface_latest])
[    4.769487] [<7f09cf68>] (process_event_esp_bootup [esp_interface_latest]) from [<7f09d3e8>] (esp_events_work+0xd4/0xe8 [esp_interface_latest])
[    4.782426] [<7f09d3e8>] (esp_events_work [esp_interface_latest]) from [<8013e1bc>] (process_one_work+0x1d4/0x43c)
[    4.792819] [<8013e1bc>] (process_one_work) from [<8013e47c>] (worker_thread+0x58/0x59c)
[    4.800949] [<8013e47c>] (worker_thread) from [<80142948>] (kthread+0x144/0x148)
[    4.808384] [<80142948>] (kthread) from [<801010e8>] (ret_from_fork+0x14/0x2c)
[    4.815634] Exception stack(0x9eb71fb0 to 0x9eb71ff8)
[    4.820699] 1fa0:                                     00000000 00000000 00000000 00000000
[... cut ...]
[    5.251946] esp_interface_latest:process_event_esp_bootup: network iterface init failed
[    5.260336] procd: - watchdog -
[    5.273953] procd: - ubus -
[    5.313631] random: ubusd: uninitialized urandom read (4 bytes read)
[    5.337697] random: ubusd: uninitialized urandom read (4 bytes read)
[    5.354988] random: ubusd: uninitialized urandom read (4 bytes read)
[    5.378856] procd: - init -
[    6.694094] urngd: v1.0.2 started.
[    6.755479] kmodloader: loading kernel modules from /etc/modules.d/*

...
```

Philipansari, 2024-01-11:
I've made no changes to the driver itself expect adjusting the IOs to the ones i need (via the pre-processor macros)

Philipansari, 2024-01-11:
It seems there is a mismatch between the used kernel version during compilation time (v5.4.154) and execution time on the target. (backports-5.10.68). There is a different offset in the `struct cfg80211_ops` struct containing the several functions. Considering the different offsets of the defined functions the warning messages started to make sense:

| v5.4.154        | offset in bytes | v5.10.68              |
|-----------------|-----------------|-----------------------|
| scan            | 8               | auth                  |
| connect         | 8               | update_connect_params |
| disconnect      | 8               | join_ibss             |
| add_key         | 0               |                       |
| del_key         | 0               |                       |
| set_default_key | 0               |                       |
| mgmt_tx         | 16              | set_power_mgmt        |
| auth            | 8               | assoc                 |
| deauth          | 8               | disassoc              |
| disassoc        | 8               | connect               |
| assoc           | 8               | deauth                |
| suspend         | 0               |                       |
| resume          | 0               |                       |
| set_wakeup      | 0               |                       |

Hacking the functions in the right place resolves the warning. So it probably has nothing to do with the esp-hosted repo, sorry for that.

closed 2024-01-16 by Philipansari