espressif/esp-hosted #315: esp-hosted-fg SPI can't use pin 33 as HS or DR

JAndrassy, 2024-01-13:
functions `set_handshake_gpio`, `reset_handshake_gpio`, `set_dataready_gpio` and `reset_dataready_gpio` use `WRITE_PERI_REG` which treats mask as a 32 bit value.

I want to make esp-hosted-fg work on Arduino Nano RP2040 Connect where the io available for DR and HS are io 0 and io 33.

mantriyogesh, 2024-01-13:
WRITE_PERI_REG are all slave side part.
Don't you use the slave and slave software as is?

You can always change the host, as per your need.

I am still trying to understand the problem, if you could detail a bit.

JAndrassy, 2024-01-13:
the io 33 mask for  WRITE_PERI_REG is `1 << 33` which doesn't fit into 32 bits, but WRITE_PERI_REG casts it to 32 bit.

I think correct is
```
static inline void set_handshake_gpio(void)
{
  gpio_set_level(GPIO_HS, 1);
}

static inline void reset_handshake_gpio(void)
{
  gpio_set_level(GPIO_HS, 0);
}

static inline void set_dataready_gpio(void)
{
  gpio_set_level(GPIO_DR, 1);
}

static inline void reset_dataready_gpio(void)
{
  gpio_set_level(GPIO_DR, 0);
}
```

mantriyogesh, 2024-01-14:
Yes. You are correct.
Either use above changes, or use
`1ULL << 33`

JAndrassy, 2024-01-14:
just 1ll is not enough
```
#define WRITE_PERI_REG(addr, val) do {                                                                                 \
            ASSERT_IF_DPORT_REG((addr), WRITE_PERI_REG);                                                               \
            (*((volatile uint32_t *)ETS_UNCACHED_ADDR(addr))) = (uint32_t)(val);                                       \
        } while(0)
```
there is `(uint32_t)(val); `

mantriyogesh, 2024-01-14:
@JAndrassy ,

Yes, you are absolutely correct.
We will correct the master with this change.

JAndrassy, 2024-03-06:
I see now, there is already have a PR https://github.com/espressif/esp-hosted/pull/248/

mantriyogesh, 2026-09-28:
This issue is fixed. The co-processor no longer uses `WRITE_PERI_REG()` with a 32-bit mask to drive the Handshake and Data Ready pins. It now uses `gpio_set_level()`, so you can use GPIO 32 and higher for these pins.

- ESP-Hosted-FG: fixed in [5c267b5](https://github.com/espressif/esp-hosted/commit/5c267b5dc887b10cff3ecdf8633e95866d9601d9), [spi_slave_api.c#L249-L252](https://github.com/espressif/esp-hosted/blob/5c267b5dc887b10cff3ecdf8633e95866d9601d9/esp_hosted_fg/esp/esp_driver/network_adapter/main/spi_slave_api.c#L249-L252)
- ESP-Hosted-MCU 2.x: [slave/main/spi_slave_api.c#L257-L260 @ v2.12.13](https://github.com/espressif/esp-hosted-mcu/blob/v2.12.13/slave/main/spi_slave_api.c#L257-L260)
- ESP-Hosted-MCU 3.x: [eh_cp_transport_spi.c#L314-L319 @ v3.0.9](https://github.com/espressif/esp-hosted-mcu/blob/v3.0.9/coprocessor/eh_cp_transport/src/eh_cp_transport_spi.c#L314-L319)

PR #248 was closed and not merged. The same fix is now in the main code.

ESP-Hosted-FG is legacy and continues as [esp-hosted-mcu](https://github.com/espressif/esp-hosted-mcu). For new designs, please use esp-hosted-mcu. If you still see the problem there, please open an issue in that repository.

closed 2026-09-28 by mantriyogesh