espressif/esp-hosted #316: Porting issue of “feature/esp_as_mcu_host” branch

Evlers, 2024-01-15:
@mantriyogesh @SohKamYung-Espressif
Before the migration of this branch began, I noticed that the esp-idf type and api were used in many places.
I couldn't port all the files from esp-idf to the project!!
In my opinion, all calls within the api and drivers folders should be made based on the wrapper, otherwise it would be difficult to use this branch as a git submodule.
I remember you mentioned before that there are internal branches in progress, can you share them to speed up my transplantation?

Evlers, 2024-01-15:
Some files I didn't even find in the esp-idf repository, such as esp_wifi_remote.h

mantriyogesh, 2024-01-15:
[esp_wifi_remote.tgz](https://github.com/espressif/esp-hosted/files/13934596/esp_wifi_remote.tgz)

mantriyogesh, 2024-01-15:
> In my opinion, all calls within the api and drivers folders should be made based on the wrapper, otherwise it would be difficult to use this branch as a git submodule.

Yes yes.. We have this plan to do, this will support any other MCUs then.
Actually, if you see, we used to support agnostic of ESP-IDF earlier.
But for ESP32-P4 as host and some other ESP chipset as slave, we wanted to just re-use all the ESP-IDF power.
ESP32-P4 is by the way new chipset, without Wi-Fi / Bluetooth. and relied on slave chipset like C3/C6/etc for getting Wi-Fi and Bluetooth.

This is later change, where we intentionally break, for easy re-use and faster development.
Anyway, we attempted to decouple the ESP-IDF, which is also in https://github.com/espressif/esp-hosted/issues/311#issuecomment-1884488097

But just that we are not able to find proper time to get these up for MCU neutral solution again.

mantriyogesh, 2024-01-15:
In short, can it work without ESP-IDF, definitely. You yourself are not bound tto ESP-IDF at present for your working solution based on master.

Just that we needed to focus on feature/esp_as_mcu_host for ESP32-P4, the intermediate branch in https://github.com/espressif/esp-hosted/issues/311#issuecomment-1884488097 is lagging.

mantriyogesh, 2024-01-15:
Some APIs can be bypassed.

esp_wifi_remote is wrapper for esp_wifi.
For MCU, you can actually bypass like stated in: 
https://github.com/espressif/esp-hosted/issues/270#issuecomment-1880575410

Evlers, 2024-01-15:
Ok, I have patched the 4840528 commit of the master branch to see if this is more suitable for porting to other MCUS

mantriyogesh, 2024-01-15:
Built ? Tested?
Pushed somewhere?

HubertXie, 2024-02-03:
> 好的，我已经修补了 master 分支的[4840528](https://github.com/espressif/esp-hosted/commit/4840528810457f393e0e65fe2bb1442dcb6dbc10)提交，看看这是否更适合移植到其他 MCU

大神，你这个移植的怎么样了，我也正在做这个事情，基于rtt系统移植hosted

Evlers, 2024-02-03:
> > 好的，我已经修补了 master 分支的[4840528](https://github.com/espressif/esp-hosted/commit/4840528810457f393e0e65fe2bb1442dcb6dbc10)提交，看看这是否更适合移植到其他 MCU
> 
> 大神，你这个移植的怎么样了，我也正在做这个事情，基于rtt系统移植hosted

这个没时间搞了。目前只移植了主分支勉强用着，准备换英飞凌方案，后面等乐鑫完成新的分支再适配。

gavin-hy, 2024-05-09:
@Evlers  你们使用蓝牙了嘛？不使用蓝牙的话，可以使用MXCHIP的EMW3080，比这个简单方便，用的是LWIP和FreeRTOS，就是WiFi端没有程序，只有bin文件，稳定性是没有问题的！

Evlers, 2024-05-10:
> @Evlers 你们使用蓝牙了嘛？不使用蓝牙的话，可以使用MXCHIP的EMW3080，比这个简单方便，用的是LWIP和FreeRTOS，就是WiFi端没有程序，只有bin文件，稳定性是没有问题的！

有些项目不需要蓝牙的。
这款模组用的什么芯片，单价多少呢，我们目前用英飞凌的方案也很稳定的，就是单价13含税有点贵了

gavin-hy, 2024-05-11:
> 有些项目不需要蓝牙的。 这款模组用的什么芯片，单价多少呢，我们目前用英飞凌的方案也很稳定的，就是单价13含税有点贵了
庆科自研的芯片，出货量很大，稳定性问题不大，出了很多年了，不支持WPA3，你们目前使用的英飞凌的哪款芯片？带蓝牙吗？模组本身带flash吗？