espressif/esp-hosted #317: How to change wifi, Bluetooth MAC address

xmlhp, 2024-01-18:
We need to use the MAC address we bought.
How to change the address? 
How to permanently save the address to esp32c3?

mantriyogesh, 2024-01-18:
Please check https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html

xmlhp, 2024-01-18:
This was changed in the firmware. How do I change through the driver on linux？

xmlhp, 2024-01-18:
# ifconfig -a
espsta0   Link encap:Ethernet  HWaddr C0:4E:30:84:AA:FC  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

# # ifconfig espsta0 hw ether c0:4E:30:84:AA:FE
# ifconfig -a
espsta0   Link encap:Ethernet  HWaddr C0:4E:30:84:AA:FC  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

# # ifconfig espsta0 hw ether c0:4E:30:84:AA:FE
# # ifconfig espsta0 up
# ifconfig -a
espsta0   Link encap:Ethernet  HWaddr C0:4E:30:84:AA:FC  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

# 
I can't config the hwadder by ifconfig espsta0 hw ether xx:xx:xx:xx:xx

mantriyogesh, 2024-01-18:
For Wi-Fi, the call is available, [wifi_set_mac()](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/common/ctrl_apis.md#16-ctrl_cmd_t--wifi_set_macctrl_cmd_t-req).

[test_set_mac_addr()](https://github.com/espressif/esp-hosted/blob/26f4fa56bb2dd8504cc24efefa03a4e5ffcd0535/esp_hosted_fg/host/linux/host_control/c_support/test_utils.c#L673) can also be used, as it calls 'wifi_set_mac()' internally.

But this clearly needs to be agnostic and need to support Bluetooth.
For bluetooth, yet not available, but little change you can re-use this API for bluetooth also.

mantriyogesh, 2024-01-18:
Make sure that the espsta0 mac address if for the 'STA' interface MAC address at ESP.
Otherwise you will face issues in delivering packets correctly.

xmlhp, 2024-01-18:
Make sure that the espsta0 mac address if for the 'STA' interface MAC address at ESP.
I don't understand. What does that mean.

mantriyogesh, 2024-01-18:
Right, https://github.com/espressif/esp-hosted/blob/26f4fa56bb2dd8504cc24efefa03a4e5ffcd0535/esp_hosted_ng/host/main.c#L349
 
 https://github.com/espressif/esp-hosted/blob/26f4fa56bb2dd8504cc24efefa03a4e5ffcd0535/esp_hosted_fg/host/linux/host_driver/esp32/main.c#L97
 
 esp_set_mac_address should be automatically used, when you set mac address on interface espsta0.
 So the thing, you had tried, should have worked.
 
1. Can you please share ESP log
2. Which git commit do you use and what is the host git commit?
 
 There is also restriction of MAC address . It is JFYI, you are not using such mac address anyway:
 
[image]

xmlhp, 2024-01-18:
Then, how set the mac for bluetooth?

mantriyogesh, 2024-01-18:
For Bluetooth yet not supported, need to add this support.