espressif/esp-hosted #318: imx6+ng  sdio  probe error

hustzju, 2024-01-18:
I have successfully implemented the SPI solution On  my imx6ul board,and now I want to change to SDIO mode,
but I encounter some problems.
if I pull up IO12,IO13,IO14,IO2,IO4, when enable ESP32, it will  restart repeatly. log like this:
rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7096
load:0xffffffff,len:-1
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

BUT if  I don't pull up  GPIOs, when load esp32 driver, Host will get errors like this:
[   61.211828] esp_sdio: probe of mmc1:0001:1 failed with error -110
[   61.218605] esp_sdio: probe of mmc1:0001:2 failed with error -110

I try to pull up the GPIOS after esp32 has reset, but no event could be received  including boot event.
Could you give me some advice?

mantriyogesh, 2024-01-18:
Check https://docs.espressif.com/projects/esp-idf/en/v4.2.3/esp32/api-reference/peripherals/sd_pullup_requirements.html#conflicts-between-bootstrap-and-sdio-on-dat2
And other issues on same page for using ESP32 as sdio slave.

hustzju, 2024-01-19:
how to burn efuse?
when I run cmd "components/esptool_py/esptool/espefuse.py  --port /dev/ttyUSB0 set_flash_voltage 3.3V", the terminal shell 
will stuck,then get error:
import: not authorized `subprocess' @ error/constitute.c/WriteImage/1028.
import: not authorized `sys' @ error/constitute.c/WriteImage/1028.
components/esptool_py/esptool/espefuse.py:行11: 未预期的符号 `subprocess.run' 附近有语法错误
components/esptool_py/esptool/espefuse.py:行11: `    sys.exit(subprocess.run([sys.executable, '-m', 'espefuse'] + sys.argv[1:]).returncode)'

by the way, before to burn efuse, ESP32 must be in download mode? 
But even if I do that ,it still doesn't work

hustzju, 2024-01-19:
I have solve the above problem  by  burning it on Windows , I don't know why it  don't work on Ubuntu.
But the issue  is still present, host log ：
[ 1861.384683] esp32_sdio: loading out-of-tree module taints kernel.
[ 1861.415028] esp32_sdio:esp_reset: triggering ESP reset.

esp32 log：
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7096
load:0x40078000,len:15584
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3876
entry 0x4008064c
[0;32mI (31) boot: ESP-IDF v5.1.1-1-gd3c99ed-dirty 2nd stage bootloader[0m
[0;32mI (31) boot: compile time Dec 15 2023 16:46:18[0m
[0;32mI (32) boot: Multicore bootloader[0m
[0;32mI (36) boot: chip revision: v1.1[0m
[0;32mI (40) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (45) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (49) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (54) boot: Enabling RNG early entropy source...[0m
[0;32mI (59) boot: Partition Table:[0m
[0;32mI (63) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (70) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (78) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (85) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (93) boot:  3 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (100) boot:  4 ota_0            OTA app          00 10 00110000 00100000[0m
[0;32mI (108) boot:  5 ota_1            OTA app          00 11 00210000 00100000[0m
[0;32mI (115) boot: End of partition table[0m
[0;32mI (119) boot: Defaulting to factory image[0m
[0;32mI (124) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=20538h (132408) map[0m
[0;32mI (181) esp_image: segment 1: paddr=00030560 vaddr=3ffbdb60 size=0542ch ( 21548) load[0m
[0;32mI (189) esp_image: segment 2: paddr=00035994 vaddr=40080000 size=0a684h ( 42628) load[0m
[0;32mI (207) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=8c004h (573444) map[0m
[0;32mI (414) esp_image: segment 4: paddr=000cc02c vaddr=4008a684 size=14870h ( 84080) load[0m
[0;32mI (465) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (465) boot: Disabling RNG early entropy source...[0m
[0;32mI (477) cpu_start: Multicore app[0m
[0;32mI (477) cpu_start: Pro cpu up.[0m
[0;32mI (477) cpu_start: Starting app cpu, entry point is 0x4008147c[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (497) cpu_start: Pro cpu start user code[0m
[0;32mI (497) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (497) cpu_start: Application information:[0m
[0;32mI (502) cpu_start: Project name:     network_adapter[0m
[0;32mI (508) cpu_start: App version:      release/ng-v1.0.2-134-gce3c50a-[0m
[0;32mI (515) cpu_start: Compile time:     Dec 15 2023 16:46:08[0m
[0;32mI (521) cpu_start: ELF file SHA256:  1440309b35e30ef9...[0m
[0;32mI (527) cpu_start: ESP-IDF:          v5.1.1-1-gd3c99ed-dirty[0m
[0;32mI (533) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (538) cpu_start: Max chip rev:     v3.99 [0m
[0;32mI (543) cpu_start: Chip rev:         v1.1[0m
[0;32mI (548) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (555) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (561) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (567) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (573) heap_init: At 3FFD13B8 len 0000EC48 (59 KiB): DRAM[0m
[0;32mI (579) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (585) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (592) heap_init: At 4009EEF4 len 0000110C (4 KiB): IRAM[0m
[0;32mI (599) spi_flash: detected chip: generic[0m
[0;32mI (603) spi_flash: flash io: dio[0m
[0;32mI (607) coexist: coex firmware version: 80b0d89[0m
[0;32mI (612) app_start: Starting scheduler on CPU0[0m
[... 18 lines cut ...]
I (693) wifi:Init data frame dynamic rx buffer num: 32
I (697) wifi:Init management frame dynamic rx buffer num: 32
I (701) wifi:Init management short buffer num: 32
I (703) wifi:Init dynamic tx buffer num: 64
I (707) wifi:Init static rx buffer size: 1600
I (711) wifi:Init static rx buffer num: 16
I (713) wifi:Init dynamic rx buffer num: 32
[0;32mI (717) wifi_init: rx ba win: 6[0m
[0;32mI (719) wifi_init: tcpip mbox: 32[0m
[0;32mI (723) wifi_init: udp mbox: 6[0m
[0;32mI (725) wifi_init: tcp mbox: 6[0m
[0;32mI (729) wifi_init: tcp tx win: 5744[0m
[0;32mI (733) wifi_init: tcp rx win: 5744[0m
[0;32mI (735) wifi_init: tcp mss: 1440[0m
[0;32mI (739) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (743) wifi_init: WiFi RX IRAM OP enabled[0m
[0;32mI (747) BTDM_INIT: BT controller compile version [946b762][0m
[0;32mI (751) BTDM_INIT: Bluetooth MAC: 08:d1:f9:00:b9:a2[0m
[0;32mI (757) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07[0m
[0;32mI (1085) FW_MAIN: ESP Bluetooth MAC addr:  8-d1-f9- 0-b9-a2
[0m
[0;32mI (1087) FW_SDIO_SLAVE: Using SDIO interface[0m
[0;32mI (1087) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [0m

No subsequent logs

hustzju, 2024-01-19:
I think  issue like #212, and  we use the same host MCU

mantriyogesh, 2024-01-19:
Can you please provide the dmesg log from boot?
Also what commands you fired and their output?

If the git commit same at ESP and host?

mantriyogesh, 2024-01-19:
> I think issue like #212, and we use the same host MCU.

I can't come to.such conclusion, yet. May / may not be.

> if hardware connection between ESP32 and host is not correct, would host print mmc CIS info?

If connections are correct, CIS will be populated. However that cannot guarantee that everything is correct or working.
Let me explain a bit.

Sdio init mode is distinctly different than normal working mode.
In init mode host pushes small clock and tries to read the sdio slave CIS (enumerate the sdio slave)
Here, all data pins may not be used.
Once this is done the normal  mode, the host will push normal clock configured as per the device tree. In normal mode, all required pins would be used, as per the bus width used.

Please cross check the Porting guide step wise.
If you don't use expected  pull-up registers or use jumper cables for sdio, you will face instability issues.

Let us know if you use sdio over the jumper cables.

hustzju, 2024-01-20:
And now I have moved forward,  Host could get boot event, but still have errors.

mantriyogesh, 2024-01-20:
1. Why the brownout is triggered at ESP, check voltages at all sd pins. Is ESP powered with correct voltage?
2. 
```
[ 96.299913] esp32_sdio:esp_reg_notifier: 629 esp_wifi_device not initialized yet
[ 96.355583] esp32_sdio:tx_process: Failed to send data: -84 512 512
```

Connection is not yet stable.
Using jumper wires? Can share setup picture?

hustzju, 2024-02-01:
Sorry for not contacting you for so long, I have since solved the last issue by rewiring the circuit board. And now I have another question, In the mass production phase, how to burn the efuse? Are there any specific tools for this purpose?

mantriyogesh, 2024-02-01:
Let me check on this.

mantriyogesh, 2024-02-01:
You can burn the efuse, whenever you download the firmware to flash.
If you burn the firmware in our module factory, we can burn efuse in the same time, if specified so.

hustzju, 2024-02-02:
[image]
we use this tool to burn firmware, but how to use it to burn efuse and "set_flash_voltage 3.3V" ? I just want to solve the gpio12 strapping conflict problem

hustzju, 2024-02-02:
Besides, I have another question. everything is seems to be OK, wlan0 is  found, 
and WIFI communication works. BUT when I check the host logs of loading esp32 driver, 
I find  words  like this "esp32_sdio:esp_reg_notifier: 629 esp_wifi_device not initialized yet ", 
Do I need to pay attention to this issue?

HOST side logs: (by the way, I changed the interface name to WLAN0)
root@myd-y6ull14x14:~# [  172.301034] mmc1: queuing unknown CIS tuple 0x01 (3 bytes)
[  172.319523] mmc1: queuing unknown CIS tuple 0x1a (5 bytes)
[  172.329065] mmc1: queuing unknown CIS tuple 0x1b (8 bytes)
[  172.342091] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  172.348046] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[  172.353802] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[  172.367158] mmc1: queuing unknown CIS tuple 0x80 (1 bytes)
[  172.372882] mmc1: queuing unknown CIS tuple 0x81 (1 bytes)
[  172.381931] mmc1: queuing unknown CIS tuple 0x82 (1 bytes)
[  172.389412] mmc1: new SDIO card at address 0001
[  172.404467] esp32_sdio:esp_probe: ESP network device detected
[  172.413054] esp32_sdio:get_firmware_data: Rx Pre ====== 0
[  172.421172] esp32_sdio:get_firmware_data: Rx Pos ======  0
[  172.428813] esp32_sdio:get_firmware_data: Tx Pre ======  0
[  172.434433] esp32_sdio:get_firmware_data: Tx Pos ======  10
[  172.448504] esp_sdio: probe of mmc1:0001:2 failed with error -22
[  173.532877] esp32_sdio:process_esp_bootup_event: 
[  173.532877] Received ESP bootup event
[  173.541454] esp32_sdio:process_event_esp_bootup: EVENT: 3
[  173.547065] esp32_sdio:process_event_esp_bootup: ESP chipset detected [esp32]
[  173.554250] esp32_sdio:process_event_esp_bootup: EVENT: 0
[  173.565257] esp32_sdio:process_capabilities: ESP peripheral capabilities: 0x1d
[  173.785141] esp32_sdio:process_capabilities: ESP Bluetooth init
[  173.798924] esp32_sdio:print_capabilities: Capabilities: 0x1d. Features supported are:
[  173.807501] esp32_sdio:print_capabilities: 	 * WLAN on SDIO
[  173.813177] esp32_sdio:print_capabilities: 	 * BT/BLE
[  173.818641] esp32_sdio:print_capabilities: 	   - HCI over SDIO
[  173.824568] esp32_sdio:print_capabilities: 	   - BT/BLE dual mode
[  173.831229] esp32_sdio:process_event_esp_bootup: EVENT: 4
[  173.836749] esp32_sdio:process_event_esp_bootup: EVENT: 1
[  173.842169] esp32_sdio:process_fw_data: ESP chipset's last reset cause:
[  173.850258] esp32_sdio:print_reset_reason: POWERON_RESET
[  173.855873] esp32_sdio:check_esp_version: esp32: ESP Firmware version: 1.0.2
[  173.863749] esp32_sdio:esp_reg_notifier: 629 esp_wifi_device not initialized yet 
[  173.941608] esp32_sdio:esp_cfg80211_get_tx_power: 

root@myd-y6ull14x14:~# ifconfig wlan0 up
root@myd-y6ull14x14:~# ifconfig wlan0   
wlan0     Link encap:Ethernet  HWaddr 08:d1:f9:00:b9:a0  
          inet6 addr: fe80::ad1:f9ff:fe00:b9a0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:26 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

qinyunliu, 2024-02-18:
blogger：I would like to ask how the spi solution is implemented as follows

kapilkedawat, 2024-02-19:
Hi @hustzju,

>> Besides, I have another question. everything is seems to be OK, wlan0 is found,
and WIFI communication works. BUT when I check the host logs of loading esp32 driver,
I find words like this "esp32_sdio:esp_reg_notifier: 629 esp_wifi_device not initialized yet ",
Do I need to pay attention to this issue?

This isn't an issue, we discard `reg set` command during driver initialization and this is expected.

closed 2024-02-22 by hustzju