espressif/esp-hosted #319: error esp-hosted-release-ng-v1.0.2??????

kangz543g, 2024-01-22:
esp-hosted-release-ng-v1.0.2.zip

make
rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=mips CROSS_COMPILE=/home/kkk/mips-gcc472-glibc216-64bit/bin/mips-linux-gnu- -C /home/kkk/Main/kernel M=/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32 clean
make[1]: 디렉터리 '/home/kkk/Main/kernel' 들어감
  CLEAN   /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/.tmp_versions
make[1]: 디렉터리 '/home/kkk/Main/kernel' 나감
make ARCH=mips CROSS_COMPILE=/home/kkk/mips-gcc472-glibc216-64bit/bin/mips-linux-gnu- -C /home/kkk/Main/kernel M=/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32 modules  # KERNEL을 KSRC로 수정
make[1]: 디렉터리 '/home/kkk/Main/kernel' 들어감
  CC [M]  /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.o
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c: In function 'esp_bt_send_frame':
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:91:19: error: invalid application of 'sizeof' to incomplete type 'struct esp_payload_header'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:92:27: error: invalid application of 'sizeof' to incomplete type 'struct esp_payload_header'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:137:25: error: invalid application of 'sizeof' to incomplete type 'struct esp_payload_header'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:139:5: error: dereferencing pointer to incomplete type
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:139:17: error: 'ESP_HCI_IF' undeclared (first use in this function)
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:139:17: note: each undeclared identifier is reported only once for each function it appears in
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:140:5: error: dereferencing pointer to incomplete type
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:141:5: error: dereferencing pointer to incomplete type
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:142:5: error: dereferencing pointer to incomplete type
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:148:30: error: 'ESP_CHECKSUM_ENABLED' undeclared (first use in this function)
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:149:6: error: dereferencing pointer to incomplete type
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:149:3: error: implicit declaration of function 'compute_checksum' [-Werror=implicit-function-declaration]
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:157:38: error: dereferencing pointer to incomplete type
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:309: /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.o] 오류 1
make[1]: *** [Makefile:1224: _module_/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32] 오류 2
make[1]: 디렉터리 '/home/kkk/Main/kernel' 나감
make: *** [Makefile:57: all] 오류 2

kangz543g, 2024-01-22:
make
rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=mips CROSS_COMPILE=/home/kkk/mips-gcc472-glibc216-64bit/bin/mips-linux-gnu- -C /home/kkk/Main/kernel M=/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32 clean
make[1]: 디렉터리 '/home/kkk/Main/kernel' 들어감
  CLEAN   /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/.tmp_versions
make[1]: 디렉터리 '/home/kkk/Main/kernel' 나감
make ARCH=mips CROSS_COMPILE=/home/kkk/mips-gcc472-glibc216-64bit/bin/mips-linux-gnu- -C /home/kkk/Main/kernel M=/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32 modules  # KERNEL을 KSRC로 수정
make[1]: 디렉터리 '/home/kkk/Main/kernel' 들어감
  CC [M]  /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.o
In file included from /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:19:0:
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:23:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:24:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:25:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:26:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:27:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:28:2: error: unknown type name 'uint16_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:29:2: error: unknown type name 'uint16_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:30:2: error: unknown type name 'uint16_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:31:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:35:3: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:36:3: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:37:3: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:43:2: error: unknown type name 'uint16_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:44:2: error: unknown type name 'uint16_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:45:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:46:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:47:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:48:2: error: unknown type name 'uint16_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:140:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:141:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:142:2: error: unknown type name 'uint16_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:143:2: error: unknown type name 'uint16_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:144:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:145:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:150:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:151:2: error: unknown type name 'uint16_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:153:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:154:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:159:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:160:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:165:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:166:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:167:2: error: unknown type name 'uint8_t'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/adapter.h:169:2: error: unknown type name 'uint8_t'
[... cut ...]
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.c:152:2: error: implicit declaration of function 'compute_checksum' [-Werror=implicit-function-declaration]
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:309: /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32/esp_bt.o] 오류 1
make[1]: *** [Makefile:1224: _module_/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_fg/host/linux/host_driver/esp32] 오류 2
make[1]: 디렉터리 '/home/kkk/Main/kernel' 나감
make: *** [Makefile:57: all] 오류 2

kangz543g, 2024-01-22:
kkk@kkk-Lenovo-ideapad-100-15IBY:~/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host$ grep -r "cfg80211_auth_request"
esp_cmd.h:		struct cfg80211_auth_request *req);
esp_cmd.c:		struct cfg80211_auth_request *req)
esp_cfg80211.c:		struct cfg80211_auth_request *req)

/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c: In function 'process_auth_event':
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:537:2: error: implicit declaration of function 'cfg80211_rx_mlme_mgmt' [-Werror=implicit-function-declaration]
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c: In function 'process_assoc_event':
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:573:2: error: implicit declaration of function 'cfg80211_rx_assoc_resp' [-Werror=implicit-function-declaration]
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c: In function 'cmd_auth_request':
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:912:45: error: 'struct cfg80211_auth_request' has no member named 'auth_data_len'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:925:26: error: 'struct cfg80211_auth_request' has no member named 'auth_data_len'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:926:28: error: 'struct cfg80211_auth_request' has no member named 'auth_data'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:926:44: error: 'struct cfg80211_auth_request' has no member named 'auth_data_len'

mantriyogesh, 2024-01-22:
@kapilkedawat ++

kapilkedawat, 2024-01-22:
@kangz543g suggest to update to latest code and retry. Also please share your kernel version.

kangz543g, 2024-01-22:
root@openipc-t21:/mnt/mmcblk0p1# insmod esp32_sdio.ko
insmod: can't insert 'esp32_sdio.ko': unknown symbol in module, or unknown parameter

kk@kkk-Lenovo-ideapad-100-15IBY:~/esp-idf/eps32_driver/esp32-linux$ ls
LICENSE         esp32_sdio.ko     esp_bt.c      esp_rb.h      main.o
Makefile        esp32_sdio.mod.c  esp_bt.o      esp_serial.c  modules.order
Module.symvers  esp32_sdio.mod.o  esp_bt_api.h  esp_serial.h  sdio
adapter.h       esp32_sdio.o      esp_if.h      hci.h         spi
esp.h           esp_api.h         esp_rb.c      main.c
kkk@kkk-Lenovo-ideapad-100-15IBY:~/esp-idf/eps32_driver/esp32-linux$ modinfo esp32_sdio.ko
filename:       /home/kkk/esp-idf/eps32_driver/esp32-linux/esp32_sdio.ko
depends:        
vermagic:       3.10.14__isvp_turkey_1.0__ preempt mod_unload MIPS32_R1 32BIT 
kkk@kkk-Lenovo-ideapad-100-15IBY:~/esp-idf/eps32_driver/esp32-linux$

kangz543g, 2024-01-22:
make
rm -rf *.o sdio/*.o spi/*.o *.ko
make ARCH=mips CROSS_COMPILE=/home/kkk/mips-gcc472-glibc216-64bit/bin/mips-linux-gnu- -C /home/kkk/Main/kernel M=/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host clean
make[1]: 디렉터리 '/home/kkk/Main/kernel' 들어감
  CLEAN   /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/.tmp_versions
make[1]: 디렉터리 '/home/kkk/Main/kernel' 나감
make ARCH=mips CROSS_COMPILE=/home/kkk/mips-gcc472-glibc216-64bit/bin/mips-linux-gnu- -C /home/kkk/Main/kernel M=/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host modules  # KERNEL을 KSRC로 수정
make[1]: 디렉터리 '/home/kkk/Main/kernel' 들어감
  CC [M]  /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_bt.o
  CC [M]  /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/main.o
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/main.c:315:8: warning: return type defaults to 'int' [-Wreturn-type]
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/main.c:315:8: warning: function declaration isn't a prototype [-Wstrict-prototypes]
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/main.c:315:8: warning: 'NDO_TX_TIMEOUT_PROTOTYPE' defined but not used [-Wunused-function]
  CC [M]  /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.o
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c: In function 'process_scan_result_event':
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:517:44: error: macro "CFG80211_INFORM_BSS" requires 11 arguments, but only 10 given
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:514:9: error: 'CFG80211_INFORM_BSS' undeclared (first use in this function)
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:514:9: note: each undeclared identifier is reported only once for each function it appears in
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c: In function 'process_auth_event':
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:539:2: error: implicit declaration of function 'cfg80211_rx_mlme_mgmt' [-Werror=implicit-function-declaration]
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c: In function 'process_assoc_event':
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:575:2: error: implicit declaration of function 'cfg80211_rx_assoc_resp' [-Werror=implicit-function-declaration]
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c: In function 'cmd_auth_request':
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:914:45: error: 'struct cfg80211_auth_request' has no member named 'auth_data_len'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:927:26: error: 'struct cfg80211_auth_request' has no member named 'auth_data_len'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:928:28: error: 'struct cfg80211_auth_request' has no member named 'auth_data'
/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.c:928:44: error: 'struct cfg80211_auth_request' has no member named 'auth_data_len'
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:309: /home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host/esp_cmd.o] 오류 1
make[1]: *** [Makefile:1224: _module_/home/kkk/esp-hosted-release-ng-v1.0.2/esp_hosted_ng/host] 오류 2
make[1]: 디렉터리 '/home/kkk/Main/kernel' 나감

mantriyogesh, 2024-01-22:
$ uname -a
output on target?

kangz543g, 2024-01-22:
uname -a
Linux openipc-t21 3.10.14__isvp_turkey_1.0__ #2 PREEMPT Sat Jan 6 15:49:53 UTC 2024 mips GNU/Linux

mantriyogesh, 2024-01-22:
Is it possible to at least move to 4.x ot 5.x kernel?

This kernel is very old.

kangz543g, 2024-01-22:
uname -a
Linux kkk-Lenovo-ideapad-100-15IBY 6.5.0-14-generic #14~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 18:15:30 UTC 2 x86_64 x86_64 x86_64 GNU/   use mips-gcc472-glibc216-64bit

mantriyogesh, 2024-01-22:
https://github.com/espressif/esp-hosted/issues/319#issuecomment-1903595101
This seems like your development environment.

https://github.com/espressif/esp-hosted/issues/319#issuecomment-1903588346
Looks target kernel version.

Is that correct?
Don't bother much, wrt Development environment kernel version .
Target kernel version is important.

kangz543g, 2024-01-22:
mips-gcc472-glibc216-64bit  --> eps32 linux usb support?

mantriyogesh, 2024-01-22:
USB is not supported.
We support SPI or SDIO or UART as if now.

So you can see https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/docs/Linux_based_host/rpi_esp32_c6_setup.jpg for reference.

The ESP (any supported) and Linux should have SPI/SDIO/UART connections as per documentation.
ESP connected in USB as serial port is needed to power the ESP and flashing ESP & get the UART logs of ESP.
But other than that, we do not have USB support yet for ESP-Hosted.

kangz543g, 2024-01-22:
https://github.com/aliengreen/esp32_uart_bridge  <----  usd esp32-c3 usb to serial.
ttyAM0 is not supported in 32bit mips liunx.

kangz543g, 2024-01-22:
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382808
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xb9c
load:0x403ce710,len:0x2e40
entry 0x403cc71a
I (15) boot: ESP-IDF v5.3-dev-1353-gb3f7e2c8a4 2nd stage bootloader
I (15) boot: compile time Jan 22 2024 20:39:40
I (16) boot: chip revision: v0.4
I (20) boot.esp32c3: SPI Speed      : 80MHz
I (25) boot.esp32c3: SPI Mode       : DIO
I (29) boot.esp32c3: SPI Flash Size : 2MB
I (34) boot: Enabling RNG early entropy source...
I (40) boot: Partition Table:
I (43) boot: ## Label            Usage          Type ST Offset   Length
I (50) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (58) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (65) boot:  2 factory          factory app      00 00 00010000 00100000
I (73) boot: End of partition table
I (77) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0b560h ( 46432) map
I (93) esp_image: segment 1: paddr=0001b588 vaddr=3fc8c400 size=012bch (  4796) load
I (95) esp_image: segment 2: paddr=0001c84c vaddr=40380000 size=037cch ( 14284) load
I (105) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1b09ch (110748) map
I (128) esp_image: segment 4: paddr=0003b0c4 vaddr=403837cc size=08b48h ( 35656) load
I (139) boot: Loaded app from partition at offset 0x10000
I (139) boot: Disabling RNG early entropy source...
I (150) cpu_start: Unicore app
I (159) cpu_start: Pro cpu start user code
I (159) cpu_start: cpu freq: 160000000 Hz
I (159) cpu_start: Application information:
I (162) cpu_start: Project name:     esp32_uart_bridge
I (168) cpu_start: App version:      f35b335
I (173) cpu_start: Compile time:     Jan 22 2024 20:41:12
I (179) cpu_start: ELF file SHA256:  e89daea07...
I (184) cpu_start: ESP-IDF:          v5.3-dev-1353-gb3f7e2c8a4
I (191) cpu_start: Min chip rev:     v0.3
I (195) cpu_start: Max chip rev:     v1.99 
I (200) cpu_start: Chip rev:         v0.4
I (205) heap_init: Initializing. RAM available for dynamic allocation:
I (212) heap_init: At 3FC8E670 len 00031990 (198 KiB): RAM
I (218) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (225) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (232) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (239) spi_flash: detected chip: generic
I (243) spi_flash: flash io: dio
W (247) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (260) sleep: Configure to isolate all GPIO pins in sleep state
I (267) sleep: Enable automatic switching of GPIO sleep configuration
I (274) main_task: Started on CPU0
I (274) main_task: Calling app_main()
I (274) gpio: GPIO[7]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (284) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (294) main_task: Returned from app_main()

reference) /dev/ttyACM0    thank you.

kangz543g, 2024-01-23:
<pre>make
make ARCH=mips CROSS_COMPILE=/home/kkk/mips-gcc472-glibc216-64bit/bin/mips-linux-gnu- -C /home/kkk/Main/kernel M=/home/kkk/esp32c3  modules
make[1]: 디렉터리 &apos;/home/kkk/Main/kernel&apos; 들어감
  CC [M]  /home/kkk/esp32c3/eps32c3_serial.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/kkk/esp32c3/eps32c3_serial.mod.o
  LD [M]  /home/kkk/esp32c3/eps32c3_serial.ko
</pre>

kangz543g, 2024-01-23:
/* Serial interface */
#define SERIAL_IF_FILE                            "/dev/ttyACM0"
I don't know how.   linux source???

kangz543g, 2024-01-23:
modinfo eps32c3_serial.ko
filename:       /home/kkk/esp32c3/eps32c3_serial.ko
license:        GPL
author:         <kangz543g@gmail.com>
description:    EPS32C3 USB to serial adaptor driver
alias:          usb:v303Ap1001d*dc*dsc*dp*ic*isc*ip*in*
depends:        
vermagic:       3.10.14__isvp_turkey_1.0__ preempt mod_unload MIPS32_R1 32BIT

root@openipc-t21:
root@openipc-t21:/mnt/mmcblk0p1# lsusb
Bus 001 Device 003: ID 303a:1001
Bus 001 Device 001: ID 1d6b:0002

root@openipc-t21:/mnt/mmcblk0p1# lsmod | grep eps32c3_serial
eps32c3_serial          6812  0 [permanent]

root@openipc-t21:ls /dev/tty*
/dev/tty    /dev/ttyp1  /dev/ttyp4  /dev/ttyp7  /dev/ttypa  /dev/ttypd
/dev/ttyS1  /dev/ttyp2  /dev/ttyp5  /dev/ttyp8  /dev/ttypb  /dev/ttype
/dev/ttyp0  /dev/ttyp3  /dev/ttyp6  /dev/ttyp9  /dev/ttypc  /dev/ttypf

root@openipc-t21:dmesg | tail
[   18.442363] wait stable.[288][cgu_cim]
[   18.442385] set sensor gpio as PA-low-10bit
[   18.442401] probe ok ------->jxf37
[   18.780041] -----jxf37_detect: 623 ret = 0, v = 0x0f
[   18.780461] -----jxf37_detect: 631 ret = 0, v = 0x37
[   18.780473] jxf37 chip found @ 0x40 (i2c0)
[   18.780482] sensor driver version H20200416a
[   18.931111] jxf37 stream on
[  134.429845] usb 1-1: USB disconnect, device number 2
[  138.228618] usb 1-1: new full-speed USB device number 3 using dwc2

kangz543g, 2024-01-23:
not resolved.

mantriyogesh, 2024-01-23:
We are still trying to understand exact issue you are facing.

kangz543g, 2024-01-23:
without /dev/ttyUSB0 or /dev/ttyAMCM0

#define ESP_SERIAL_MAJOR      		221
#define ESP_SERIAL_MINOR_MAX  	2

mknod /dev/ttyUSB0 c 221 2   ?????

root@openipc-t21:/usr/bin# microcom /dev/ttyUSB0 -s 38400
microcom: can't open '/dev/ttyUSB0': No such device or address

kangz543g, 2024-01-24:
eps32c3_serial.ko is not supported.

root@openipc-t21:/mnt/mmcblk0p1# insmod cdcacm.ko

root@openipc-t21:/mnt/mmcblk0p1# ls /dev/tty*
/dev/tty /dev/ttyp1 /dev/ttyp5 /dev/ttyp9 /dev/ttypd
/dev/ttyACM1 /dev/ttyp2 /dev/ttyp6 /dev/ttypa /dev/ttype
/dev/ttyS1 /dev/ttyp3 /dev/ttyp7 /dev/ttypb /dev/ttypf
/dev/ttyp0 /dev/ttyp4 /dev/ttyp8 /dev/ttypc

root@openipc-t21:/mnt/mmcblk0p1# ls /dev/ttyA*
/dev/ttyACM1

root@openipc-t21:/mnt/mmcblk0p1#   esp32c3 usb driver ok!!!!    the end.

mantriyogesh, 2024-01-24:
Do you have any GitHub repo to refer the code?

kangz543g, 2024-01-24:
https://codebrowser.dev/linux/linux/drivers/usb/class/cdc-acm.c.html  <---   /dev/ttyACMx supported.

kangz543g, 2024-01-24:
ESP32C3 APPLICATION PROGRAM(concept possible)  - i am a deaf. linux yes,esp32 yes!!!!

[CODE]
/*

serial_usb_simple_arduino - For communicating over USB serial. Send it a '1' (character one)
and it will make the builtin LED start blinking every one second. Send it a '0'
(character zero) and it will make it stop blinking.
Each time it receives one of the commands, it sends back an 'A' for acknowledge.
But send it a commmand it doesn't recognize and it sends back an 'E' for error.
*/
bool blinking = false;
bool led_on = false;
int target_time;

void setup() {
Serial.begin(115200);
while (!Serial) {
; // wait for serial port to connect. Needed for native USB
}
pinMode(8, OUTPUT);
}

void loop() {
char c;

if (Serial.available() > 0) {
c = Serial.read();
switch (c) {
case '0':
// stop blinking
blinking = false;
if (led_on) {
digitalWrite(8, LOW);
}
Serial.write("A", 1);
break;
case '1':
// start blinking
if (blinking == false) {
blinking = true;
digitalWrite(8, HIGH);
led_on = true;
target_time = millis() + 100; // turn off in 1 tenth of a second (100 milliseconds)
}
Serial.write("A", 1);
break;
default:
Serial.write("E", 1);
break;
}
} else if (blinking) {
if (millis() >= target_time) {
if (led_on) {
digitalWrite(8, LOW);
led_on = false;
target_time = millis() + 100; // turn on in 1 tenth of a second (100 milliseconds)
} else {
digitalWrite(8, HIGH);
led_on = true;
target_time = millis() + 100; // turn off in 1 tenth of a second (100 milliseconds)
}
}
}
}

[Serial Testing]
root@openipc-t21:~# ls /dev/tty*
/dev/tty /dev/ttyp1 /dev/ttyp5 /dev/ttyp9 /dev/ttypd
/dev/ttyACM0 /dev/ttyp2 /dev/ttyp6 /dev/ttypa /dev/ttype
/dev/ttyS1 /dev/ttyp3 /dev/ttyp7 /dev/ttypb /dev/ttypf
/dev/ttyp0 /dev/ttyp4 /dev/ttyp8 /dev/ttypc

root@openipc-t21:# echo 1 > /dev/ttyACM0
root@openipc-t21:# echo 0 > /dev/ttyACM0
root@openipc-t21:~# echo 1 > /dev/ttyACM0
[image]