espressif/esp-hosted #322: esp_hosted_fg ：cannot connect to the Internet in softap mode

wky19971002, 2024-01-25:
We are now having a problem.We used the 4g modulus as the dhcp server,esp32 successfully obtains the dhcp arp broadcast message,and dhcp arp ack message from  4g modulus is sent to esp32，it seems  get a right IP address，but it still cannot work.
esp log:
[image]
[esp_log.zip](https://github.com/espressif/esp-hosted/files/14048092/esp_log.zip)

when I ping 8.8.8.8, I can see this packet's log in wlan_ap_rx_callback(),it released by this judgment,
[image]
esp log:
[image]

So I wonder if the broadcast package is handled differently than the icmp package.

mantriyogesh, 2024-01-25:
Hello @wky19971002 ,

Thanks for the raising this issue.

Can you please comment;
https://github.com/espressif/esp-hosted/blob/fc02a0abee729dc374ed9e565eda56c884a1d52a/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c#L224-L229

and try again?
We are going to remove this in master.

wky19971002, 2024-01-25:
> Hello @wky19971002 ,
> 
> Thanks for the raising this issue.
> 
> Can you please comment;
> 
> https://github.com/espressif/esp-hosted/blob/fc02a0abee729dc374ed9e565eda56c884a1d52a/esp_hosted_fg/esp/esp_driver/network_adapter/main/app_main.c#L224-L229
> 
> and try again? We are going to remove this in master.

thanks,I'll try.

mantriyogesh, 2024-04-24:
Closing due to inactivity

closed 2024-04-24 by mantriyogesh