espressif/esp-hosted #323: build error at esp_hosted_ng with latest master branch

Psychesnet, 2024-01-25:
Hi, 

I try to test esp_hosted_ng from master branch, there is an error, how to resolve it? Thanks.
```
/Users/eric/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(cmd.c.obj):(.literal.sta_connection+0xc): undefined reference to `esp_wifi_connect_internal'
/Users/eric/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(cmd.c.obj):(.literal.process_auth_request+0x34): undefined reference to `esp_wifi_issue_auth_internal'
/Users/eric/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(cmd.c.obj):(.literal.process_assoc_request+0x4): undefined reference to `esp_wifi_issue_assoc_internal'
/Users/eric/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(cmd.c.obj): in function `sta_connection':
/Users/eric/git/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:127: undefined reference to `esp_wifi_connect_internal'
/Users/eric/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(cmd.c.obj): in function `process_auth_request':
/Users/eric/git/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:1110: undefined reference to `esp_wifi_issue_auth_internal'
/Users/eric/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/main/libmain.a(cmd.c.obj): in function `process_assoc_request':
/Users/eric/git/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/main/cmd.c:1225: undefined reference to `esp_wifi_issue_assoc_internal'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /Users/eric/git/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/log/idf_py_stderr_output_15463 and /Users/eric/git/esp-hosted/esp_hosted_ng/esp/esp_driver/network_adapter/build/log/idf_py_stdout_output_15463
```

The latest commit is fc02a0abee729dc374ed9e565eda56c884a1d52a.

Following is how to build this branch, something wrong? Thanks.
```
  git clone --recurse-submodules https://github.com/espressif/esp-hosted.git
  cd esp-hosted/
  git submodule update --init --recursive
  cd esp_hosted_ng/esp/esp_driver/
  source esp-idf/export.sh 
  cd network_adapter/
  cp sdkconfig.defaults.esp32 sdkconfig
  idf.py build
```

mantriyogesh, 2024-01-25:
@kapilkedawat ++

kapilkedawat, 2024-01-25:
Please copy libraries as well https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/esp/esp_driver/setup.sh#L11

Psychesnet, 2024-01-26:
Hi @kapilkedawat 

Thanks. I works.

closed 2024-01-26 by Psychesnet

Psychesnet, 2024-01-26:
Hi @kapilkedawat 

Just want to confirm one thing, looks like esp_hosted_ng did not support soft-ap, am I right? Thanks.

```
Following operations supported as of now:

Scanning
Connect to AP
Disconnect from AP

```

kapilkedawat, 2024-01-26:
Hi @Psychesnet, Yes softAP is not supported at the moment.

Psychesnet, 2024-01-26:
Hi @kapilkedawat , Thanks. Have a good day.