espressif/esp-hosted #325: esp_hosted_ng: the behavior support?

Psychesnet, 2024-01-29:
Hi,

Currently, my device under esp_hosted_ng, it can work. I want to check if the following behavior support or not. Thanks.

1. My device has two chips, first is ARM and second is ESP32.
2. At normal case, both chips are power on. ARM system use wpa_supplicant to control ESP32 and connect to the internet.
    Question: Can ESP32 create connection to Amazon server and send/recv package from internet at ESP32 side? 
    Explain: We hope ESP32 has private connection to send/recv MQTT package and handle status LEDs.
3. At power saving case, ARM is off and ESP32 is on.
    Question: Can ESP32 watch the incoming data from internet and wake up the ARM system?
    Explain: At normal case, ARM system already have the session to keep wifi connection. We hope ESP32 can share the session between ARM system and ESP32. Once ESP32 watch incoming data and wake up ARM system, ARM system still can use the same connection to response data.
    
  I hope anything is clear. If not, please let me know. Thanks.

mantriyogesh, 2024-01-29:
(3) is completely supported, please check [host_sleep.md](https://github.com/espressif/esp-hosted/blob/master/esp_hosted_ng/docs/host_sleep.md)

(2) for connecting to amazon/web, you would need IP connectivity.
This can only be done from host, as host runs TCP/TP protocols and not the slave.

obviously, you can run ip suing standalone esp chipset, but that would not be ESP-Hosted as such then.
 ESP-Hosted needs a transport, say SPI/SDIO. So Apart from the transport currently in use, you can use all other things like GPIOs, peripherals etc from slave also. but for reporting as such, you can send the accumulated data to host and send using IP address.

mantriyogesh, 2024-01-29:
Any specific details you think needed, we can definitely discuss here further.
You also can customize the code as per your need, as the code is all open-source.

If you need some kind of  software support as such, can contact https://www.espressif.com/en/contact-us/technical-inquiries

Psychesnet, 2024-01-31:
Hi @mantriyogesh 

Thanks. I will study the source code first.

closed 2024-01-31 by Psychesnet

mantriyogesh, 2024-01-31:
Shoot your questions any time.
Happy to anwer.