espressif/esp-hosted #327: esp32c3 as a network module

lipengvita, 2024-02-01:
I want to use esp32c3 chip and esp-hosted-ng code to Implementing Linux to control ESP32C3 to achieve wireless network card function。
I have two question
1 Dose esp32c3 and esp-hosted-ng support wifi+bt Implement the translation of wireless network card functions only by spi , not use uart
2 IF I use esp32c3 and a linux box , What are the sizes of DDR and FLASH that you recommend?

mantriyogesh, 2024-02-01:
Hello !
(1)Yes. Definitely supported.
(2) Flash size at C3? 4MB should be enough.
Kernel module has networking stack integration. Accordingly, the RAM could be used.
the general dynamic allocation only happens in perspective of packet, otherwise there is not much to reserve.
Do you run wpa_supplicant? If so, NG would be better.

If you have very restricted environment, FG also could be used.

qinyunliu, 2024-02-18:
I also want to know the solution of this aspect, may I ask if it has been solved

mantriyogesh, 2024-02-18:
Your question is answered already. If you have nothing more to ask in this regard, we can close this issue.

mantriyogesh, 2024-02-18:
Please check the documentation and let us know if you have any specific question as such
The chipses, ESP32, S2,S3,C3,C6 are already supported for both flavors, FG and NG.

closed 2024-02-19 by lipengvita