espressif/esp-hosted #328: Error in rpi_init.sh

NikolayTsvetkovv, 2024-02-01:
When you set ESP_SLAVE_CHIPSET="esp32c6" and it goes to this case:
[image]

It expects '-' between "32" and "c"

mantriyogesh, 2024-02-01:
Both cases are covered, right?
https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L211-L218

Basically. `ESP32C6` or `ESP32-C6` both should be able to be used.

NikolayTsvetkovv, 2024-02-01:
It seems I used old version. Yes you are right, the problem is fixed in this file: https://github.com/espressif/esp-hosted/blob/master/esp_hosted_fg/host/linux/host_control/rpi_init.sh#L211-L218

closed 2024-02-20 by mantriyogesh