espressif/esp-hosted #330: esp32-c3 -mini-1 + ESP-Hosted-FG-release_0.0.5 no work

wujique, 2024-02-02:
mcu:stm32f407
just spi
esp32-c3 -mini-1 
ESP-Hosted-FG-release_0.0.5

unknown type 4

----------------------data log---------------------
ESP-Hosted for ESP32-C2/C3/C6/S2/S3
retval:1600
00000000:  04 00 0E 00 0C 00 3F 01  00 00 00 00 00 0C 02 01  ......?.........
00000010:  05 01 01 1E 00 01 E8 03  01 00 00 00 00 00 00 00  ................
00000020:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000030:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000040:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000050:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000060:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000070:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000080:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000090:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000000A0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000000B0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000000C0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000000D0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000000E0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000000F0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000100:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000110:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000120:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000130:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000140:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000150:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000160:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000170:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000180:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000190:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000001A0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000001B0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000001C0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000001D0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000001E0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000001F0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000200:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000210:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000220:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000230:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000240:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000250:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000260:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000270:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000280:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000290:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000002A0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000002B0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000002C0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000002D0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000002E0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000002F0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000300:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000310:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000320:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000330:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000340:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000350:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000360:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000370:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000380:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00000390:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000003A0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
000003B0:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
[... 119 lines cut ...]
000004C0:  15 C5 C0 9A 8F 1D CC 6A  3A 37 88 4B 84 AF 9E E5  .......j:7.K....
000004D0:  4B 09 23 C0 2E 95 81 34  E8 37 BB FA 64 7C B2 45  K.#....4.7..d|.E
000004E0:  3C 94 DD 0E C3 B9 EA 0E  4B 4E E5 53 11 26 27 D2  <.......KN.S.&'.
000004F0:  54 2C 85 57 06 53 75 BB  D5 94 22 14 00 74 17 DF  T,.W.Su..."..t..
00000500:  6D 18 24 70 2A 1A 42 78  A0 C2 F2 2A 7A 4A 12 BF  m.$p*.Bx...*zJ..
00000510:  48 DB D8 32 ED 18 50 7D  96 23 A7 6F 73 6D AD 9D  H..2..P}.#.osm..
00000520:  B7 07 FA 29 93 50 F8 51  C4 60 B5 DB 81 F3 94 2C  ...).P.Q.`.....,
00000530:  A3 B3 C9 7E 0A 19 B8 81  33 99 70 CF C6 B7 9F 7F  ...~....3.p.....
00000540:  5E F0 3D 9D B3 EB 79 CF  39 E0 11 00 BA 34 A4 5D  ^.=...y.9....4.]
00000550:  84 E5 80 BF 03 AD 93 BA  17 31 34 29 37 17 4C 1E  .........14)7.L.
00000560:  9B C6 9C 1D EB 42 9B F8  D1 3D 68 94 14 0F 8E 8A  .....B...=h.....
00000570:  4A BE 8A F4 73 15 8B 6D  E5 93 F3 BF BB 1A 7F C0  J...s..m........
00000580:  A7 6A 3A 21 A5 05 FA C4  9E 1A 56 EB 42 8E 54 AB  .j:!......V.B.T.
00000590:  9B E5 DE 1A 18 C6 2E F4  C6 ED AF 9D C0 11 D7 51  ...............Q
000005A0:  8C 22 AC 72 82 20 42 B5  0D 2D 78 5A F1 2E 5B 78  .".r. B..-xZ..[x
000005B0:  2E 19 4F 09 CD 72 FA EA  5B D5 3F 94 4A 9B FB F5  ..O..r..[.?.J...
000005C0:  23 7A 7A 22 06 32 70 1E  A1 AE EF F5 28 0F A3 61  #zz".2p.....(..a
000005D0:  B9 DF 8F CC 32 D9 B2 3E  D5 B2 3E B3 1C 16 78 69  ....2..>..>...xi
000005E0:  1A 32 79 CA A6 51 2C 70  3D 2B 79 F9 54 07 CD 7A  .2y..Q,p=+y.T..z
000005F0:  88 D5 86 8A E3 15 11 D9  3F 0C 58 B1 D0 6E 92 BD  ........?.X..n..
00000600:  9D 46 91 8D EB B3 28 C3  41 3B D0 71 C2 E7 80 4B  .F....(.A;.q...K
00000610:  E8 C9 90 B8 DB DC 1B 20  B3 B1 D5 24 9D AC 3F D3  ....... ...$..?.
00000620:  31 8B 07 77 90 52 A2 0D  DD 78 D2 A1 B3 28 C1 7C  1..w.R...x...(.|
00000630:  B1 5F 7D 45 CD 34 43 ED  7B 18 60 9D 3C B3 4F 5E  ._}E.4C.{.`.<.O^
checksum:0550

wujique, 2024-02-03:
the checksum is ok。

mantriyogesh, 2024-02-03:
I can see that the unknown type.
Are you using same `git commit` at host and slave?

wujique, 2024-02-03:
build a new ESP-Hosted-FG， work
ESP-Hosted-FG-release_0.0.5 x

mantriyogesh, 2024-02-03:
Great. You can use master, as it has many fixes pushed.