espressif/esp-hosted #332: c3 driver will panic when reset password

xmlhp, 2024-02-06:
[esp32c3 panic.zip](https://github.com/espressif/esp-hosted/files/14174189/esp32c3.panic.zip)
After I connect to the hotspot normally.
I want to reconfigure the password, then connect, I follow the steps below, the driver will crash
kill udhpc
2.kill wpa_supplicant
3. Reconfigure wpa_supplicant (SSID and password). This step is not in the diary, so wpa_supplicant does not need to change SSID and password
Reopen wpa_supplicant
5. Try udhcpc again, it will crash.

If it will be after step2, rmmod esp32c3.ko, power down the module, and then on, and then reload ko will not.
However, we also run bluetooth service, Bluetooth service can not stop at this time.

xmlhp, 2024-02-19:
Excuse me, is there a solution to this problem @mantriyogesh @kapilkedawat

mantriyogesh, 2024-02-19:
Apologies for the delay from our side.
Let me check this again

mantriyogesh, 2024-02-19:
Can you please attach the ESP logs as well?
Trying to understand if ESP had crashed, and which messages it had received / responded before crash at host

xmlhp, 2024-02-19:
[esp32c3 panic.zip](https://github.com/espressif/esp-hosted/files/14174189/esp32c3.panic.zip)
It is the log @linux?Do you want the log@esp32c3 side?

mantriyogesh, 2024-02-19:
yes, tested at the same time, both side logs.

kapilkedawat, 2024-02-19:
@xmlhp Please try with attached patch.
[0001-esp_hosted_ng-fixed-some-concurrency-issues.patch](https://github.com/espressif/esp-hosted/files/14326652/0001-esp_hosted_ng-fixed-some-concurrency-issues.patch)

Also please check the device connections, commands are getting timed out.

xmlhp, 2024-02-19:
[linux_side.zip](https://github.com/espressif/esp-hosted/files/14327192/linux_side.zip)
[c3_side.zip](https://github.com/espressif/esp-hosted/files/14327194/c3_side.zip)
linux_side.zip ,the log file @ linux side 
c3_side.zip the esp32c3 side log 
@mantriyogesh @kapilkedawat

xmlhp, 2024-02-19:
@xmlhp Please try with attached patch.
[0001-esp_hosted_ng-fixed-some-concurrency-issues.patch](https://github.com/espressif/esp-hosted/files/14326652/0001-esp_hosted_ng-fixed-some-concurrency-issues.patch)

Also please check the device connections, commands are getting timed out.

@kapilkedawat ,    what  git commits for this patch?

xmlhp, 2024-02-19:
I didn't found the file sp_stats.h esp_utils.h at host_ng
@kapilkedawat @mantriyogesh

kapilkedawat, 2024-02-19:
Hi @xmlhp, are you using latest master?

xmlhp, 2024-02-19:
Now, I can build ko for lastet master.

kapilkedawat, 2024-02-19:
@xmlhp  this change has been merged in latest master. You can also pull latest master and check if you are still facing this crash.

kapilkedawat, 2024-02-20:
@xmlhp please let us know if you are still facing this issue.

xmlhp, 2024-02-20:
@kapilkedawat ，now, I can reset ssid and password.

xmlhp, 2024-02-20:
@kapilkedawat Later, if you have any questions, wii let me know

kapilkedawat, 2024-02-20:
@xmlhp we will close this issue in that case, feel free to reopen or file another issue in case you face any other issue :).

closed 2024-02-20 by mantriyogesh

xmlhp, 2024-02-21:
@kapilkedawat ，ok